coolsnowwolf/lede #6590: k2p 选了wpad 编译失败

yjd, 2021-03-23:
### (1) 具体问题

`最新代码配置选了 Network --> WirelessAPD -> wpad 编译就失败`

### (2) 路由器型号和固件版本

`k2p A2`

### (3) 详细日志

```
<artificial>:(.text.ap_sta_add+0x172): undefined reference to `sta_track_claim_taxonomy_info'
collect2: error: ld returned 1 exit status
Makefile:680: recipe for target '/home/yjd/openWRT_x96/lede/build_dir/target-mipsel_24kc_musl/hostapd-wpad-full-internal/hostapd-2020-06-08-5a8b3662/.built' failed
make[3]: *** [/home/yjd/openWRT_x96/lede/build_dir/target-mipsel_24kc_musl/hostapd-wpad-full-internal/hostapd-2020-06-08-5a8b3662/.built] Error 1
make[3]: Leaving directory '/home/yjd/openWRT_x96/lede/package/network/services/hostapd'
time: package/network/services/hostapd/wpad-full-internal/compile#69.43#5.39#72.02
package/Makefile:111: recipe for target 'package/network/services/hostapd/compile' failed
make[2]: *** [package/network/services/hostapd/compile] Error 2
make[2]: Leaving directory '/home/yjd/openWRT_x96/lede'
package/Makefile:107: recipe for target '/home/yjd/openWRT_x96/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/yjd/openWRT_x96/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/yjd/openWRT_x96/lede'
/home/yjd/openWRT_x96/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2
```

AmadeusGhost, 2021-03-23:
k2p 不是用的闭源驱动么，为什么要选 wpad。。

yjd, 2021-03-23:
> k2p 不是用的闭源驱动么，为什么要选 wpad。。
原来是这样，谢谢了！
用不上802.11R了。

Neverends4, 2021-03-24:
如果要换开源驱动先把闭源那一套取消，否则会编译出错。
开源驱动的优点是支持kvr和WPA3/SAE

yjd, 2021-03-24:
> 如果要换开源驱动先把闭源那一套取消，否则会编译出错。
> 开源驱动的优点是支持kvr和WPA3/SAE

我看默认是选了 kmod-mt7615d kmod-mt7615d_dbdc
开源的是哪一项？还是说要自己找驱动

AmadeusGhost, 2021-03-24:
k2p 这种放弃开源驱动吧，luci 上无线的显示和控制还有问题

yjd, 2021-03-24:
> k2p 这种放弃开源驱动吧，luci 上无线的显示和控制还有问题

好的。那就不搞了。

Neverends4, 2021-03-24:
> 
> 我看默认是选了 kmod-mt7615d kmod-mt7615d_dbdc
> 开源的是哪一项？还是说要自己找驱动

我的红米2100是：
-mtwifi、mt7603e、mt7615d
+mt7603、mt7615-fireware、 mt7615e，以及wpad-basic-openssl（如果你要组mesh就用wpad-openssl）

开源驱动去年还很差，但是现在已经很不错了，至少我用着很好，和LS说的相反，Luci可以显示2.4/5g分别连接的客户端，闭源的却不行。L大说用闭源的原因是支持硬件NAT，但是目前还没实现，我试了开源没有NAT测速能到300M，不玩游戏足够了，你可以自己试。

TXH201622, 2021-03-24:
> 我的红米2100是：
> -mtwifi、mt7603e、mt7615d
> +mt7603、mt7615-fireware、 mt7615e，以及wpad-basic-openssl（如果你要组mesh就用wpad-openssl） [...]

楼上没有说错，K2P是一个7615同时发送2.4G和5G，和你的红米情况完全不一样，开源驱动图形界面现在是有问题的。

closed 2021-03-25 by yjd

siren202101, 2021-07-29:
> > 我看是选了kmod-mt7615d kmod-mt7615d_dbdc
> > 默认开源数学？还是说要自己找驱动
> 
> 我的红米2100是：
> -mtwifi、mt7603e、mt7615d
> +mt7603、mt7615-fireware、mt7615e，以及wpad-basic-openssl（如果你要组网就用wpad-openssl）
> 
> 开源驱动去年还差，但现在已经很不错了，至少我用着很好，和LS说的相反，Luci可以显示2.4/5g分别的客户端，闭源的不连接。L大说用闭。源的原因是支持硬件NAT，但目前还没有实现，我可以尝试开源没有NAT测速能到300M，不玩游戏网络了，你自己试一下。
大佬您好，请问呢一下-mtwifi、mt7603e、mt7615d，这个前面带个-号是什么意思？

Neverends4, 2021-07-30:
> > 大佬您好，请问呢一下-mtwifi、mt7603e、mt7615d，这个前面带个-号是什么意思？

取消make menuconfig里的勾选。+是增加勾选。
现在最新的源码默认就是开源wifi驱动，不用再改了
另外不要在关闭的issue回复。