coolsnowwolf/lede #6591: mlx5驱动模块故障

Aes64X, 2021-03-24:
kmod-mlx5-core
先屏蔽或者删除吧，这个驱动模块在5.4和5.10内核下会造成内核崩溃，上电加载一闪就崩溃的那种。。。。根本来不及看错误记录屏幕就闪黑重启，然后就是无限循环重启。
但只有在相应硬件存在的情况下，比如 Mellanox ConnectX-4 或 ConnectX-4 Lx 系列
不插卡就没事，插上 MCX 4121-ACAT 必崩。。。。。。
kmod-mlx4-core 支持的 ConnectX-3 及ConnectX-3 PRO 系列未知

kmod-mlx5-core 目前测试的只在4.19下工作正常

hyird, 2021-03-24:
[image]
我表示没有遇到

closed 2021-03-27 by Aes64X

nasbdh9, 2021-03-31:
可以试试https://github.com/nasbdh9/openwrt/commit/c0dc6010934e88223182b3cb0affbb1ada478539

Aes64X, 2021-04-01:
> 可以试试[nasbdh9/openwrt@c0dc601](https://github.com/nasbdh9/openwrt/commit/c0dc6010934e88223182b3cb0affbb1ada478539)

试过了，仍然重启，目前怀疑只有Lean的有这个问题，用MCX4的不少，OP主线那边居然没动静。。。。。
（另外多谢大佬。。。。从你那抄来的bnxt驱动，之前我都是直接配置内核 - -）