coolsnowwolf/lede #6597: newifi3 编译失败

00LT00, 2021-03-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
编译失败，不太清楚具体是哪部分模块不能选，或者说到底这个app编译成功了没

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
有，但都没人回答，然后过了半年左右大佬直接close了

## 2. 详细叙述

### (1) 具体问题

A：如题，编译失败

### (2) 路由器型号和固件版本

A：newifi3，版本就是最新的源码克隆的

### (3) 详细日志

A：异常开始的日志如下
```
cp -fpR /home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips /home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root.orig-ramips
/home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/arpbind: line 5: /lib/functions.sh: No such file or directory
/home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/arpbind: line 6: /lib/functions/network.sh: No such file or directory
mkdir: cannot create directory '/etc/config': Permission denied
/home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/guest-wifi: line 10: uci: command not found
/home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/guest-wifi: line 11: uci: command not found
/home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/guest-wifi: line 12: uci: command not found
/home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/guest-wifi: line 13: uci: command not found
/home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/guest-wifi: line 14: uci: command not found
/home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/guest-wifi: line 15: uci: command not found
/home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/guest-wifi: line 16: uci: command not found
/home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/guest-wifi: line 17: uci: command not found
/home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/guest-wifi: line 18: uci: command not found
/home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/guest-wifi: line 19: uci: command not found
/home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/guest-wifi: line 20: uci: command not found
/home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/guest-wifi: line 21: uci: command not found
/home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/openclash: line 3: /usr/share/openclash/openclash_ps.sh: No such file or directory
/home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/openclash: line 4: /usr/share/openclash/ruby.sh: No such file or directory
/home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/openclash: line 13: uci: command not found
/home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/openclash: line 14: uci: command not found
/home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/openclash: line 17: uci: command not found
sed: can't read /usr/lib/lua/luci/dispatcher.lua: No such file or directory
sed: can't read /usr/lib/lua/luci/dispatcher.lua: No such file or directory
chmod: cannot access '/sbin/mtkwifi': No such file or directory
/home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/nginx: line 12: /usr/bin/nginx-util: No such file or directory
Enabling AdGuardHome
Enabling adbyby
./etc/init.d/arpbind: line 5: /lib/functions.sh: No such file or directory
./etc/init.d/arpbind: line 6: /lib/functions/network.sh: No such file or directory
Enabling arpbind
Enabling autoreboot
Enabling boot
Enabling bootcount
Enabling clash
Enabling cron
Enabling ddns
Enabling dnsmasq
Enabling done
Enabling dropbear
Enabling etherwake
Enabling firewall
chmod: cannot access '/sbin/mtkwifi': No such file or directory
Enabling firmware.sh
Enabling flowoffload
Enabling fstab
Enabling gpio_switch
mkdir: cannot create directory '/etc/config': Permission denied
./etc/init.d/guest-wifi: line 10: uci: command not found
./etc/init.d/guest-wifi: line 11: uci: command not found
./etc/init.d/guest-wifi: line 12: uci: command not found
./etc/init.d/guest-wifi: line 13: uci: command not found
./etc/init.d/guest-wifi: line 14: uci: command not found
./etc/init.d/guest-wifi: line 15: uci: command not found
./etc/init.d/guest-wifi: line 16: uci: command not found
./etc/init.d/guest-wifi: line 17: uci: command not found
./etc/init.d/guest-wifi: line 18: uci: command not found
./etc/init.d/guest-wifi: line 19: uci: command not found
./etc/init.d/guest-wifi: line 20: uci: command not found
./etc/init.d/guest-wifi: line 21: uci: command not found
Enabling guest-wifi
Enabling haproxy
Enabling jd-dailybonus
Enabling kms
Enabling led
Enabling log
Enabling miniupnpd
Enabling mwan3
Enabling mwan3helper
Enabling network
./etc/init.d/nginx: line 12: /usr/bin/nginx-util: No such file or directory
Enabling nginx
Enabling nlbwmon
./etc/init.d/openclash: line 3: /usr/share/openclash/openclash_ps.sh: No such file or directory
./etc/init.d/openclash: line 4: /usr/share/openclash/ruby.sh: No such file or directory
./etc/init.d/openclash: line 13: uci: command not found
./etc/init.d/openclash: line 14: uci: command not found
./etc/init.d/openclash: line 17: uci: command not found
Enabling openclash
Enabling passwall
Enabling passwall_server
Enabling rpcd
Enabling set-irq-affinity
Enabling shadowsocksr
Enabling sysctl
Enabling sysfixtime
Enabling sysntpd
Enabling system
Enabling uhttpd
Enabling umount
Enabling unblockmusic
Enabling urandom_seed
Enabling urngd
Enabling uwsgi
Enabling vsftpd
Enabling wrtbwmon
Enabling xlnetacc
Enabling zerotier
sed -i "s/Installed-Time: .*/Installed-Time: 1616574061/" /home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/opkg/status
rm -rf /home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/tmp/*
rm -f /home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/opkg/lists/*
rm -f /home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/usr/lib/opkg/info/*.postinst*rm -f /home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/var/lock/*.lock
rm -rf /home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/boot
find /home/zerokirin/lede/build_dir/target-mipsel_24kc_musl/root-ramips/ -mindepth 1 -execdir touch -hcd "@1616574061" "{}" +
find: The relative path 'Files' is included in the PATH environment variable, which is insecure in combination with the -execdir action of find.  Please remove that entry from $PATH
package/Makefile:65: recipe for target 'package/install' failed
make[2]: *** [package/install] Error 1
make[2]: Leaving directory '/home/zerokirin/lede'
package/Makefile:108: recipe for target '/home/zerokirin/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install' failed
make[1]: *** [/home/zerokirin/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2make[1]: Leaving directory '/home/zerokirin/lede'
/home/zerokirin/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2
```

这里前面明明有openclash和guest-wifi的报错，但后面怎么又有enabling openclash，看不太懂，求大佬提点呐

EnnawYang, 2021-03-25:
aWSL

AndyLin2020, 2021-03-25:
搭梯子再编译

AmadeusGhost, 2021-03-25:
`find: The relative path 'Files' is included in the PATH environment variable, which is insecure in combination with the -execdir action of find.  Please remove that entry from $PATH`
看不懂就去百度这一行

00LT00, 2021-03-25:
懂了，谢谢大哥们，禁用windows的环境变量就ok了

closed 2021-03-25 by 00LT00

smdx023, 2021-04-23:
网上搜索到的禁用Windows变量教程
1、2在wsl运行，3在win的cmd运行

1、 sudo vim /etc/wsl.conf

2、 输入
# 不加载Windows中的PATH内容
[interop]
appendWindowsPath = false
# 不自动挂载Windows系统所有磁盘分区
[automount]
enabled = false

3、类似重启WSL
wsl --list
# 适用于 Linux 的 Windows 子系统:
# Ubuntu-18.04 (默认)
wsl --terminate Ubuntu-18.04