coolsnowwolf/lede #6599: 网络-接口状态信息 数据包显示负数

ostrichii, 2021-03-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
网络-接口状态信息 数据包显示负数
[image]

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
LAN和WAN接口的数据包显示 为负数
[image]

### (1) 具体问题

A：

### (2) 路由器型号和固件版本
路由型号：x86 3865u 
固件版本：Powered by LuCI Master (git-21.035.34289-13f96d1) / OpenWrt R21.2.1
A：

### (3) 详细日志

A：

Cathgao, 2021-03-27:
应该是数据类型溢出

coolsnowwolf, 2021-03-27:
感谢报告 bug，这个是 jsonc 的 INT64 超过上限了，换用双精度显示就好了。

已修复

[image]

ostrichii, 2021-03-27:
> 感谢报告 bug，这个是 jsonc 的 INT64 超过上限了，换用双精度显示就好了。
> 
> 已修复
> 
> [image]

多谢大雕，地球上最快的男人

closed 2021-06-17 by coolsnowwolf

ostrichii, 2021-11-12:
这个问题貌似又被带上来了