coolsnowwolf/lede #6603: make memu报错

SuperPau1, 2021-03-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
make memu报错
Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
### (1) 具体问题

A：feeds.conf.default去掉最后一行的注释，然后make memu就报错了，请问怎么解决

[image]
[image]

### (2) 路由器型号和固件版本

A：x64

### (3) 详细日志

A：

OleStore, 2021-03-26:
update& install
建议你看首页说明
https://github.com/coolsnowwolf/lede

SuperPau1, 2021-03-26:
> update& install
> 建议你看首页说明
> https://github.com/coolsnowwolf/lede

首页没有任何信息，还是说你指的是更新feeds？编译无数次了不可能漏掉这一步的

SuperPau1, 2021-03-26:
> update& install
> 建议你看首页说明
> https://github.com/coolsnowwolf/lede

为了避免误解，我更新一下，下载源代码后打开feeds.conf.default去掉helloword的注释，update和install feeds后make memu报错

1715173329, 2021-03-27:
什么叫报错？没日志让网友给你算命？

closed 2021-06-17 by coolsnowwolf