coolsnowwolf/lede #6607: WIFI 5G信号与USB 3.0冲突？

DINKHo, 2021-03-27:
### (1) 具体问题
最近编译新的Lean's Openwrt刷入后，手机常常找不到WIFI 5G信号，重新设置5G WIFI或重启路由器都没用，最后发现只要把USB硬盘拔掉，手机马上就连上WIFI 5G信号了（如下核心日志），请问是哪里有冲突了吗？

### (2) 路由器型号和固件版本
Newifi3-D2
OpenWrt R21.2.1
核心版本 5.4.105

### (3) 核心日志
[21560.296196] usb 2-1: USB disconnect, device number 2
[21560.302654] blk_update_request: I/O error, dev sda, sector 0 op 0x1:(WRITE) flags 0x800 phys_seg 0 prio class 0
[21560.359758] sd 0:0:0:0: [sda] Synchronizing SCSI cache
[21560.365296] sd 0:0:0:0: [sda] Synchronize Cache(10) failed: Result: hostbyte=0x01 driverbyte=0x00
[21572.322747] update_associated_mac_entry(): Peer's PhyCap=>Mode:VHT, BW:80M
[21572.536700] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(11-11-11-11-11-11) !!!
[21572.544522] Receive EAPoL-Key frame from STA 11-11-11-11-11-11 wcid(1)
[21572.555419] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(11-11-11-11-11-11) !!!
[21572.563177] Receive EAPoL-Key frame from STA 11-11-11-11-11-11 wcid(1)
[21572.569796] AP SETKEYS DONE - WPA2, AuthMode(7)=WPA2PSK, WepStatus(6)=AES, GroupWepStatus(4)=TKIP
[21572.569796]
[21572.712235] Rcv Wcid(1) AddBAReq
[21572.715468] Start Seq = 00000000
[21575.414342] update_associated_mac_entry(): Peer's PhyCap=>Mode:VHT, BW:80M
[21575.627538] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(22-22-22-22-22-22) !!!
[21575.635666] Receive EAPoL-Key frame from STA 22-22-22-22-22-22 wcid(2)
[21575.647286] WpaEAPOLKeyAction: CliIdx == 0xFF  pHeader->Addr2(22-22-22-22-22-22) !!!
[21575.655392] Receive EAPoL-Key frame from STA 22-22-22-22-22-22 wcid(2)
[21575.662232] AP SETKEYS DONE - WPA2, AuthMode(7)=WPA2PSK, WepStatus(6)=AES, GroupWepStatus(4)=TKIP
[21575.662232]
[21575.753303] Rcv Wcid(2) AddBAReq
[21575.756566] Start Seq = 00000000

minlang112, 2021-03-28:
屏蔽不良的USB3.0设备，对2.4G频段的WiFi有干扰，对5G频段干扰不大……

Cathgao, 2021-03-28:
U3干扰的是2.4G，跟5G一点关系没有

DINKHo, 2021-03-28:
我知道USB3.0跟2.4G WIFI会干扰，但USB3.0硬盘插上时手机可以正常使用2.4G WIFI，却搜寻不到5G WIFI信号，USB3.0硬盘拔掉，手机马上就连上WIFI 5G信号了

brightsm, 2021-03-30:
是不是USB3.0的硬盘供电不足压？

DINKHo, 2021-04-01:
> 是不是USB3.0的硬盘供电不足压？

我以前用旧版的Lean's Openwrt是5G WIFI信号跟USB3.0硬盘都正常的，所以我猜应该不是供电问题，现在是12V 2A，我也没有更大的电源可以测试了

closed 2021-06-17 by coolsnowwolf