coolsnowwolf/lede #6609: 低性能设备建议关闭 FULLCONE NAT

FH0, 2021-03-27:
```python
import socket
import secrets
import time

udp_socket = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
udp_socket.bind(("0.0.0.0", 0))

buf = secrets.token_bytes(1000)
while True:
    for i in range(1, 65536):
        udp_socket.sendto(buf, ("1.2.3.4", i))
        time.sleep(0.0005)
```

上面是一个用 python 写的 udp 压力测试脚本，可以通过调整 time.sleep 来调整每秒的发送量。
在 k2(7620) 设备上
- 开启 FULLCONE NAT，每秒 500 个 udp 包就能让路由器宕机
- 关闭 FULLCONE NAT，每秒 2000 个 udp 包，此时路由器的 CPU 剩余在 60% - 90% 波动，提升巨大

[image]

Neverends4, 2021-03-27:
所以说FULLCONE和启用硬件NAT/SFE导致的随机重启有没有联系？还是像MT762X这种设备导致重启的是不是开启FULLCONE造成的？

supppig, 2021-03-28:
还有这个问题？关了 fullcone试试看。。。

mjyhj, 2021-03-29:
> 不编译SFE/HWNAT然后开启FULLCONE看会不会随机重启
不开SFE，只开FULLCONE是不会重启的。
开FLOW和FULLCONE，也不会重启

mjyhj, 2021-03-29:
> 所以说FULLCONE和启用硬件NAT/SFE导致的随机重启有没有联系？还是像MT762X这种设备导致重启的是不是开启FULLCONE造成的？
> 决定自己试一下，不编译SFE/HWNAT然后开启FULLCONE看会不会随机重启。

不开SFE，只开FULLCONE是不会重启的。
开FLOW和FULLCONE，也不会重启

realconnor, 2021-03-30:
是的，其实不用这个脚本，打开比特彗星效果也是一样的。。。

realconnor, 2021-03-30:
> 是的，其实不用这个脚本，打开比特彗星效果也是一样的。。。

参见 https://github.com/coolsnowwolf/lede/issues/6612 和 https://github.com/coolsnowwolf/lede/issues/6427

TXH201622, 2021-03-30:
> > 不开SFE，只开FULLCONE是不会重启的。
> > 开FLOW和FULLCONE，也不会重启
> 
> 应该是架构不同，我的红米2100用SFE/FLOWOFFLOAD不管开不开FULLCONE都会重启，但是貌似不开FULLCONE能撑久一点？感觉从启动后到头一回随机重启可以撑好几天，但是只要随机重启第一次后再重启时间间隔就很短了，有时半天都不到。
> #6489
> 目前没编译SFE/FLOWOFFLOAD 开FULLCONE 5天还没重启，需要再观察，我觉得至少要20天不重启才合格，因为网络负荷并不大，只是偶尔有P2P和UPNP需求，局域网全部连接数99%的时间都低于1500/s。

K2P 官方master分支+闭源驱动，开flow offload，半个月没有重启。

closed 2021-06-17 by coolsnowwolf