coolsnowwolf/lede #6612: R4S在电脑开启比特彗星后会越来越卡，过几分钟时间恢复

llzzll, 2021-03-27:
不管是不是在下载，只要开着就会，3月24晚上编译的固件。
卡的期间连luci界面都进不去，ssh也进不去。
开启了SFE和fullconenat，ipv6helper，

以下是卡机前后的日志：
`Sat Mar 27 18:29:40 2021 daemon.notice miniupnpd[4057]: remove port mapping 34509 UDP because it has expired
Sat Mar 27 19:56:36 2021 kern.info kernel: [248699.710313] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 19:56:36 2021 kern.info kernel: [248699.712552] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 19:56:36 2021 kern.info kernel: [248699.713507] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 19:56:36 2021 kern.info kernel: [248699.714267] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 19:56:36 2021 kern.info kernel: [248699.715710] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 19:56:36 2021 kern.info kernel: [248699.716429] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 19:56:36 2021 kern.info kernel: [248699.717820] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 19:56:36 2021 kern.info kernel: [248699.719244] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 19:56:36 2021 kern.info kernel: [248699.720177] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 19:56:36 2021 kern.info kernel: [248699.721442] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 19:56:44 2021 kern.warn kernel: [248707.694547] net_ratelimit: 1095 callbacks suppressed
Sat Mar 27 19:56:44 2021 kern.info kernel: [248707.694550] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 19:56:44 2021 kern.info kernel: [248707.695824] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 19:56:44 2021 kern.info kernel: [248707.710672] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 19:56:44 2021 kern.info kernel: [248707.718060] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 19:56:44 2021 kern.info kernel: [248707.731923] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 19:56:44 2021 kern.info kernel: [248707.732693] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 19:56:44 2021 kern.info kernel: [248707.739228] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 19:56:44 2021 kern.info kernel: [248707.740418] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 19:56:44 2021 kern.info kernel: [248707.746374] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 19:56:44 2021 kern.info kernel: [248707.747711] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 20:26:38 2021 kern.warn kernel: [250501.464388] net_ratelimit: 18 callbacks suppressed
Sat Mar 27 20:26:38 2021 kern.info kernel: [250501.464391] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 20:26:38 2021 kern.info kernel: [250501.467222] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 20:26:38 2021 kern.info kernel: [250501.468076] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 20:26:38 2021 kern.info kernel: [250501.468783] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 20:26:38 2021 kern.info kernel: [250501.469730] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 20:26:38 2021 kern.info kernel: [250501.470486] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 20:26:38 2021 kern.info kernel: [250501.471259] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 20:26:38 2021 kern.info kernel: [250501.472066] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 20:26:38 2021 kern.info kernel: [250501.472939] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 20:26:38 2021 kern.info kernel: [250501.473744] neighbour: arp_cache: neighbor table overflow!
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.264632] ------------[ cut here ]------------
Sat Mar 27 20:27:42 2021 kern.info kernel: [250565.265078] NETDEV WATCHDOG: eth1 (r8168): transmit queue 0 timed out
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.265751] WARNING: CPU: 2 PID: 0 at net/sched/sch_generic.c:448 dev_watchdog+0x2f4/0x300
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.266485] Modules linked in: fast_classifier xt_FULLCONENAT pppoe ppp_async pppox ppp_mppe ppp_generic ipt_REJECT xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm slhc sch_cake r8168 nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan iptable_raw
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.266752]  iptable_nat iptable_mangle iptable_filter ipt_ECN ip_tables exfat crc_ccitt asn1_decoder tcp_bbr sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb sit tunnel4 ip_tunnel tun vfat fat nls_utf8 nls_cp437 shortcut_fe_ipv6 shortcut_fe crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha1_generic ecb authenc arc4 uas gpio_button_hotplug antfs
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.281784] CPU: 2 PID: 0 Comm: swapper/2 Not tainted 5.4.105 #0
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.282323] Hardware name: FriendlyElec NanoPi R4S (DT)
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.282799] pstate: 40000005 (nZcv daif -PAN -UAO)
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.283241] pc : dev_watchdog+0x2f4/0x300
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.283614] lr : dev_watchdog+0x2f4/0x300
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.283982] sp : ffff800010013dc0
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.284287] x29: ffff800010013dc0 x28: ffff0000f00c0c80
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.284774] x27: 0000000000000004 x26: 0000000000000140
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.285259] x25: 00000000ffffffff x24: 0000000000000002
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.285745] x23: ffff0000ef23845c x22: ffff0000ef238000
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.286231] x21: ffff0000ef238480 x20: ffff800010b76000
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.286717] x19: 0000000000000000 x18: 0000000000000000
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.287202] x17: 0000000000000000 x16: 0000000000000000
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.287686] x15: 0000000000000000 x14: ffff800010bfefc2
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.288172] x13: 0000000000000000 x12: ffff800010bfe000
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.288657] x11: ffff800010b89000 x10: 0000000000000010
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.289141] x9 : 0000000000000000 x8 : 2030206575657571
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.289626] x7 : 2074696d736e6172 x6 : 00000000000001a2
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.290110] x5 : 0000000000000001 x4 : ffff0000f6f91708
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.290596] x3 : 0000000000000006 x2 : 0000000000000007
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.291081] x1 : ffff0000f2230b80 x0 : 0000000000000039
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.291567] Call trace:
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.291808]  dev_watchdog+0x2f4/0x300
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.292152]  call_timer_fn.isra.35+0x20/0x78
Sat Mar 27 20:27:42 2021 kern.warn kernel: [250565.292547]  run_timer_softirq+0x378/0x388
[... cut ...]
Sat Mar 27 20:27:53 2021 kern.info kernel: [250576.800502] br-lan: port 1(eth1) entered blocking state
Sat Mar 27 20:27:53 2021 daemon.notice netifd: Network device 'eth1' link is up
Sat Mar 27 20:27:53 2021 daemon.notice netifd: bridge 'br-lan' link is up
Sat Mar 27 20:27:53 2021 daemon.notice netifd: Interface 'lan' has link connectivity
Sat Mar 27 20:27:53 2021 kern.info kernel: [250576.801012] br-lan: port 1(eth1) entered forwarding state
Sat Mar 27 20:29:36 2021 daemon.notice miniupnpd[4057]: remove port mapping 45241 UDP because it has expired
`

TXH201622, 2021-03-27:
两个方法
1.关闭nat1
2.将BT彗星的UDP发包数从1000改为100

coolsnowwolf, 2021-03-27:
比特彗星 的 Bug，不断的发送 ARP 更新表，把路由器 ARP Table 充满了

llzzll, 2021-03-27:
> 比特彗星 的 Bug，不断的发送 ARP 更新表，把路由器 ARP Table 充满了

那解决方案就是改udp发包数或者换fullcone吗？还有就是请教一下为什么这么明显的bug在其他路由器上没有见过呢？

Neverends4, 2021-03-28:
> > 比特彗星 的 Bug，不断的发送 ARP 更新表，把路由器 ARP Table 充满了
> 
> 那解决方案就是改udp发包数或者换fullcone吗？还有就是请教一下为什么这么明显的bug在其他路由器上没有见过呢？

貌似大部分路由器默认都没有FULLCONE NAT，尤其是三剑客。

jxph1123, 2021-03-31:
将BT彗星的UDP发包数从1000改为100就可以了

closed 2021-06-17 by coolsnowwolf