coolsnowwolf/lede #6613: 时间显示与人类认知习惯有冲突：0h 60m 36s

xxp2277, 2021-03-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
### (1) 具体问题

A：luci界面：网络，接口中，显示的wan口的运行时间进制有点问题，显示为“0h 60m 36s”，我认为应该是“1h 0m 36s”才符合人类的认知习惯。
其它接口由于未在合适的时间截图，不清楚是否有相同问题。

### (2) 路由器型号和固件版本

A：wndr3800

### (3) 详细日志
A：[image]

lphgor, 2021-03-27:
应该是这里的问题吧
https://github.com/coolsnowwolf/luci/blob/50d5286ab365f4c1c6646cb976c6be27a57cf6d4/modules/luci-base/htdocs/luci-static/resources/cbi.js#L1453-L1476

lphgor, 2021-03-29:
可以复现，页面按F12进控制台，依次输入这几行：
```javascript
String.format('%t', 60);
String.format('%t', 3600);
String.format('%t', 86400);
```
这里的问题吧，判断的时候应该`ts > 59`、`tm > 59`、`th > 23`
https://github.com/coolsnowwolf/luci/blob/3fe64081aa5c14895d25dbadf7484dde61ed7a1d/modules/luci-base/htdocs/luci-static/resources/cbi.js#L1453-L1478

xxp2277, 2021-03-29:
> 
> 
> 可以复现，页面按F12进控制台，依次输入这几行：
> 
> ```js
> String.format('%t', 60);
> String.format('%t', 3600); //3600 to 3659
> String.format('%t', 86400); //86400 to 89999
> ```
> 
> 这里的问题吧，判断的时候应该`ts > 59`、`tm > 59`、`th > 23`
> https://github.com/coolsnowwolf/luci/blob/3fe64081aa5c14895d25dbadf7484dde61ed7a1d/modules/luci-base/htdocs/luci-static/resources/cbi.js#L1453-L1478
> 
> openwrt 官方的代码也是这样的。
大神有能力就提个PR改了吧！

xxp2277, 2021-04-01:
辛苦了。

closed 2021-04-01 by xxp2277