coolsnowwolf/lede #6615: overture 1.7编译失败，1.6.1版可以正常编译

NagaseKouichi, 2021-03-28:
L大的feed中带的overture是1.6.1版的，如果把overture的Makefile文件中的版本改成1.7版，PKG_HASH也相应修改之后，编译会报错：

Building targets
../../core/inbound/server.go:18:2: missing go.sum entry for module providing package github.com/coredns/coredns/plugin/pkg/dnsutil (imported by github.com/shawn1m/overture/core/inbound); to add:
        go get github.com/shawn1m/overture/core/inbound
../../core/inbound/server.go:19:2: missing go.sum entry for module providing package github.com/coredns/coredns/plugin/pkg/doh (imported by github.com/shawn1m/overture/core/inbound); to add:
        go get github.com/shawn1m/overture/core/inbound
../../core/inbound/server.go:20:2: missing go.sum entry for module providing package github.com/coredns/coredns/plugin/pkg/response (imported by github.com/shawn1m/overture/core/inbound); to add:
        go get github.com/shawn1m/overture/core/inbound

make[2]: *** [Makefile:61: /home/suhao/lede/build_dir/target-x86_64_musl/overture-1.7/.built] Error 1
make[2]: Leaving directory '/home/suhao/lede/package/overture'
time: package/overture/compile#0.61#0.27#0.88
make[1]: *** [package/Makefile:113: package/overture/compile] Error 2
make[1]: Leaving directory '/home/suhao/lede'
make: *** [/home/suhao/lede/include/toplevel.mk:224: package/overture/compile] Error 2

请问是golang版本的问题吗？我看了下，L大的feed中带的golang是1.16的。而用openwrt官方的sdk编译overture 1.7版可以正常编译，openwrt官方的sdk带的golang是1.13版的，版本号还要低一些。

NagaseKouichi, 2021-03-28:
在go项目也搜到类似的反馈，貌似在go升级到1.16之后有些go软件编译的时候就会报这个错误。
https://github.com/golang/go/issues/44129

AmadeusGhost, 2021-03-31:
```
--- a/go.sum
+++ b/go.sum
@@ -91,6 +91,7 @@ github.com/cncf/udpa/go v0.0.0-20191209042840-269d4d468f6f/go.mod h1:M8M6+tZqaGX
 github.com/cockroachdb/datadriven v0.0.0-20190809214429-80d97fb3cbaa/go.mod h1:zn76sxSg3SzpJ0PPJaLDCu+Bu0Lg3sKTORVIj19EIF8=
 github.com/codahale/hdrhistogram v0.0.0-20161010025455-3a0bb77429bd/go.mod h1:sE/e/2PUdi/liOCUjSTXgM1o87ZssimdTWN964YiIeI=
 github.com/coredns/caddy v1.1.0/go.mod h1:A6ntJQlAWuQfFlsd9hvigKbo2WS0VUs2l1e2F+BawD4=
+github.com/coredns/coredns v1.8.0 h1:U6Z6Ljuy5D8VKDquF1dUotUtFeoNt0PVCwSfIVYFC6s=
 github.com/coredns/coredns v1.8.0/go.mod h1:R/TQ2yPVKdafwpVXOh6NFN13ikgto21kOMPvOeFPS8o=
 github.com/coreos/go-semver v0.2.0/go.mod h1:nnelYz7RCh+5ahJtPPxZlU+153eP4D4r3EedlOD2RNk=
 github.com/coreos/go-systemd v0.0.0-20180511133405-39ca1b05acc7/go.mod h1:F5haX7vjVVG0kc13fIWeqUViNPyEJxv/OmvnBo0Yme4=
@@ -277,6 +278,7 @@ github.com/kr/pretty v0.1.0/go.mod h1:dAy3ld7l9f0ibDNOQOHHMYYIIbhfbHSm3C4ZsoJORN
 github.com/kr/pretty v0.2.0 h1:s5hAObm+yFO5uHYt5dYjxi2rXrsnmRpJx4OYvIWUaQs=
 github.com/kr/pretty v0.2.0/go.mod h1:ipq/a2n7PKx3OHsz4KJII5eveXtPO4qwEXGdVfWzfnI=
 github.com/kr/pty v1.1.1/go.mod h1:pFQYn66WHrOpPYNljwOMqo10TkYh1fy3cYio2l3bCsQ=
+github.com/kr/text v0.1.0 h1:45sCR5RtlFHMR4UwH9sdQ5TC8v0qDQCHnXt+kaKSTVE=
 github.com/kr/text v0.1.0/go.mod h1:4Jbv+DJW3UT/LiOwJeYQe1efqtUx/iVham/4vfdArNI=
 github.com/lightstep/lightstep-tracer-common/golang/gogo v0.0.0-20190605223551-bc2310a04743/go.mod h1:qklhhLq1aX+mtWk9cPHPzaBjWImj5ULL6C7HFJtXQMM=
 github.com/lightstep/lightstep-tracer-go v0.18.1/go.mod h1:jlF1pusYV4pidLvZ+XD0UBX0ZE6WURAspgAczcDHrL4=
@@ -356,6 +358,7 @@ github.com/pkg/errors v0.8.0/go.mod h1:bwawxfHBFNV+L2hUp1rHADufV3IMtnDRdf1r5NINE
 github.com/pkg/errors v0.8.1/go.mod h1:bwawxfHBFNV+L2hUp1rHADufV3IMtnDRdf1r5NINEl0=
 github.com/pkg/errors v0.9.1/go.mod h1:bwawxfHBFNV+L2hUp1rHADufV3IMtnDRdf1r5NINEl0=
 github.com/pkg/profile v1.2.1/go.mod h1:hJw3o1OdXxsrSjjVksARp5W95eeEaEfptyVZyv6JUPA=
+github.com/pmezard/go-difflib v1.0.0 h1:4DBwDE0NGyQoBHbLQYPwSUPoCMWR5BEzIk/f1lZbAQM=
 github.com/pmezard/go-difflib v1.0.0/go.mod h1:iKH77koFhYxTK1pcRnkKkqfTogsbg7gZNVY4sRDYZ/4=
 github.com/posener/complete v1.1.1/go.mod h1:em0nMJCgc9GFtwrmVmEMR/ZL6WyhyjMBndrE9hABlRI=
 github.com/prometheus/client_golang v0.9.1/go.mod h1:7SWBe2y4D6OKWSNQJUaRYU/AaXPKyh/dDVn+NZz0KFw=
@@ -717,6 +720,7 @@ gopkg.in/fsnotify.v1 v1.4.7/go.mod h1:Tz8NjZHkW78fSQdbUxIjBTcgA1z1m8ZHf0WmKUhAMy
 gopkg.in/gcfg.v1 v1.2.3/go.mod h1:yesOnuUOFQAhST5vPY4nbZsb/huCgGGXlipJsBn0b3o=
 gopkg.in/inf.v0 v0.9.1/go.mod h1:cWUDdTG/fYaXco+Dcufb5Vnc6Gp2YChqWtbxRZE0mXw=
 gopkg.in/resty.v1 v1.12.0/go.mod h1:mDo4pnntr5jdWRML875a/NmxYqAlA73dVijT2AXvQQo=
+gopkg.in/tomb.v1 v1.0.0-20141024135613-dd632973f1e7 h1:uRGJdciOHaEIrze2W8Q3AKkepLTh2hOroT7a+7czfdQ=
 gopkg.in/tomb.v1 v1.0.0-20141024135613-dd632973f1e7/go.mod h1:dt/ZhP58zS4L8KSrWDmTeBkI65Dw0HsyUHuEVlX15mw=
 gopkg.in/warnings.v0 v0.1.2/go.mod h1:jksf8JmL6Qr/oQM2OXTHunEvvTAsrWBLb6OOjuVWRNI=
 gopkg.in/yaml.v2 v2.0.0-20170812160011-eb3733d160e7/go.mod h1:JAlM8MvJe8wmxCU4Bli9HhUf9+ttbYbLASfIpnQbh74=
```
打个 patch 应该就好了

NagaseKouichi, 2021-03-31:
在L大的openwrt编译成功了，问题解决。

closed 2021-03-31 by NagaseKouichi