coolsnowwolf/lede #6623: 最新5.4源码ipsec手机连接失败

zyh6111, 2021-03-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：Tue Mar 30 15:36:37 2021 authpriv.info ipsec_starter[25777]: Starting strongSwan 5.8.4 IPsec [starter]...
Tue Mar 30 15:36:37 2021 daemon.info : 00[DMN] Starting IKE charon daemon (strongSwan 5.8.4, Linux 5.4.106, x86_64)
Tue Mar 30 15:36:37 2021 daemon.info : 00[CFG] loading ca certificates from '/etc/ipsec.d/cacerts'
Tue Mar 30 15:36:37 2021 daemon.info : 00[CFG] loading aa certificates from '/etc/ipsec.d/aacerts'
Tue Mar 30 15:36:37 2021 daemon.info : 00[CFG] loading ocsp signer certificates from '/etc/ipsec.d/ocspcerts'
Tue Mar 30 15:36:37 2021 daemon.info : 00[CFG] loading attribute certificates from '/etc/ipsec.d/acerts'
Tue Mar 30 15:36:37 2021 daemon.info : 00[CFG] loading crls from '/etc/ipsec.d/crls'
Tue Mar 30 15:36:37 2021 daemon.info : 00[CFG] loading secrets from '/etc/ipsec.secrets'
Tue Mar 30 15:36:37 2021 daemon.info : 00[CFG]   loaded IKE secret for %any
Tue Mar 30 15:36:37 2021 daemon.info : 00[CFG]   loaded EAP secret for 1
Tue Mar 30 15:36:37 2021 daemon.info : 00[LIB] loaded plugins: charon aes sha1 random nonce x509 pubkey gmp xcbc hmac kernel-netlink socket-default stroke updown xauth-generic
Tue Mar 30 15:36:37 2021 daemon.info : 00[JOB] spawning 16 worker threads
Tue Mar 30 15:36:37 2021 authpriv.info ipsec_starter[25777]: charon (25781) started after 20 ms
Tue Mar 30 15:36:37 2021 daemon.info : 07[CFG] received stroke: add connection 'xauth_psk'
Tue Mar 30 15:36:37 2021 daemon.info : 07[CFG] adding virtual IP address pool 192.168.0.200/24
Tue Mar 30 15:36:37 2021 daemon.info : 07[CFG] added configuration 'xauth_psk'
Tue Mar 30 15:37:11 2021 daemon.info : 04[NET] received packet: from 223.104.189.118[45510] to 182.35.239.87[500] (444 bytes)
Tue Mar 30 15:37:11 2021 daemon.info : 04[ENC] parsed ID_PROT request 0 [ SA V V V V V V ]
Tue Mar 30 15:37:11 2021 daemon.info : 04[IKE] received XAuth vendor ID
Tue Mar 30 15:37:11 2021 daemon.info : 04[IKE] received DPD vendor ID
Tue Mar 30 15:37:11 2021 daemon.info : 04[IKE] received Cisco Unity vendor ID
Tue Mar 30 15:37:11 2021 daemon.info : 04[IKE] received FRAGMENTATION vendor ID
Tue Mar 30 15:37:11 2021 daemon.info : 04[IKE] received NAT-T (RFC 3947) vendor ID
Tue Mar 30 15:37:11 2021 daemon.info : 04[IKE] received draft-ietf-ipsec-nat-t-ike-02\n vendor ID
Tue Mar 30 15:37:11 2021 daemon.info : 04[IKE] 223.104.189.118 is initiating a Main Mode IKE_SA
Tue Mar 30 15:37:11 2021 authpriv.info : 04[IKE] 223.104.189.118 is initiating a Main Mode IKE_SA
Tue Mar 30 15:37:11 2021 daemon.info : 04[CFG] selected proposal: IKE:AES_CBC_128/HMAC_SHA1_96/PRF_HMAC_SHA1/MODP_1024
Tue Mar 30 15:37:11 2021 daemon.info : 04[ENC] generating ID_PROT response 0 [ SA V V V V ]
Tue Mar 30 15:37:11 2021 daemon.info : 04[NET] sending packet: from 182.35.239.87[500] to 223.104.189.118[45510] (160 bytes)
Tue Mar 30 15:37:12 2021 daemon.info : 06[NET] received packet: from 223.104.189.118[45510] to 182.35.239.87[500] (244 bytes)
Tue Mar 30 15:37:12 2021 daemon.info : 06[ENC] parsed ID_PROT request 0 [ KE No NAT-D NAT-D ]
Tue Mar 30 15:37:12 2021 daemon.info : 06[IKE] remote host is behind NAT
Tue Mar 30 15:37:12 2021 daemon.info : 06[ENC] generating ID_PROT response 0 [ KE No NAT-D NAT-D ]
Tue Mar 30 15:37:12 2021 daemon.info : 06[NET] sending packet: from 182.35.239.87[500] to 223.104.189.118[45510] (244 bytes)
Tue Mar 30 15:37:12 2021 daemon.info : 11[NET] received packet: from 223.104.189.118[55489] to 182.35.239.87[4500] (76 bytes)
Tue Mar 30 15:37:12 2021 daemon.info : 11[ENC] parsed ID_PROT request 0 [ ID HASH ]
Tue Mar 30 15:37:12 2021 daemon.info : 11[CFG] looking for XAuthInitPSK peer configs matching 182.35.239.87...223.104.189.118[10.94.134.183]
Tue Mar 30 15:37:12 2021 daemon.info : 11[CFG] selected peer config "xauth_psk"
Tue Mar 30 15:37:12 2021 daemon.info : 11[ENC] generating ID_PROT response 0 [ ID HASH ]
Tue Mar 30 15:37:12 2021 daemon.info : 11[NET] sending packet: from 182.35.239.87[4500] to 223.104.189.118[55489] (76 bytes)
Tue Mar 30 15:37:12 2021 daemon.info : 11[ENC] generating TRANSACTION request 1053627833 [ HASH CPRQ(X_USER X_PWD) ]
Tue Mar 30 15:37:12 2021 daemon.info : 11[NET] sending packet: from 182.35.239.87[4500] to 223.104.189.118[55489] (76 bytes)
Tue Mar 30 15:37:12 2021 daemon.info : 08[NET] received packet: from 223.104.189.118[55489] to 182.35.239.87[4500] (92 bytes)
Tue Mar 30 15:37:12 2021 daemon.info : 08[ENC] parsed TRANSACTION response 1053627833 [ HASH CPRP(X_USER X_PWD) ]
Tue Mar 30 15:37:12 2021 daemon.info : 08[IKE] XAuth authentication of '1' successful
Tue Mar 30 15:37:12 2021 daemon.info : 08[ENC] generating TRANSACTION request 1430421060 [ HASH CPS(X_STATUS) ]
Tue Mar 30 15:37:12 2021 daemon.info : 08[NET] sending packet: from 182.35.239.87[4500] to 223.104.189.118[55489] (76 bytes)
Tue Mar 30 15:37:12 2021 daemon.info : 12[NET] received packet: from 223.104.189.118[55489] to 182.35.239.87[4500] (76 bytes)
Tue Mar 30 15:37:12 2021 daemon.info : 12[ENC] parsed TRANSACTION response 1430421060 [ HASH CPA(X_STATUS) ]
Tue Mar 30 15:37:12 2021 daemon.info : 12[IKE] IKE_SA xauth_psk[1] established between 182.35.239.87[182.35.239.87]...223.104.189.118[10.94.134.183]
Tue Mar 30 15:37:12 2021 authpriv.info : 12[IKE] IKE_SA xauth_psk[1] established between 182.35.239.87[182.35.239.87]...223.104.189.118[10.94.134.183]
Tue Mar 30 15:37:12 2021 daemon.info : 12[IKE] scheduling reauthentication in 10171s
Tue Mar 30 15:37:12 2021 daemon.info : 12[IKE] maximum IKE_SA lifetime 10711s
Tue Mar 30 15:37:12 2021 daemon.info : 10[NET] received packet: from 223.104.189.118[55489] to 182.35.239.87[4500] (76 bytes)
Tue Mar 30 15:37:12 2021 daemon.info : 10[ENC] parsed TRANSACTION request 807592675 [ HASH CPRQ(ADDR DNS) ]
Tue Mar 30 15:37:12 2021 daemon.info : 10[IKE] peer requested virtual IP %any
Tue Mar 30 15:37:12 2021 daemon.info : 10[CFG] assigning new lease to '1'
Tue Mar 30 15:37:12 2021 daemon.info : 10[IKE] assigning virtual IP 192.168.0.200 to peer '1'
Tue Mar 30 15:37:12 2021 daemon.info : 10[ENC] generating TRANSACTION response 807592675 [ HASH CPRP(ADDR DNS) ]
Tue Mar 30 15:37:12 2021 daemon.info : 10[NET] sending packet: from 182.35.239.87[4500] to 223.104.189.118[55489] (92 bytes)
Tue Mar 30 15:37:12 2021 daemon.info : 16[NET] received packet: from 223.104.189.118[55489] to 182.35.239.87[4500] (412 bytes)
Tue Mar 30 15:37:12 2021 daemon.info : 16[ENC] parsed QUICK_MODE request 902374648 [ HASH SA No ID ID ]
Tue Mar 30 15:37:12 2021 daemon.info : 16[CFG] selected proposal: ESP:AES_CBC_128/HMAC_SHA1_96/NO_EXT_SEQ
Tue Mar 30 15:37:12 2021 daemon.info : 16[IKE] received 10800s lifetime, configured 3600s
Tue Mar 30 15:37:12 2021 daemon.info : 16[ENC] generating QUICK_MODE response 902374648 [ HASH SA No ID ID ]
Tue Mar 30 15:37:12 2021 daemon.info : 16[NET] sending packet: from 182.35.239.87[4500] to 223.104.189.118[55489] (172 bytes)
Tue Mar 30 15:37:12 2021 daemon.info : 15[NET] received packet: from 223.104.189.118[55489] to 182.35.239.87[4500] (60 bytes)
Tue Mar 30 15:37:12 2021 daemon.info : 15[ENC] parsed QUICK_MODE request 902374648 [ HASH ]
Tue Mar 30 15:37:12 2021 daemon.info : 15[KNL] received netlink error: No such file or directory (2)
Tue Mar 30 15:37:12 2021 daemon.info : 15[KNL] unable to add SAD entry with SPI cb7b566f (FAILED)
Tue Mar 30 15:37:12 2021 daemon.info : 15[KNL] received netlink error: No such file or directory (2)
Tue Mar 30 15:37:12 2021 daemon.info : 15[KNL] unable to add SAD entry with SPI cc98fbf6 (FAILED)
Tue Mar 30 15:37:12 2021 daemon.info : 15[IKE] unable to install inbound and outbound IPsec SA (SAD) in kernel
Tue Mar 30 15:37:12 2021 daemon.info : 15[KNL] deleting policy 192.168.0.200/32 === 0.0.0.0/0 in failed, not found
Tue Mar 30 15:37:12 2021 daemon.info : 15[KNL] deleting policy 192.168.0.200/32 === 0.0.0.0/0 fwd failed, not found
Tue Mar 30 15:37:12 2021 daemon.info : 15[IKE] sending DELETE for ESP CHILD_SA with SPI cc98fbf6
Tue Mar 30 15:37:12 2021 daemon.info : 15[ENC] generating INFORMATIONAL_V1 request 116582077 [ HASH D ]
Tue Mar 30 15:37:12 2021 daemon.info : 15[NET] sending packet: from 182.35.239.87[4500] to 223.104.189.118[55489] (76 bytes)
Tue Mar 30 15:37:12 2021 daemon.info : 03[NET] received packet: from 223.104.189.118[55489] to 182.35.239.87[4500] (92 bytes)
Tue Mar 30 15:37:12 2021 daemon.info : 03[ENC] parsed INFORMATIONAL_V1 request 2539167535 [ HASH D ]
Tue Mar 30 15:37:12 2021 daemon.info : 03[IKE] received DELETE for IKE_SA xauth_psk[1]
Tue Mar 30 15:37:12 2021 daemon.info : 03[IKE] deleting IKE_SA xauth_psk[1] between 182.35.239.87[182.35.239.87]...223.104.189.118[10.94.134.183]
Tue Mar 30 15:37:12 2021 authpriv.info : 03[IKE] deleting IKE_SA xauth_psk[1] between 182.35.239.87[182.35.239.87]...223.104.189.118[10.94.134.183]
Tue Mar 30 15:37:12 2021 daemon.info : 03[CFG] lease 192.168.0.200 by '1' went offline

hv54, 2021-05-17:
我的也是一样

blackshot-zard, 2021-05-30:
我的是可以连接，也不断连，就是没有网络

blackshot-zard, 2021-05-30:
> 我的是可以连接，也不断连，就是没有网络
内外网都没有，好像只发送心跳包，确认是连接的。。

closed 2021-06-17 by coolsnowwolf