coolsnowwolf/lede #6628: Phicomm PSG1218 Ax  (K2) 重启或断电, 丢失配置, 自动重置, 恢复出厂设置

dengxin6, 2021-04-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
在breed里直接刷openwrt-ramips-mt7620-phicomm_psg1218a-initramfs-kernel 没有这个问题 ;
更新openwrt-ramips-mt7620-phicomm_psg1218a-squashfs-sysupgrade之后, 重启或断电, 丢失配置

### (2) 路由器型号和固件版本

PSG1218 Ax  
R21.3.27

### (3) 详细日志

Wed Mar 31 12:32:02 2021 kern.notice kernel: [    0.000000] Linux version 5.4.108 (cnmj@localhost.localdomain) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r2829-e68d170)) #0 Wed Mar 31 04:31:52 2021
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] Board has DDR2
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] Analog PMU set to hw control
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] Digital PMU set to hw control
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7620A ver:2 eco:6
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] CPU0 revision is: 00019650 (MIPS 24KEc)
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] MIPS: machine is Phicomm PSG1218 rev.A
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Wed Mar 31 12:32:02 2021 kern.warn kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Wed Mar 31 12:32:02 2021 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] Zone ranges:
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] Movable zone start for each node
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] Early memory node ranges
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
Wed Mar 31 12:32:02 2021 kern.debug kernel: [    0.000000] On node 0 totalpages: 16384
Wed Mar 31 12:32:02 2021 kern.debug kernel: [    0.000000]   Normal zone: 144 pages used for memmap
Wed Mar 31 12:32:02 2021 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Wed Mar 31 12:32:02 2021 kern.debug kernel: [    0.000000]   Normal zone: 16384 pages, LIFO batch:3
Wed Mar 31 12:32:02 2021 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Wed Mar 31 12:32:02 2021 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
Wed Mar 31 12:32:02 2021 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] Writing ErrCtl register=0007e007
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] Readback ErrCtl register=0007e007
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] Memory: 57116K/65536K available (4973K kernel code, 208K rwdata, 1100K rodata, 1196K init, 209K bss, 8420K reserved, 0K cma-reserved)
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] NR_IRQS: 256
Wed Mar 31 12:32:02 2021 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x308/0x4f4 with crng_init=0
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] CPU Clock: 620MHz
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] clocksource: systick: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 583261500 ns
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] systick: enable autosleep mode
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] systick: running - mult: 214748, shift: 32
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6165356278 ns
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.000010] sched_clock: 32 bits at 310MHz, resolution 3ns, wraps every 6927366142ns
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.015464] Calibrating delay loop... 410.62 BogoMIPS (lpj=821248)
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.055559] pid_max: default: 32768 minimum: 301
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.064923] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.079329] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.100885] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.120190] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.133773] pinctrl core: initialized pinctrl subsystem
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.144653] NET: Registered protocol family 16
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.416077] PCI host bridge /pcie@10140000 ranges:
Wed Mar 31 12:32:02 2021 kern.info kernel: [    0.425474]  MEM 0x0000000020000000..0x000000002fffffff
[... 247 lines cut ...]
Wed Mar 31 12:33:00 2021 kern.info kernel: [   81.171465] br-lan: port 3(wlan0) entered blocking state
Wed Mar 31 12:33:00 2021 kern.info kernel: [   81.182100] br-lan: port 3(wlan0) entered forwarding state
Wed Mar 31 12:33:00 2021 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Wed Mar 31 12:33:00 2021 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->HT_SCAN
Wed Mar 31 12:33:00 2021 daemon.notice hostapd: wlan1: interface state COUNTRY_UPDATE->ENABLED
Wed Mar 31 12:33:00 2021 daemon.notice hostapd: wlan1: AP-ENABLED
Wed Mar 31 12:33:01 2021 kern.info kernel: [   81.436480] br-lan: port 3(wlan0) entered disabled state
Wed Mar 31 12:33:01 2021 daemon.notice procd: /etc/rc.d/S90kms: KMS Server has started.
Wed Mar 31 12:33:01 2021 daemon.notice procd: /etc/rc.d/S90kms: uci: Entry not found
Wed Mar 31 12:33:01 2021 kern.info kernel: [   82.321813] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Wed Mar 31 12:33:01 2021 kern.info kernel: [   82.334829] br-lan: port 3(wlan0) entered blocking state
Wed Mar 31 12:33:01 2021 kern.info kernel: [   82.345476] br-lan: port 3(wlan0) entered forwarding state
Wed Mar 31 12:33:01 2021 daemon.notice hostapd: wlan0: interface state HT_SCAN->ENABLED
Wed Mar 31 12:33:01 2021 daemon.notice hostapd: wlan0: AP-ENABLED
Wed Mar 31 12:33:02 2021 daemon.notice netifd: Network device 'wlan1' link is up
Wed Mar 31 12:33:02 2021 daemon.notice netifd: Network device 'wlan0' link is up
Wed Mar 31 12:33:03 2021 daemon.notice procd: /etc/rc.d/S90kms: udhcpc: started, v1.31.1
Wed Mar 31 12:33:03 2021 daemon.notice procd: /etc/rc.d/S90kms: udhcpc: sending discover
Wed Mar 31 12:33:05 2021 daemon.notice procd: /etc/rc.d/S90kms: udhcpc: no lease, failing
Wed Mar 31 12:33:06 2021 daemon.info dnsmasq[3177]: Connected to system UBus
Wed Mar 31 12:33:07 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Mar 31 12:33:07 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Mar 31 12:33:10 2021 daemon.info dnsmasq[3394]: Connected to system UBus
Wed Mar 31 12:33:11 2021 daemon.info procd: - init complete -
Wed Mar 31 12:33:12 2021 daemon.info urandom_seed[3519]: Seed saved (/etc/urandom.seed)

dengxin6, 2021-04-01:
是不是固件太大, 把空间沾满了?

EnnawYang, 2021-04-01:
8MB Flash

Neverends4, 2021-04-01:
df -h 看看空间？

dengxin6, 2021-04-02:
就是默认配置固件太大的问题,  可以关闭了

AmadeusGhost, 2021-04-03:
你可以自行关闭 issue

closed 2021-04-03 by dengxin6