coolsnowwolf/lede #6629: 树莓派刷系统后使用不了设备本身的蓝牙

debugpro, 2021-04-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：刷系统后使用不了设备本身的蓝牙

### (2) 路由器型号和固件版本

A：树莓派4B + 20210331的代码

### (3) 详细日志

A：
root@OpenWrt:/lib/firmware/brcm# hciattach /dev/ttyAMA0 bcm43xx 115200 noflow -b
cm43xx_init
bcm43xx_init
Initialization timed out.
root@OpenWrt:/lib/firmware/brcm# hcitool dev
Devices:
root@OpenWrt:/lib/firmware/brcm# bluetoothctl
Agent registered
[bluetooth]# list
[bluetooth]# devices
No default controller available
[bluetooth]#

EnnawYang, 2021-04-01:
op要蓝牙做什么？

debugpro, 2021-04-01:
开发的产品需要用到蓝牙，外接USB蓝牙倒是可以，设备本身的用不了

debugpro, 2021-04-01:
> op要蓝牙做什么？
开发的产品需要用到蓝牙，外接USB蓝牙倒是可以，设备本身的用不了

AmadeusGhost, 2021-04-01:
openwrt 默认禁用掉蓝牙改成串口输出了，就当不支持好了。

debugpro, 2021-04-01:
> openwrt 默认禁用掉蓝牙改成串口输出了，就当不支持好了。

能启用吗？怎么改代码呢

closed 2021-06-17 by coolsnowwolf