coolsnowwolf/lede #6631: 闭源驱动如何使用8021X

wysizhh, 2021-04-02:
开源驱动可以安装WPAD和FreeRADIUS然后设置一下就行。闭源驱动不能用WPAD，按照开源驱动时设置FreeRADIUS后连接不成功。小白请教要怎么安装设置才能用

Metric-Void, 2021-04-05:
安装luci-app-mtwifi，在它提供的无线界面里面配置闭源驱动

wysizhh, 2021-04-06:
> 安装luci-app-mtwifi，在它提供的无线界面里面配置闭源驱动

除了下图设置还有其他设置吗?   之后证书要放什么地方?
[image]

Metric-Void, 2021-04-06:
> > 安装luci-app-mtwifi，在它提供的无线界面里面配置闭源驱动
> 
> 除了下图设置还有其他设置吗? 之后证书要放什么地方?
> [image]

有的，需要手动编辑配置文件，下面有个文本框那就是

以及，什么协议的证书？

wysizhh, 2021-04-07:
> > > 安装luci-app-mtwifi，在它提供的无线界面里面配置闭源驱动
> > 
> > 
> > 除了下图设置还有其他设置吗? 之后证书要放什么地方?
> > [image]
> 
> 有的，需要手动编辑配置文件，下面有个文本框那就是
> 
> 以及，什么协议的证书？

例如我用PEAP认证,就要用到证书啊

closed 2021-06-17 by coolsnowwolf