coolsnowwolf/lede #6632: 云编译到forked-daapd报错

Ailina0728, 2021-04-02:
反馈bug/问题模板，提建议请删除

1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)

- [ ] 没有类似的issue

 2. 详细叙述

 (1) 具体问题

A：云编译到forked-daapd报错

 (2) 路由器型号和固件版本

A：arm

 (3) 详细日志

1. 

A：

OpenWrt-libtool: link: rm -fr  .libs/forked-daapd-sqlext.la .libs/forked-daapd-sqlext.lai .libs/forked-daapd-sqlext.so
OpenWrt-libtool: link: aarch64-openwrt-linux-musl-gcc -shared  -fPIC -DPIC  .libs/sqlext.o   -L/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/lib -L/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/lib -L/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/lib -L/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libiconv-stub/lib -L/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libintl-stub/lib -lsqlite3 -lunistring  -Os -mcpu=cortex-a53 -fstack-protector -Wl,-z -Wl,now -Wl,-z -Wl,relro -Wl,-rpath-link=/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libiconv-stub/lib -Wl,-rpath-link=/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/libintl-stub/lib -Wl,-rpath-link -Wl,/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/lib   -Wl,-soname -Wl,forked-daapd-sqlext.so -o .libs/forked-daapd-sqlext.so
OpenWrt-libtool: link: ( cd ".libs" && rm -f "forked-daapd-sqlext.la" && ln -s "../forked-daapd-sqlext.la" "forked-daapd-sqlext.la" )
make[6]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/forked-daapd-27.1/sqlext'
Making all in src
make[6]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/forked-daapd-27.1/src'
/usr/bin/antlr3 -Xconversiontimeout 30000  -fo . RSP.g
/usr/bin/antlr3: 4: exec: java: not found
make[6]: *** [Makefile:1058: RSPLexer.c] Error 127
make[6]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/forked-daapd-27.1/src'
make[5]: *** [Makefile:626: all-recursive] Error 1
make[4]: *** [Makefile:468: all] Error 2
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/forked-daapd-27.1'
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/forked-daapd-27.1'
make[3]: *** [Makefile:85: /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/forked-daapd-27.1/.built] Error 2
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/sound/forked-daapd'
time: package/feeds/packages/forked-daapd/compile#0.26#0.05#0.31
make[2]: *** [package/Makefile:113: package/feeds/packages/forked-daapd/compile] Error 2
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:107: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:224: world] Error 2
Error: Process completed with exit code 2.

AmadeusGhost, 2021-04-04:
`/usr/bin/antlr3: 4: exec: java: not found` java 依赖没安装

Ailina0728, 2021-04-05:
> `/usr/bin/antlr3: 4: exec: java: not found` java依赖没安装

之前用p3大佬脚本正常编译成功，后面再用就会各种报错，都是L大源码，不应该缺依赖啊

closed 2021-06-17 by coolsnowwolf