coolsnowwolf/lede #6635: 在更换内核编译时报错，应该如何解决？

yiqifengba, 2021-04-02:
## 1.关于你要提交的问题
我想更换openwrt社区官方编译版本相同的内核，利用了官方19.07.7SDK的部分文件，但在kernel的编译过程仍然报错，是SDK里面的内核相关文件存在问题？还是说其他文件没有修改到位？
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：替换了镜像站上SDK的target/linux/generic bcm53xx ，并对应修改了kernel-version中的小版本号与其哈希值，能帮我看看是配置文件哪里修改得还不够到位。

### (2) 路由器型号和固件版本

A：斐讯k3 更换的是19.07.7版本的4.14.221的内核

### (3) 详细日志

A：
make[4]: Entering directory '/usr/local/lede/build_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/linux-4.14.221'
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf  --oldconfig Kconfig
net/sched/Kconfig:44: warning: menuconfig statement without prompt
drivers/mtd/Kconfig:30: can't open file "drivers/mtd/mtdsplit/Kconfig"
scripts/kconfig/Makefile:87: recipe for target 'oldconfig' failed
make[5]: *** [oldconfig] Error 1
Makefile:527: recipe for target 'oldconfig' failed
make[4]: *** [oldconfig] Error 2
make[4]: Leaving directory '/usr/local/lede/build_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/linux-4.14.221'
Makefile:104: recipe for target '/usr/local/lede/build_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/linux-4.14.221/.configured' failed
make[3]: *** [/usr/local/lede/build_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/linux-4.14.221/.configured] Error 2
make[3]: Leaving directory '/usr/local/lede/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#11.90#5.54#104.66
toolchain/Makefile:98: recipe for target 'toolchain/kernel-headers/compile' failed
make[2]: *** [toolchain/kernel-headers/compile] Error 2
make[2]: Leaving directory '/usr/local/lede'
toolchain/Makefile:96: recipe for target '/usr/local/lede/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/stamp/.toolchain_compile' failed
make[1]: *** [/usr/local/lede/staging_dir/toolchain-arm_cortex-a9_gcc-8.4.0_musl_eabi/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/usr/local/lede'
/usr/local/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

AmadeusGhost, 2021-04-02:
sdk 编译不了内核

yiqifengba, 2021-04-02:
我是看这两个文件相对路径相同，才尝试调换了一下。那应该怎么改，能细嗦吗

yiqifengba, 2021-04-02:
而且我只是更换了target/linux下的文件夹，还是在lean的源码目录编译的，应该与SDK没什么关系吧

closed 2021-06-17 by coolsnowwolf