coolsnowwolf/lede #6639: 编译出的X86固件添加到ESXI下，硬盘大小总有零头出现，导致不能编辑设置

dwj0, 2021-04-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

如本来是256M的固件显示大小为256.50048828125, 我设置到257M ESXI提示不能设置。可否修正一下
[image]

hxlls, 2021-04-03:
这个是你转换软件问题和固件没关系

minlang112, 2021-04-03:
用StarWind V2V Image Converter 这个软件转就会这样……

pmdayu, 2021-04-03:
> [coolsnowwolf/lede] 编译出的X86固件添加到ESXI下，硬盘大小总有零头出现，导致不能编辑设置 (#6639)

我的就很正常发自我的iPhone------------------ 原始邮件 ------------------发件人: 狠心言少  ***@***.***>发送时间:

"target": "https://github.com/coolsnowwolf/lede/issues/6639#issuecomment-812821899",
"url": "https://github.com/coolsnowwolf/lede/issues/6639#issuecomment-812821899",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

dwj0, 2021-04-03:
是的， 我是用StarWind V2V Image Converter转换的，但很奇怪，我用openwrt官方源码编译出来的固件，就不存在这个问题

dwj0, 2021-04-03:
> 用StarWind V2V Image Converter 这个软件转就会这样……

我是用StarWind V2V Image Converter转换的，但很奇怪，我用openwrt官方源码编译出来的固件，就不存在这个问题

hxlls, 2021-04-03:
一个三方的转换软件出了问题，并不能判定固件有问题，如果是官方的转换工具出问题，那才是问题，但是现在我用vmfstool 转换没问题

cocci810123, 2021-04-03:
我是直接挂载vmdk文件，把vmdk文件上传到exsi后，去ssh用命令转换一下文件，否则会报scsi0_0的磁盘类型不受支持无法开机，然后就可以用了，最后重新关闭OP，直接编辑硬盘容量，把默认的容量直接扩大到16G，开机一点问题没有，您可以试试看

cocci810123, 2021-04-03:
转换命令vmkfstools -i oldfile.vmdk newfile.vmdk -d thin

dwj0, 2021-04-03:
> 转换命令vmkfstools -i oldfile.vmdk newfile.vmdk -d thin

你这个方法确实可用，thanks.
但我想根本问题还是出在固件大小上，下面一张图左边是用官方源码编译的大小，其大小刚好是256.5M， 所以用StarWind V2V Image Converter转换没有零头出现。右边的比左边多出了532字节，我想零头问题应该出在这里。
[image]

fhbyljj, 2021-05-22:
其实百度就有解决办法

closed 2021-06-17 by coolsnowwolf