coolsnowwolf/lede #6644: 适配硬改极路由B70的固件使用正常, 只有"状态"-"概览"页面报错

lon91ong, 2021-04-04:
报错信息: `Cannot set property 'portstate' of undefined`
其它功能使用正常, 只有**概览**页面会报错这个信息
问题文件锁定在[29_eth.js](https://github.com/garypang13/autocore/blob/master/files/switch/29_eth.js)
[mt7621_hiwifi_hc5962.dts](https://github.com/lon91ong/OpenWrt/blob/hiwifi_hc5962-spi/devices/hiwifi-hc5962/mt7621_hiwifi_hc5962.dts)文件中的`switch0`定义:
```
&switch0 {
	ports {
		port@1 {
			status = "okay";
			label = "lan1";
		};

		port@2 {
			status = "okay";
			label = "lan2";
		};

		port@3 {
			status = "okay";
			label = "lan3";
		};

		port@4 {
			status = "okay";
			label = "wan";
		};
	};
};
```
还对[02_network](https://github.com/coolsnowwolf/lede/blob/master/target/linux/ramips/mt7621/base-files/etc/board.d/02_network)文件做了修改:
```
--- a/target/linux/ramips/mt7621/base-files/etc/board.d/02_network
+++ b/target/linux/ramips/mt7621/base-files/etc/board.d/02_network
@@ -25,6 +25,7 @@
 		ucidef_set_interface_lan "eth0.1 ra0 rai0"
 		;;
 	hiwifi,hc5962|\
+	hiwifi,hc5962-32m|\
 	xiaomi,mir3p)
 		ucidef_set_interfaces_lan_wan "lan1 lan2 lan3 ra0 rai0" "wan"
 		;;
@@ -119,6 +120,8 @@
 		wan_mac="$(grep -m1 mac= "/dev/mtd${index}" | cut -d= -f2)"
 		lan_mac=$wan_mac
 		;;
+	hiwifi,hc5962|\
+	hiwifi,hc5962-32m|\
 	d-team,newifi-d2)
 		lan_mac=$(cat /sys/class/net/eth0/address)
 		wan_mac=$(mtd_get_mac_binary factory 0xe006)
@@ -136,12 +139,6 @@
 		lan_mac=$(cat /sys/class/net/eth0/address)
 		wan_mac=$(macaddr_add "$lan_mac" 1)
 		;;
-	hiwifi,hc5962)
-		lan_mac=$(mtd_get_mac_ascii bdinfo "Vfac_mac ")
-		label_mac=$lan_mac
-		[ -n "$lan_mac" ] || lan_mac=$(cat /sys/class/net/eth0/address)
-		wan_mac=$(macaddr_add "$lan_mac" 1)
-		;;
 	iodata,wnpr2600g)
 		wan_mac=$(mtd_get_mac_ascii u-boot-env wanaddr)
 		label_mac=$wan_mac
```
编译成功, 其余功能都正常使用, 唯独"状态"-"概览"页面报错
希望大侠指点
[image]

AmadeusGhost, 2021-04-04:
```
-	hiwifi,hc5962)
-		lan_mac=$(mtd_get_mac_ascii bdinfo "Vfac_mac ")
-		label_mac=$lan_mac
-		[ -n "$lan_mac" ] || lan_mac=$(cat /sys/class/net/eth0/address)
-		wan_mac=$(macaddr_add "$lan_mac" 1)
-		;;
```
这一段应该不能删掉吧，还有就是你硬改到 SPI 上之后 bdinfo 分区也要复制过去并且体现在 dts 里面，毕竟 b70 的 mac 地址是放在 bdinfo 里面而不是在 factory 分区。

lon91ong, 2021-04-04:
> ```
> -	hiwifi,hc5962)
> -		lan_mac=$(mtd_get_mac_ascii bdinfo "Vfac_mac ")
> -		label_mac=$lan_mac
> -		[ -n "$lan_mac" ] || lan_mac=$(cat /sys/class/net/eth0/address)
> -		wan_mac=$(macaddr_add "$lan_mac" [...]

我是参考同配置的[newifi-d1](https://github.com/coolsnowwolf/lede/blob/master/target/linux/ramips/dts/mt7621_lenovo_newifi-d1.dts)和[youku-L2](https://github.com/coolsnowwolf/lede/blob/master/target/linux/ramips/dts/mt7621_youku_yk-l2.dts)的dts文件拼凑的. 这两个机型都没有`bdinfo`分区, 而是直接指定了地址
```
&gmac0 {
	mtd-mac-address = <&factory 0xe000>;
};
```
搜索了一下代码, 似乎只有hiwifi是使用bdinfo分区的, 比如:[mt7620a_hiwifi_hc5x61.dtsi](https://github.com/coolsnowwolf/lede/blob/master/target/linux/ramips/dts/mt7620a_hiwifi_hc5x61.dtsi)
没有找到类似配置16m SPI闪存的使用bdinfo分区的例子, 地址不知道怎么算

还有一点奇怪之处, 我这么瞎拼凑的dts文件编译出来的固件居然也能用!!! 5G信号没有测试, 2.4G一切正常!!!

lon91ong, 2021-04-04:
> ```
> -	hiwifi,hc5962)
> -		lan_mac=$(mtd_get_mac_ascii bdinfo "Vfac_mac ")
> -		label_mac=$lan_mac
> -		[ -n "$lan_mac" ] || lan_mac=$(cat /sys/class/net/eth0/address)
> -		wan_mac=$(macaddr_add "$lan_mac" [...]

今天折腾了一番, 按你说的添加了bdinfo分区, 参考[mt7620a_hiwifi_hc5x61.dtsi](https://github.com/coolsnowwolf/lede/blob/master/target/linux/ramips/dts/mt7620a_hiwifi_hc5x61.dtsi)的格式, 对[mt7621_hiwifi_hc5962.dts](https://github.com/lon91ong/OpenWrt/commit/e720e6e38bb952d21c77a56192b11c84ebee2ea0)修改如下图:
[image]
同时对应修改了[配套的mt7621.mk和02_network](https://github.com/lon91ong/OpenWrt/blob/hiwifi_hc5962-spi/devices/hiwifi-hc5962/patches/hc5962.patch)
编译成功了, 刷入后"状态"-"概览"依旧报错
而且**多出了错误**: "网络"-"接口"中的MAC地址与EEPROM中的配置不符

**原因推测**:
1. bdinfo分区配置有误
2. 之前没有bdinfo分区的dts可以正确读取eeprom的mac配置, 只是29_eth.js无法读取接口数据, 可能是29_eth.js的配套问题

lon91ong, 2021-04-05:
@AmadeusGhost 放弃调试dts了, 有什么方法可以去掉"状态"-"概览"页面的有线接口部分?

AmadeusGhost, 2021-04-05:
> 没有找到类似配置16m SPI闪存的使用bdinfo分区的例子, 地址不知道怎么算

spi 上自己划一个分区出来就好，得把备份的 bdinfo 刷回去：`mtd write bdinfo.bin bdinfo`，mac 读取的方式还用原来的

lon91ong, 2021-04-05:
> > 没有找到类似配置16m SPI闪存的使用bdinfo分区的例子, 地址不知道怎么算
> 
> spi 上自己划一个分区出来就好，得把备份的 bdinfo 刷回去：`mtd write bdinfo.bin bdinfo`，mac 读取的方式还用原来的

没有备份锅bdinfo分区, 还是不折腾这个了

现在只想要把overview页面的有线接口部分屏蔽掉, 请教该如何操作?

lon91ong, 2021-04-05:
> > 没有找到类似配置16m SPI闪存的使用bdinfo分区的例子, 地址不知道怎么算
> 
> spi 上自己划一个分区出来就好，得把备份的 bdinfo 刷回去：`mtd write bdinfo.bin bdinfo`，mac 读取的方式还用原来的

@AmadeusGhost `bdinfo`分区是`readonly`, 无法写入。只能是刷入固件之后，导出编程器固件，在WinHex中将备份的`bdinfo.bin`内容写入编程器固件, 刷入...

奈何这么一番折腾下来, 尽管可以正确读取网卡地址了, 但是**"状态"**-**"概览"**页面依旧报错...

lon91ong, 2021-04-05:
最后的结果是: `CONFIG_PACKAGE_autocore-switch=n`可以阉割掉"状态"-"概览"页面的"接口"条目

closed 2021-04-05 by lon91ong