coolsnowwolf/lede #6645: 最新源码编译时无网络诊断中的IPv6路由跟踪组件

liuxichi, 2021-04-04:
如题 在~~编译时~~无法找到iputils-traceroute6

用/搜索都不能找到。希望大雕关注下

AmadeusGhost, 2021-04-04:
不需要了，有 busybox 提供

liuxichi, 2021-04-04:
> 不需要了，由 busybox 提供

但 编译后 网络诊断页面提示需要并且无法实现ipv6路由跟踪，由 busybox 提供是指需要在此勾选还是什么意思能说详细点吗？

Neverends4, 2021-04-05:
LS的意思是不需要那个包，ssh进路由 后直接打traceroute "ip" 命令即可。

liuxichi, 2021-04-05:
> LS的意思是不需要那个包，ssh进路由 后直接打traceroute "ip" 命令即可。

但是 我需要网页内可以直接用，并且有提示缺少很不爽。有什么办法可以自己添加吗？

AmadeusGhost, 2021-04-05:
https://github.com/coolsnowwolf/luci/pull/56 照着改一下就可以了

Dimitri2020007, 2021-04-05:
那个在编译菜单里改掉了

[image]

把黄色圈出来那部分改下再重新编译就出来了

liuxichi, 2021-04-05:
> 那个在编译菜单里改掉了
> 
> [image]
> 
> 把黄色圈出来那部分改下再重新编译就出来了

之前把黄色中4和5选上后依然没有，你说的改是指勾选上吗？你测试过成功没啊

Dimitri2020007, 2021-04-05:
我的已经成功

Dimitri2020007, 2021-04-05:
因为luci下的诊断链接的就是黄色圈出来的那部分，如果没有请git pull一下

liuxichi, 2021-04-05:
> 我的已经成功

那就怪事 我之前勾选上4和5，依然在路由管理页面提示路由跟踪需要安装支持ipv6

Dimitri2020007, 2021-04-05:
git pull下最新的代码然后feed更新再安装就是了

Dimitri2020007, 2021-04-05:
给你那个截图是今天最新的代码

liuxichi, 2021-04-05:
> 因为luci下的诊断链接的就是黄色圈出来的那部分，如果没有请git pull一下，然后执行./script/feeds update -a 和./script/feeds install -
行我再去试试昨天代码确定没有

closed 2021-04-06 by coolsnowwolf