coolsnowwolf/lede #6657: 如何让openwrt在x86设备上开启S3睡眠（suspend to ram）功能？

NagaseKouichi, 2021-04-07:
X86设备本身是支持S3睡眠的，但是在openwrt下运行 echo mem > /sys/power/state 会提示：
ash: write error: Function not implemented

是不是编译的时候有什么内核设置需要修改呢？
如果能支持S3睡眠的话，用来做NAS之类的设备就更方便了

minlang112, 2021-04-07:
让路由器S3睡眠不是个奇怪的想法么……

NagaseKouichi, 2021-04-07:
> 让路由器S3睡眠不是个奇怪的想法么……

openwrt也不只可以做路由器吧

Neverends4, 2021-04-07:
确实是小众需求，不过NAS也需要睡眠？如果觉得功耗高，或许你可以考虑用ARM box之类的做NAS，我自己DIY了一个，4盘+4K播放，待机功耗40w。

NagaseKouichi, 2021-04-07:
> 确实是小众需求，不过NAS也需要睡眠？如果觉得功耗高，或许你可以考虑用ARM box之类的做NAS，我自己DIY了一个，4盘+4K播放，待机功耗40w。

我有多台NAS，主NAS是24小时开机，其他的机器平时都睡眠了，需要的时候随时开机。
威联通有的NAS也是官方支持睡眠功能的

sxx1314, 2021-04-09:
- make kernel_menuconfig
- 自行打开 Suspend to RAM and standby 选项即可

NagaseKouichi, 2021-04-09:
> * make kernel_menuconfig
> * 自行打开 Suspend to RAM and standby 选项即可
> * 另外不止x86可以 很多mips方案也一样可以开启
> * 再提醒一下，如果你想低功耗还可以开启 Runtime PM。

感谢大佬指教，我试了下，运行make kernel_menuconfig之后的设置是保存在 build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.109 下面的 .config 文件，那如果用 github action 云编译，是可以把改好的 .config 文件预先放到这个目录吗？

另外 Runtime PM 这个选项没有找到呢。

closed 2021-06-17 by coolsnowwolf