coolsnowwolf/lede #6666: openwrt怎么更换安装的python版本？

debugpro, 2021-04-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
openwrt怎么更换安装的python版本？  默认是3.8.3
A：

### (2) 路由器型号和固件版本
rpi 4B
A：

### (3) 详细日志

A：

sxx1314, 2021-04-09:
- 追新请使用官方的package https://github.com/openwrt/packages,直接就是最新的3.9.4
- 想要降级  自行修改version.mk即可

debugpro, 2021-04-09:
> * 追新请使用官方的package [https://github.com/openwrt/packages,直接就是最新的3.9.4](https://github.com/openwrt/packages,%E7%9B%B4%E6%8E%A5%E5%B0%B1%E6%98%AF%E6%9C%80%E6%96%B0%E7%9A%843.9.4)
> * 想要降级  自行修改version.mk即可

我修改了python3-version.mk中的版本为3.7.0，现在报错如下：
Applying ./patches/001-enable-zlib.patch using plaintext: 
can't find file to patch at input line 3
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|--- a/Modules/Setup
|+++ b/Modules/Setup
--------------------------
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
Patch failed!  Please fix ./patches/001-enable-zlib.patch!
Makefile:354: recipe for target '/home/op/lede2/lede/build_dir/hostpkg/Python-3.7.0/.prepared40a7be0c4c2ef9c103a99aec425b043b_6664517399ebbbc92a37c5bb081b5c53' failed
make[3]: *** [/home/op/lede2/lede/build_dir/hostpkg/Python-3.7.0/.prepared40a7be0c4c2ef9c103a99aec425b043b_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/op/lede2/lede/feeds/packages/lang/python/python3'
time: package/feeds/packages/python3/host-compile#0.46#0.89#1.32
package/Makefile:111: recipe for target 'package/feeds/packages/python3/host/compile' failed
make[2]: *** [package/feeds/packages/python3/host/compile] Error 2
make[2]: Leaving directory '/home/op/lede2/lede'
package/Makefile:107: recipe for target '/home/op/lede2/lede/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile' failed
make[1]: *** [/home/op/lede2/lede/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/op/lede2/lede'
/home/op/lede2/lede/include/toplevel.mk:222: recipe for target 'world' failed

默认的python版本会进行patch, 如果版本改了patch也就对应不上了，how to do?

sxx1314, 2021-04-09:
> ## 我修改了python3-version.mk中的版本为3.7.0，现在报错如下：
> Applying ./patches/001-enable-zlib.patch using plaintext:
> can't find file to patch at input line 3
> Perhaps you used the wrong -p or --strip option?
> The [...]

自行修改patch

closed 2021-06-17 by coolsnowwolf