coolsnowwolf/lede #6667: Orangepi Zero 4.19升级5.4内核后内存预留激增

hjh142857, 2021-04-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：之前一直是一年前编译的4.19内核的固件，最近准备给旁路由Orangepi zero升级下系统，发现升级到5.4内核后reserved men激增了60+m，从13748K增长到了82516K，因为板子本身也只有256M的内存，5.4内核下free查看内存total是177MB，4.19下为244MB，挺影响使用的。

尝试过全新环境下git源码，只修改了target/linux/sunxi/Makefile中的内核版本，分别编译了默认配置的5.4内核固件和4.19内核固件，问题依旧。尝试自己查找资料没有解决，内核日志中的检索DMA和CMA相关的信息也是相同的。

因为目前找到的显示cpu温度的patch只能支持5.4内核，所以想更新到5.4内核使用。希望有大佬能指点下，先提前谢谢各位大佬了O(∩_∩)O

先附上发现问题的部分内核日志，5.4和4.19的完整内核日志在下方
```
## 5.4内核下部分内核日志
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x4bc6e000-0x4fc6e000] (64MB)
[    0.000000] Memory: 179628K/262144K available (6463K kernel code, 416K rwdata, 1784K rodata, 2048K init, 247K bss, 82516K reserved, 0K cma-reserved, 0K highmem)

## 4.19下部分内核日志
[    0.000000] Memory: 248396K/262144K available (5938K kernel code, 394K rwdata, 1644K rodata, 2048K init, 244K bss, 13748K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xd0800000 - 0xff800000   ( 752 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xd0000000   ( 256 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0x(ptrval) - 0x(ptrval)   (7955 kB)
[    0.000000]       .init : 0x(ptrval) - 0x(ptrval)   (2048 kB)
[    0.000000]       .data : 0x(ptrval) - 0x(ptrval)   ( 395 kB)
[    0.000000]        .bss : 0x(ptrval) - 0x(ptrval)   ( 245 kB)
```

### (2) 路由器型号和固件版本

A：Xunlong Orange Pi Zero OpenWrt R21.3.27 / LuCI Master (git-21.096.35788-b200c7b) 

### (3) 详细日志

A：
5.4完整内核日志
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.109 (ubuntu@ip-172-31-29-20) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3147-c41c174be)) #0 SMP Thu Apr 8 08:22:08 2021
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=30c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Xunlong Orange Pi Zero
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 65536
[    0.000000]   Normal zone: 768 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 65536 pages, LIFO batch:15
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: Using PSCI v0.1 Function IDs from DT
[    0.000000] percpu: Embedded 15 pages/cpu s30860 r8192 d22388 u61440
[    0.000000] pcpu-alloc: s30860 r8192 d22388 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64768
[    0.000000] Kernel command line: console=ttyS0,115200 earlyprintk root=/dev/mmcblk0p2 rootwait
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x4bc6e000-0x4fc6e000] (64MB)
[... 496 lines cut ...]
[   10.330684] cryptodev: driver 1.10 loaded.
[   10.333387] random: crng init done
[   10.338188] random: 1 urandom warning(s) missed due to ratelimiting
[   10.354856] xt_time: kernel timezone is -0000
[   10.383736] PPP generic driver version 2.4.2
[   10.389195] PPP MPPE Compression module registered
[   10.394855] NET: Registered protocol family 24
[   10.408878] kmodloader: done loading kernel modules from /etc/modules.d/*
[   13.750833] Generic PHY 0.1:01: attached PHY driver [Generic PHY] (mii_bus:phy_addr=0.1:01, irq=POLL)
[   13.762233] dwmac-sun8i 1c30000.ethernet eth0: No Safety Features support found
[   13.769588] dwmac-sun8i 1c30000.ethernet eth0: No MAC Management Counters available
[   13.779771] dwmac-sun8i 1c30000.ethernet eth0: PTP not supported by HW
[   13.787279] br-lan: port 1(eth0) entered blocking state
[   13.792719] br-lan: port 1(eth0) entered disabled state
[   13.798294] device eth0 entered promiscuous mode
[   13.804934] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
[   14.846431] fast-classifier (PBR safe v2.1.4a): starting up
[   14.852520] fast-classifier: registered
[   17.911859] dwmac-sun8i 1c30000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   17.920451] br-lan: port 1(eth0) entered blocking state
[   17.925724] br-lan: port 1(eth0) entered forwarding state
[   17.932996] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   32.471370] vcc3v0: disabling
[   32.474363] vcc5v0: disabling
```

AmadeusGhost, 2021-04-09:
Maybe refer to this commit: https://github.com/torvalds/linux/commit/36c4bef372a114222266858b1f342f7c03fabdd7

hxlls, 2021-04-09:
应该加入了视频处理的硬件驱动把，所以预留了内存

hjh142857, 2021-04-10:
> Maybe refer to this commit: [torvalds/linux@36c4bef](https://github.com/torvalds/linux/commit/36c4bef372a114222266858b1f342f7c03fabdd7)

> 应该加入了视频处理的硬件驱动把，所以预留了内存

感谢两位的提醒，已经定位到问题。并不是驱动的原因，5.4内核开启CONFIG_ARM_LPAE这个选项，会自动开启SWIOTLB，这货默认就吃掉64MB内存。
这是5.4内核的[Kconfig](https://github.com/torvalds/linux/blob/v5.4/arch/arm/mm/Kconfig)中关于ARM_LPAE的定义，相比4.19多出了`select SWIOTLB`
```
config ARM_LPAE
	bool "Support for the Large Physical Address Extension"
	depends on MMU && CPU_32v7 && !CPU_32v6 && !CPU_32v5 && \
		!CPU_32v4 && !CPU_32v3
	select PHYS_ADDR_T_64BIT
	select SWIOTLB
	help
	  Say Y if you have an ARMv7 processor supporting the LPAE page
	  table format and you would like to access memory beyond the
	  4GB limit. The resulting kernel image will not run on
	  processors without the LPA extension.

	  If unsure, say N.
```
根据解释，因为并没有4GB以上内存访问需求，所以只需要在lean源码中将`target/linux/sunxi/config-5.4`中的`CONFIG_ARM_LPAE=y`改成n即可，目前已经完全恢复正常

closed 2021-04-10 by hjh142857