coolsnowwolf/lede #6670: 自行编译的固件刷入后没有overlay分区, 无法保存配置

lon91ong, 2021-04-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue
现象描述与 [#6628](https://github.com/coolsnowwolf/lede/issues/6628) 类似, 但原因不同
## 2. 详细叙述

### (1) 具体问题

A：固件没有加载overlay分区, 无法保存配置

### (2) 路由器型号和固件版本

A：硬改16m闪存+256内存的K2(PSG1218A)

### (3) 详细日志
[image]
[image]

```
Thu Apr  8 16:22:31 2021 kern.notice kernel: [    0.000000] Linux version 5.4.109 (runner@fv-az168-856) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3147-c41c174be)) #0 Thu Apr 8 08:22:08 2021
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] Board has DDR2
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] Analog PMU set to hw control
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] Digital PMU set to hw control
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7620A ver:2 eco:6
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] CPU0 revision is: 00019650 (MIPS 24KEc)
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] MIPS: machine is Phicomm PSG1218 rev.A
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Thu Apr  8 16:22:31 2021 kern.warn kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Thu Apr  8 16:22:31 2021 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] Zone ranges:
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] Movable zone start for each node
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] Early memory node ranges
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
Thu Apr  8 16:22:31 2021 kern.debug kernel: [    0.000000] On node 0 totalpages: 65536
Thu Apr  8 16:22:31 2021 kern.debug kernel: [    0.000000]   Normal zone: 576 pages used for memmap
Thu Apr  8 16:22:31 2021 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Thu Apr  8 16:22:31 2021 kern.debug kernel: [    0.000000]   Normal zone: 65536 pages, LIFO batch:15
Thu Apr  8 16:22:31 2021 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Thu Apr  8 16:22:31 2021 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
Thu Apr  8 16:22:31 2021 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] Writing ErrCtl register=00067140
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] Readback ErrCtl register=00067140
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] Memory: 243084K/262144K available (5071K kernel code, 208K rwdata, 1116K rodata, 9852K init, 209K bss, 19060K reserved, 0K cma-reserved)
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] NR_IRQS: 256
Thu Apr  8 16:22:31 2021 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x308/0x4fc with crng_init=0
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] CPU Clock: 580MHz
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] clocksource: systick: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 583261500 ns
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] systick: enable autosleep mode
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] systick: running - mult: 214748, shift: 32
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6590553264 ns
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.000010] sched_clock: 32 bits at 290MHz, resolution 3ns, wraps every 7405115902ns
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.015479] Calibrating delay loop... 385.02 BogoMIPS (lpj=770048)
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.059577] pid_max: default: 32768 minimum: 301
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.068975] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.083383] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.105541] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.124852] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.138445] pinctrl core: initialized pinctrl subsystem
Thu Apr  8 16:22:31 2021 kern.info kernel: [    0.149386] NET: Registered protocol family 16
[... 144 lines cut ...]
Thu Apr  8 16:22:31 2021 user.info kernel: [   25.074154] kmodloader: done loading kernel modules from /etc/modules.d/*
Thu Apr  8 16:22:32 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Thu Apr  8 16:22:32 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Apr  8 16:22:32 2021 daemon.info dnsmasq[1656]: Connected to system UBus
Thu Apr  8 16:22:34 2021 kern.notice kernel: [   37.433919] random: crng init done
Thu Apr  8 16:22:48 2021 daemon.notice wpa_supplicant[1839]: Successfully initialized wpa_supplicant
Thu Apr  8 16:22:49 2021 user.notice : Added device handler type: 8021ad
Thu Apr  8 16:22:49 2021 user.notice : Added device handler type: 8021q
Thu Apr  8 16:22:49 2021 user.notice : Added device handler type: macvlan
Thu Apr  8 16:22:49 2021 user.notice : Added device handler type: veth
Thu Apr  8 16:22:49 2021 user.notice : Added device handler type: bridge
Thu Apr  8 16:22:49 2021 user.notice : Added device handler type: Network device
Thu Apr  8 16:22:49 2021 user.notice : Added device handler type: tunnel
Thu Apr  8 16:22:53 2021 daemon.notice procd: /etc/rc.d/S50autoreboot: Command failed: Not found
Thu Apr  8 16:22:53 2021 daemon.notice procd: /etc/rc.d/S50autoreboot: Auto REBOOT has started.
Thu Apr  8 16:22:53 2021 cron.err crond[2048]: crond (busybox 1.31.1) started, log level 5
Thu Apr  8 16:22:55 2021 daemon.notice procd: /etc/rc.d/S60flowoffload: net.ipv4.tcp_congestion_control = cubic
Thu Apr  8 16:22:56 2021 kern.info kernel: [   60.137213] 8021q: adding VLAN 0 to HW filter on device eth0
Thu Apr  8 16:22:56 2021 kern.info kernel: [   60.194903] br-lan: port 1(eth0.1) entered blocking state
Thu Apr  8 16:22:56 2021 kern.info kernel: [   60.205764] br-lan: port 1(eth0.1) entered disabled state
Thu Apr  8 16:22:56 2021 kern.info kernel: [   60.216955] device eth0.1 entered promiscuous mode
Thu Apr  8 16:22:56 2021 kern.info kernel: [   60.226559] device eth0 entered promiscuous mode
Thu Apr  8 16:22:56 2021 kern.info kernel: [   60.370053] br-lan: port 1(eth0.1) entered blocking state
Thu Apr  8 16:22:56 2021 kern.info kernel: [   60.380859] br-lan: port 1(eth0.1) entered forwarding state
```

coolsnowwolf, 2021-04-10:
需要更改dts分区

lon91ong, 2021-04-11:
> 需要更改dts分区

我对dts文件的修改只有一条, 就是适配16m闪存的:
`sed -i 's/<0x50000 0x7b0000>/<0x50000 0xfb0000>/g' target/linux/ramips/dts/*psg1218*`

lon91ong, 2021-04-12:
改16m闪存忘了还有一处必改的:
`sed -i 's/7872k/16064k/g' target/linux/ramips/image/mt7620.mk`

closed 2021-04-12 by lon91ong