coolsnowwolf/lede #6672: ssr+编译失败

rhww, 2021-04-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译ssr+失败

### (2) 路由器型号和固件版本

A：x64

### (3) 详细日志

A：

Building targets
/home/rhw/lede/dl/go-mod-cache/github.com/golang/protobuf@v1.5.2/proto/buffer.go:1:1: expected 'package', found 'EOF'
/home/rhw/lede/dl/go-mod-cache/github.com/marten-seemann/qtls-go1-16@v0.1.3/alert.go:1:1: expected 'package', found 'EOF'
/home/rhw/lede/dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20210322153248-0c34fe9e7dc2/chacha20/chacha_generic.go:1:1: expected 'package', found 'EOF'
/home/rhw/lede/dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20210322153248-0c34fe9e7dc2/chacha20poly1305/chacha20poly1305.go:1:1: expected 'package', found 'EOF'
/home/rhw/lede/dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20210322153248-0c34fe9e7dc2/hkdf/example_test.go:1:1: expected 'package', found 'EOF'
/home/rhw/lede/dl/go-mod-cache/google.golang.org/protobuf@v1.26.0/proto/bench_test.go:1:1: expected 'package', found 'EOF'
/home/rhw/lede/dl/go-mod-cache/google.golang.org/protobuf@v1.26.0/reflect/protoreflect/methods.go:1:1: expected 'package', found 'EOF'
/home/rhw/lede/dl/go-mod-cache/google.golang.org/protobuf@v1.26.0/runtime/protoimpl/impl.go:1:1: expected 'package', found 'EOF'

Makefile:71: recipe for target '/home/rhw/lede/build_dir/target-x86_64_musl/v2ray-plugin-4.37.2/.built' failed
make[3]: *** [/home/rhw/lede/build_dir/target-x86_64_musl/v2ray-plugin-4.37.2/.built] Error 1
make[3]: Leaving directory '/home/rhw/lede/package/lean/v2ray-plugin'
time: package/lean/v2ray-plugin/compile#0.41#0.47#0.54
package/Makefile:111: recipe for target 'package/lean/v2ray-plugin/compile' failed
make[2]: *** [package/lean/v2ray-plugin/compile] Error 2
make[2]: Leaving directory '/home/rhw/lede'
package/Makefile:107: recipe for target '/home/rhw/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/rhw/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/rhw/lede'
/home/rhw/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

AmadeusGhost, 2021-04-11:
`rm -rf dl/go-mod-cache` and try again.

xiehuayi75, 2021-04-19:
网络代理问题和某些依赖不不全，依赖问题可以执行 lede/package$下运行 或者openwrt/package$下运行

 git clone https://github.com/kenzok8/small.git

closed 2021-06-17 by coolsnowwolf