coolsnowwolf/lede #6675: OrangePi R1 plus 启动失败

garlicsoft, 2021-04-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译成功，写入sd卡无法启动

### (2) 路由器型号和固件版本

A：OrangePi R1 plus

### (3) 详细日志

A：
 entries: 512 (order: 2, 16384 bytes, linear)
[    0.102688] UDP-Lite hash table entries: 512 (order: 2, 16384 bytes, linear)
[    0.103587] NET: Registered protocol family 1
[    0.104022] PCI: CLS 0 bytes, default 64
[    0.105997] workingset: timestamp_bits=46 max_order=18 bucket_order=0
[    0.115661] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.116200] jffs2: version 2.2 (NAND) (SUMMARY) (ZLIB) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.144727] Block layer SCSI generic (bsg) driver version 0.4 loaded (major 252)
[    0.145436] io scheduler mq-deadline registered
[    0.152164] dma-pl330 ff1f0000.dmac: Loaded driver for PL330 DMAC-241330
[    0.152790] dma-pl330 ff1f0000.dmac: .DBUFF-128x8bytes Num_Chans-8 Num_Peri-20 Num_Events-16
[    0.156003] Serial: 8250/16550 driver, 4 ports, IRQ sharing enabled
[    0.158534] ff120000.serial: ttyS1 at MMIO 0xff120000 (irq = 12, base_baud = 1500000) is a 16550A
[    0.160203] ff130000.serial: ttyS2 at MMIO 0xff130000 (irq = 13, base_baud = 1500000) is a 16550A
[    0.161160] printk: console [ttyS2] enabled
[    0.161160] printk: console [ttyS2] enabled
[    0.161912] printk: bootconsole [uart8250] disabled
[    0.161912] printk: bootconsole [uart8250] disabled
[    0.164687] rockchip-drm display-subsystem: [drm:rockchip_drm_platform_probe] *ERROR* No available vop found for display-subsystem.
[    0.166315] lima ff300000.gpu: IRQ pmu not found
[    0.166933] lima ff300000.gpu: IRQ ppmmu2 not found
[    0.167373] lima ff300000.gpu: IRQ ppmmu3 not found
[    0.167810] lima ff300000.gpu: IRQ ppmmu4 not found
[    0.168248] lima ff300000.gpu: IRQ ppmmu5 not found
[    0.168683] lima ff300000.gpu: IRQ ppmmu6 not found
[    0.169161] lima ff300000.gpu: IRQ ppmmu7 not found
[    0.169616] lima ff300000.gpu: gp - mali450 version major 0 minor 0
[    0.170223] lima ff300000.gpu: pp0 - mali450 version major 0 minor 0
[    0.170840] lima ff300000.gpu: pp1 - mali450 version major 0 minor 0
[    0.171438] lima ff300000.gpu: IRQ pp2 not found
[    0.171854] lima ff300000.gpu: IRQ pp3 not found
[    0.172269] lima ff300000.gpu: IRQ pp4 not found
[    0.172682] lima ff300000.gpu: IRQ pp5 not found
[    0.173113] lima ff300000.gpu: IRQ pp6 not found
[    0.173529] lima ff300000.gpu: IRQ pp7 not found
[    0.173947] lima ff300000.gpu: l2 cache 8K, 4-way, 64byte cache line, 128bit external bus
[    0.174673] lima ff300000.gpu: l2 cache 64K, 4-way, 64byte cache line, 128bit external bus
[    0.176000] lima ff300000.gpu: bus rate = 163840000
[    0.176443] lima ff300000.gpu: mod rate = 163840000
[    0.177300] [drm] Initialized lima 1.0.0 20190217 for ff300000.gpu on minor 0
[    0.186009] loop: module loaded
[    0.186308] mtip32xx Version 1.3.1
[    0.190060] spi-nor spi0.0: mx25l12805d (16384 Kbytes)
[    0.194281] libphy: Fixed MDIO Bus: probed
[    0.195918] rk_gmac-dwmac ff540000.ethernet: IRQ eth_wake_irq not found
[    0.196510] rk_gmac-dwmac ff540000.ethernet: IRQ eth_lpi not found
[    0.197287] rk_gmac-dwmac ff540000.ethernet: PTP uses main clock
[    0.197863] rk_gmac-dwmac ff540000.ethernet: phy regulator is not available yet, deferred probing
[    0.200226] dwc3 ff600000.dwc3: Failed to get clk 'ref': -2
[    0.201342] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[... 72 lines cut ...]
[    0.424815] rk_gmac-dwmac ff540000.ethernet: TX Checksum insertion supported
[    0.425479] rk_gmac-dwmac ff540000.ethernet: Wake-Up On Lan supported
[    0.426054] rk_gmac-dwmac ff540000.ethernet: Normal descriptors
[    0.426577] rk_gmac-dwmac ff540000.ethernet: Ring mode enabled
[    0.427096] rk_gmac-dwmac ff540000.ethernet: Enable RX Mitigation via HW Watchdog Timer
[    0.428015] libphy: stmmac: probed
[    0.497674] random: fast init done
[    0.499433] rk808-rtc rk808-rtc: setting system clock to 2016-01-21T08:50:08 UTC (1453366208)
[    0.501885] Waiting for root device PARTUUID=5452574f-02...
[    0.632938] usb 4-1: new SuperSpeed Gen 1 USB device number 2 using xhci-hcd
[    0.967223] dwmmc_rockchip ff500000.dwmmc: Busy; trying anyway
[    1.467754] mmc_host mmc0: Timeout sending command (cmd 0x202000 arg 0x0 status 0x80202000)
[    1.504657] mmc0: error -110 whilst initialising SD card
[    2.006082] dwmmc_rockchip ff500000.dwmmc: Busy; trying anyway
[    2.506614] mmc_host mmc0: Timeout sending command (cmd 0x202000 arg 0x0 status 0x80202000)
[    2.521201] mmc_host mmc0: Bus speed (slot 0) = 300000Hz (slot req 300000Hz, actual 300000HZ div = 0)
[    3.052235] dwmmc_rockchip ff500000.dwmmc: Busy; trying anyway
[    3.552768] mmc_host mmc0: Timeout sending command (cmd 0x202000 arg 0x0 status 0x80202000)
[    3.567365] mmc_host mmc0: Bus speed (slot 0) = 200000Hz (slot req 200000Hz, actual 200000HZ div = 0)
[    4.105186] dwmmc_rockchip ff500000.dwmmc: Busy; trying anyway
[    4.605720] mmc_host mmc0: Timeout sending command (cmd 0x202000 arg 0x0 status 0x80202000)
[    4.620321] mmc_host mmc0: Bus speed (slot 0) = 100000Hz (slot req 100000Hz, actual 100000HZ div = 0)
[    5.178334] dwmmc_rockchip ff500000.dwmmc: Busy; trying anyway
[    5.678867] mmc_host mmc0: Timeout sending command (cmd 0x202000 arg 0x0 status 0x80202000)
[   31.708935] vcc_sd: disabling

AmadeusGhost, 2021-04-11:
要不换张 sd 卡试试吧

garlicsoft, 2021-04-11:
> 要不换张 sd 卡试试吧

用过balenaEtcher跟dd命令写入过，都失败了。同一张卡用orangepi官方提供的img文件是可以启动的主要是。

AmadeusGhost, 2021-04-11:
试试看这个 commit 吧：https://github.com/AmadeusGhost/lede/commit/186da22398ffb786858a15638a7eacb8ce1119d7
```
wget https://github.com/AmadeusGhost/lede/commit/186da22398ffb786858a15638a7eacb8ce1119d7.patch
git apply 186da22398ffb786858a15638a7eacb8ce1119d7.patch
```

garlicsoft, 2021-04-14:
> 试试看这个 commit 吧：[AmadeusGhost@186da22](https://github.com/AmadeusGhost/lede/commit/186da22398ffb786858a15638a7eacb8ce1119d7)
> 
> ```
> wget https://github.com/AmadeusGhost/lede/commit/186da22398ffb786858a15638a7eacb8ce1119d7.patch
> git apply 186da22398ffb786858a15638a7eacb8ce1119d7.patch
> ```

感谢，不过我编译成功了还是无法启动，最后还是用官方提供的openwrt仓库，配合helloworld源编译出来一份包含ssr plus的镜像。
也挺多波折的（主要是官方git版本里的openwrt版本比较低，但是feeds源又有变化，以及ssr需要一些lede的包需要手动拷进去）

BINGHUAice, 2021-04-17:
官方提供的固件能够使用不代表编译的lean的固件能够使用，我最近也在折腾R1P，我手里很多张TF卡，我发现不管什么固件总有TF卡不能够引导。除了更换TF卡与固件，这里给出一些自己摸索出来的建议：1：更换写盘工具：比如Win32DiskImager，2：更换固件的打包方式，3：更换TF卡读卡器（没错，这个也会影响，我还遇到了）4：写入固件之前可以尝试格盘，并且软件可以多试几个如SD Card Formatter，LLFTOOL等。

garlicsoft, 2021-04-19:
> 官方提供的固件能够使用不代表编译的lean的固件能够使用，我最近也在折腾R1P，我手里很多张TF卡，我发现不管什么固件总有TF卡不能够引导。除了更换TF卡与固件，这里给出一些自己摸索出来的建议：1：更换写盘工具：比如Win32DiskImager，2：更换固件的打包方式，3：更换TF卡读卡器（没错，这个也会影响，我还遇到了）4：写入固件之前可以尝试格盘，并且软件可以多试几个如SD Card Formatter，LLFTOOL等。

感谢，我的手上只有两张空闲的tf卡（sandisk）都引导不能，换了官方代码编译出来的固件可以引导。目前暂时先把思路放在lede的机器补丁问题上，之前有替换代码编译完成，但也没能成功引导，可能还有些地方没有注意到。

AmadeusGhost, 2021-04-21:
要不再试试这个更改吧，不能启动的话最好能有 ttl log 方便分析问题🤣
```
git pull
git revert 6aa3ed2b19a680b406fafade10bf01095f5f18be
wget https://github.com/AmadeusGhost/lede/commit/39833b3bb6a153bc6c6e8600266d9a8d1159233e.patch
git apply 39833b3bb6a153bc6c6e8600266d9a8d1159233e.patch
```

closed 2021-06-17 by coolsnowwolf