coolsnowwolf/lede #6677: make download下载便秘, 怎么用国内镜像替换?

lon91ong, 2021-04-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
本地编译下载阶段`make download -j8` 多次下载总是卡在此处:
```
time: tools/sed/download#0.03#0.07#0.09
  2 14.3M    2  319k    0     0   5778      0  0:43:27  0:00:56  0:42:31  3400
```
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：feed.conf中涉及github的地址都用https://gitclone.com/github.com/替换可以提高下载速度
用`sed -i 's/github.com\//github.com.cnpmjs.org\//g' 'grep -l -e github.com/.*/.*\.git ./package/lean/*/Makefile'`替换可以提高package中涉及github部分的下载速度,
到了make download 阶段, 死活下不动了, 看了一下`lede\tools\sed\Makefile`中的定义`PKG_SOURCE_URL:=@GNU/$(PKG_NAME)`
不知这个`@GNU`是在哪里定义的, 想要用国内GNU镜像地址替换一下, 不然总是卡死在此处

AmadeusGhost, 2021-04-11:
已经有国内镜像了：https://github.com/coolsnowwolf/lede/blob/master/scripts/download.pl#L217

lon91ong, 2021-04-12:
> 已经有国内镜像了：https://github.com/coolsnowwolf/lede/blob/master/scripts/download.pl#L217

这个已经是主分支， 怎么还是会下不动？
怎么做才能手动触发？？？

Cathgao, 2021-04-12:
你可以试试域名劫持，如果url格式一样的话

jeasonliang, 2021-04-16:
推荐一个东西 v2raya
这玩意好使  https://github.com/v2rayA/v2rayA

lon91ong, 2021-04-16:
> 推荐一个东西 v2raya
> 这玩意好使 https://github.com/v2rayA/v2rayA

我用的是[v2rayN](https://github.com/2dust/v2rayN), 这个好用与否主要取决于机场吧, 跟用什么客户端没太大关系

jeasonliang, 2021-04-17:
> > 推荐一个东西 v2raya
> > 这玩意好使 https://github.com/v2rayA/v2rayA
> 
> 我用的是[v2rayN](https://github.com/2dust/v2rayN), 这个好用与否主要取决于机场吧, 跟用什么客户端没太大关系

v2rayA也是基于v2rayN的，只不过适用于cli下使用！

closed 2021-06-17 by coolsnowwolf