coolsnowwolf/lede #6678: 稳定版请求更新

tttt2001tttt, 2021-04-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

最新稳定版编译出错，无法编译Docker

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
请求更新稳定版的库。谢谢大佬！
A：

### (2) 路由器型号和固件版本
4417U软路由
A：

### (3) 详细日志
rm -f /home/happy/openwrt/build_dir/target-x86_64_musl/docker-ce-19.03.8/.built
touch /home/happy/openwrt/build_dir/target-x86_64_musl/docker-ce-19.03.8/.built_check
( export GOOS="linux" GOARCH="amd64" GO386="" GOARM="" GOMIPS="" GOMIPS64="" CGO_ENABLED=1 CC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/happy/openwrt/build_dir/target-x86_64_musl/docker-ce-19.03.8:docker-ce-19.03.8 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/happy/openwrt/staging_dir/target-x86_64_musl/usr/include -I/home/happy/openwrt/staging_dir/target-x86_64_musl/include -I/home/happy/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/include -I/home/happy/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include/fortify -I/home/happy/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/happy/openwrt/build_dir/target-x86_64_musl/docker-ce-19.03.8:docker-ce-19.03.8 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/happy/openwrt/staging_dir/target-x86_64_musl/usr/lib -L/home/happy/openwrt/staging_dir/target-x86_64_musl/lib -L/home/happy/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/usr/lib -L/home/happy/openwrt/staging_dir/toolchain-x86_64_gcc-7.5.0_musl/lib -znow -zrelro" GOPATH="/home/happy/openwrt/build_dir/target-x86_64_musl/docker-ce-19.03.8/.go_work/build" GOCACHE="/home/happy/openwrt/tmp/go-build" GOMODCACHE="/home/happy/openwrt/dl/go-mod-cache" GOENV=off GITCOMMIT=afacb8b7f0  DOCKER_GITCOMMIT=afacb8b7f0  DOCKER_BUILDTAGS='' VERSION=19.03.8 && echo "Compiling CLI..." && cd /home/happy/openwrt/build_dir/target-x86_64_musl/docker-ce-19.03.8/cli && ./scripts/build/binary && echo "Compiling Engine..." && cd /home/happy/openwrt/build_dir/target-x86_64_musl/docker-ce-19.03.8/engine && ./hack/make.sh binary )
Compiling CLI...
Building statically linked build/docker-linux-amd64
no required module provides package github.com/docker/cli/cmd/docker: go.mod file not found in current directory or any parent directory; see 'go help modules'
Makefile:118: recipe for target '/home/happy/openwrt/build_dir/target-x86_64_musl/docker-ce-19.03.8/.built' failed
make[3]: *** [/home/happy/openwrt/build_dir/target-x86_64_musl/docker-ce-19.03.8/.built] Error 1
make[3]: Leaving directory '/home/happy/openwrt/feeds/packages/utils/docker-ce'
time: package/feeds/packages/docker-ce/compile#0.64#5.98#7.98
package/Makefile:111: recipe for target 'package/feeds/packages/docker-ce/compile' failed
make[2]: *** [package/feeds/packages/docker-ce/compile] Error 2
make[2]: Leaving directory '/home/happy/openwrt'
package/Makefile:107: recipe for target '/home/happy/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/happy/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/happy/openwrt'
/home/happy/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
A：

vodott, 2021-04-16:
求稳定版地址。谢谢

diamondt5353, 2021-04-26:
[image]

closed 2021-06-17 by coolsnowwolf