coolsnowwolf/lede #6684: wan口没有自动添加进负载均衡

Dimitri2020007, 2021-04-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

2021.4.6 之后编译的openwrt不再将wan口自动添加进负载均衡里了

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：

ntlf9t, 2021-04-13:
config globals 'globals'
	option mmx_mask '0x3F00'
	option local_source 'lan'

config interface 'wan'
	option enabled '0'
	list track_ip '114.114.114.114'
	list track_ip '114.114.115.115'
	option family 'ipv4'
	option reliability '2'
	option count '1'
	option timeout '2'
	option failure_latency '1000'
	option recovery_latency '500'
	option failure_loss '20'
	option recovery_loss '5'
	option interval '5'
	option down '3'
	option up '8'

config member 'wan_m1_w3'
	option interface 'wan'
	option metric '1'
	option weight '3'

config policy 'balanced'
	option last_resort 'unreachable'
	list use_member 'wan_m1_w3'

config rule 'https'
	option sticky '1'
	option dest_port '443'
	option proto 'tcp'
	option use_policy 'balanced'

config rule 'default_rule'
	option dest_ip '0.0.0.0/0'
	option use_policy 'balanced'

Dimitri2020007, 2021-04-14:
> config globals 'globals'
> option mmx_mask '0x3F00'
> option local_source 'lan'
>
> config interface 'wan'
> option enabled '0'
> list track_ip '114.114.114.114'
> list track_ip '114.114.115.115'
> option family [...]

这个可以在默认设置文件中添加吗？

Dimitri2020007, 2021-04-14:
已经找到修改位置了，谢谢

closed 2021-04-14 by Dimitri2020007