coolsnowwolf/lede #6687: 单线多拨怎么定时重拨？

lexin8, 2021-04-14:
/sbin/ifup pppoe-vwan1 或者手动会有一定几率无法拨号成功！
目前在syncdial界面手动**重新拨号**或者**路由重启**这两种方式比较完美。

sqliuchang, 2021-04-14:
巧了，多拨的代码我之前研究过一些
这个issue也提到过你的问题
#5296 

相关代码主要集中在这几个文件里面
https://github.com/coolsnowwolf/lede/blob/master/package/network/services/shellsync/src/shellsync.c
https://github.com/coolsnowwolf/lede/blob/fae07ff3d97d37a1684b7d61fb1d58501bb3c3ed/package/network/services/ppp/files/ppp.sh#L234-L240
(ps: 这里dialwait参数并没有被传递进shellsync函数，而是固定为10s的超时时间)
https://github.com/coolsnowwolf/lede/blob/master/package/lean/luci-app-syncdial

至于你说的
> 目前在syncdial界面手动重新拨号或者路由重启这两种方式比较完美。

看这个luci的代码
https://github.com/coolsnowwolf/lede/blob/fae07ff3d97d37a1684b7d61fb1d58501bb3c3ed/package/lean/luci-app-syncdial/luasrc/controller/syncdial.lua#L13
可以知道那个按钮其实就是执行
`killall -9 pppd`
强制杀死所有pppd进程，然后由procd进程守护自动重新拨号。

---
题外话：
多拨最初是蝈蝈大佬写的代码，commit的历史记录这里有
https://github.com/981213/luci/commits/master/applications/luci-app-syncdial
pppconnectcheck这个脚本没了也是他删除的 https://github.com/981213/luci/commit/61295ca880e1b35f056f2e6a795f375bef799433
> 因为这个功能在大部分情况下是废的……

上面提到的dialwait参数也是用在pppconnectcheck脚本里面的，而不是给shellsync用的，所以其实这个参数可以去掉了
需要的话可以去 https://github.com/981213/luci/commits/1597d11cba9f8eaa7c519a9b1f36ab3adb53b305/applications/luci-app-syncdial/root/bin/pppconnectcheck 这里把这个脚本给补回去，不过我个人也觉得不太用得到

lexin8, 2021-04-14:
感谢大神回复！命令方式运行后其他网口也会断开，有部分多拨没拨上，可能插件就这样吧，很久没有维护了。

sqliuchang, 2021-04-14:
> 感谢大神回复！命令方式运行后其他网口也会断开，有部分多拨没拨上，可能插件就这样吧，很久没有维护了。

并发多拨需要断开所有拨号再重新并发，你想单独重拨一个的话就会出现你刚开始的情况，有时可以有时不可以
你要是执行`killall -9 pppd`还有部分拨不上的话，就可以考虑把pppconnectcheck还原回去，执行
```
killall -9 pppd
/bin/pppconnectcheck &
```

pppconnectcheck这个脚本也有挺多版本的，蝈蝈的最后一版 https://github.com/981213/luci/blob/1597d11cba9f8eaa7c519a9b1f36ab3adb53b305/applications/luci-app-syncdial/root/bin/pppconnectcheck
koolshare的版本
```
#!/bin/sh
#PPPoE拨号在线接口数检测及重拨脚本
#Copyright (C) 2015 GuoGuo<gch981213@gmail.com>
. /lib/functions.sh

[ -f /etc/config/syncdial ] || {
	return 1
}

get_ppp_num(){
	if [ -n "$(ubus call network.interface.$1 status 2>/dev/null | jsonfilter -e '@.route[@].nexthop' 2>/dev/null)" ]; then
		ppp_num=1
	else
		ppp_num=0
	fi
	echo -n $ppp_num
}

action_interface(){
	local action=$1
	local wan_action=$2
	local first_device=$3
	for device in $first_device
	do
		if [ "$device" = "$wan_action" ]; then
			[ "$old_frame" -eq 0 ] && {
				logger -t PPPoE-IFChecker "Interfaces: $wan_action; Action: $action"
				$action $wan_action
			}
		else
			logger -t PPPoE-IFChecker "Interfaces: $device; Action: $action"
			$action $device
		fi
	done
}

sync_redial(){
	local wan_select=$1
	local dialnum=$2
	p_ifname=`uci -q get network.$wan_select.ifname`
	ppp_num=$(get_ppp_num $wan_select)
	if [ "$old_frame" -eq 0 ]; then
		s_ifname=`uci show network| grep macvlandev_macvlan | grep "ifname='$p_ifname'"|awk -F'.' '{print $2}' |awk -F'_' '{print $2}'`
	else
		s_ifname=`uci show network| grep "ifname='$p_ifname" | awk -F'.' '{print $2}' | grep vwan`
	fi
	first_device=$wan_select
	for ifname in $s_ifname
	do
		s_device=vwan${ifname:7}
		s_online_num=$(get_ppp_num $s_device)
		ppp_num=$(($ppp_num + $s_online_num))
		first_device="$first_device $s_device"
	done
	
	logger -t PPPoE-IFChecker "$ppp_num interfaces come from $wan_select are online."
	if [ "$ppp_num" -lt $dialnum ]; then
		logger -t PPPoE-IFChecker "Stop all interfaces created by $wan_select."
		action_interface ifdown $wan_select "$first_device"
		sleep $dialwait
		logger -t PPPoE-IFChecker "Start all interfaces created by $wan_select."
		action_interface ifup $wan_select "$first_device"
		logger -t PPPoE-IFChecker "Interfaces come from $wan_select have been restarted."
		logger -t PPPoE-IFChecker "Another checker will be started 60s later."
		sleep 60 && pppconnectcheck &
	else
		logger -t PPPoE-IFChecker "Nothing to do for the origin interface: $wan_select. Exit."
	fi
}

config_load "syncdial"
config_get_bool enabled config enabled 0
config_get_bool dialchk config dialchk 0
config_get_bool old_frame config old_frame
config_get dial config dialnum 0
config_get dial2 config dialnum2 0
config_get dialwait config dialwait 20
config_get wanselect config wanselect
config_get dial_type config dial_type
[ "$dial_type" -eq 2 ] && {
	config_get wanselect2 config wanselect2
}

[ "$enabled" = "0" ] && {
	echo "syncdial is disabled."
	logger -t PPPoE-IFChecker "Syncdial is not enabled. Exit pppconnectcheck."
	return 0
}

[ "$dialchk" = "0" ] && {
	echo "dial check disabled."
	return 0
}

[ $(ps | grep -v grep | grep -c "pppconnectcheck") -gt 2 ] && logger -t PPPoE-IFChecker "Another checker is running.exit." && return 1

sleep 5s
if [ "$dial_type" -eq 2 ]; then
	#二条线掉线检测
	sync_redial $wanselect $dial &
	sync_redial $wanselect2 $dial2
else
	#一条线掉线检测
	sync_redial $wanselect $dial
fi

return 0
```

badcrazy, 2021-04-14:
killall -HUP pppd 加到计划任务里

lexin8, 2021-04-16:
killall -9 pppd/killall -HUP pppd都是一样的，会有一路拨不上，openwrt可能就这样了吧，其他固件可以单一个拨。