coolsnowwolf/lede #6690: 这是什么错误

qqsir-dev, 2021-04-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
/home/qqsir/lede/staging_dir/host/bin/grub-mkimage -d /home/qqsir/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/grub-core              -p /boot/grub -O "x86_64"-efi -c ./files/grub-early.cfg -o /home/qqsir/lede/staging_dir/target-x86_64_musl/image/grub2/iso-bootx64.efi at_ke             yboard boot chain configfile fat iso9660 linux ls part_msdos part_gpt reboot serial test efi_gop efi_uga
touch /home/qqsir/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/.pkgdir/grub2-efi.installed
mkdir -p /home/qqsir/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/qqsir/lede/tmp/.root-copy.flock -c 'cp -fpR /home/qqsir/lede/build_dir/target-x86_64_musl/linux-x86_64/grub-efi/grub-2.04/             .pkgdir/grub2-efi/. /home/qqsir/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/qqsir/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.grub2-efi_installed
echo "grub2-efi" >> /home/qqsir/lede/staging_dir/target-x86_64_musl/pkginfo/grub2.efi.install
make[3]: Leaving directory '/home/qqsir/lede/package/boot/grub2'
time: package/boot/grub2/efi/compile#89.25#10.79#94.84
make[2]: Leaving directory '/home/qqsir/lede'
make[1]: *** [package/Makefile:107: /home/qqsir/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/qqsir/lede'
make: *** [/home/qqsir/lede/include/toplevel.mk:224: world] Error 2
qqsir@code:~/lede$

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

wqzxhpx, 2021-04-14:
我也是这个错误，不清楚这个grub2是什么

ricksuzade-maker, 2021-04-14:
我今天也编译错误，不过是在编译luci-app-samba的时候，也是跳出来toplevel.mk这句

ma20at, 2021-04-14:
不知道最新的源码哪里有问题，git pull更新之后就编译不过去了
make[1]: *** [package/Makefile:107: /home/openwrt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
Removing child 0x55edc098d9a0 PID 691322 from chain.
make[1]: Leaving directory '/home/openwrt/lede'
正在中止失败的子进程 0x5647a2dfd8f0 PID 684686 
make: *** [/home/openwrt/lede/include/toplevel.mk:224：world] 错误 2
从链中删除子进程 0x5647a2dfd8f0 PID 684686。

noyjcl, 2021-04-16:
更新后编译出现同样的问题。

noyjcl, 2021-04-16:
make: *** [/home/yjiang/lede/include/toplevel.mk:224：world] 错误 2

galaxyskyknight, 2021-04-18:
单线程编译，这种我后来发现都是多线程的锅

closed 2021-06-17 by coolsnowwolf

honey11155, 2021-07-27:
我就是开多线程的........