coolsnowwolf/lede #6693: 请及时更新v2ray-plugin-4.37.3的go.sum校验和

Dimitri2020007, 2021-04-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

使用github actions在线编译，编译到v2ray-plugin-4.37.3出现go校验和错误，源与下载的文件校验和不一致

### (1) 具体问题

A：

2021-04-15T03:12:57.3419590Z verifying github.com/v2fly/v2ray-core/v4@v4.37.3: checksum mismatch
2021-04-15T03:12:57.3421350Z 	downloaded: h1:FGmf5rtUo3xGHyZGJi8jEa7AKin05ROfPtegmGbkCKg=
2021-04-15T03:12:57.3422836Z 	go.sum:     h1:Kf9Qot2lLYccEzQfoCdt9img6zo5Vf+PEaoTMfa2S8k=

### (2) 路由器型号和固件版本

A：x86-64 efi

### (3) 详细日志

A：

2021-04-15T03:12:55.7891039Z make[3]: Entering directory '/workdir/openwrt/package/lean/v2ray-plugin'
2021-04-15T03:12:55.9414213Z rm -f /workdir/openwrt/build_dir/target-x86_64_musl/v2ray-plugin-4.37.3/.built
2021-04-15T03:12:55.9437984Z touch /workdir/openwrt/build_dir/target-x86_64_musl/v2ray-plugin-4.37.3/.built_check
2021-04-15T03:12:55.9478398Z CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/workdir/openwrt/tmp/go-build" GO_MOD_CACHE_DIR="/workdir/openwrt/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/shadowsocks/v2ray-plugin" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/shadowsocks/v2ray-plugin/..." GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/workdir/openwrt/build_dir/target-x86_64_musl/v2ray-plugin-4.37.3" GO_BUILD_DIR="/workdir/openwrt/build_dir/target-x86_64_musl/v2ray-plugin-4.37.3/.go_work/build" GO_BUILD_BIN_DIR="/workdir/openwrt/build_dir/target-x86_64_musl/v2ray-plugin-4.37.3/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/workdir/openwrt/staging_dir/target-x86_64_musl/usr/share/gocode/src" GOOS="linux" GOARCH="amd64" GO386="" GOARM="" GOMIPS="" GOMIPS64="" CGO_ENABLED=1 CC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/v2ray-plugin-4.37.3=v2ray-plugin-4.37.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/workdir/openwrt/staging_dir/target-x86_64_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -ffile-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/v2ray-plugin-4.37.3=v2ray-plugin-4.37.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib -L/workdir/openwrt/staging_dir/target-x86_64_musl/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -znow -zrelro" GOPATH="/workdir/openwrt/build_dir/target-x86_64_musl/v2ray-plugin-4.37.3/.go_work/build" GOCACHE="/workdir/openwrt/tmp/go-build" GOMODCACHE="/workdir/openwrt/dl/go-mod-cache" GOENV=off /usr/bin/env bash /workdir/openwrt/feeds/packages/lang/golang//golang-build.sh build -v -trimpath -ldflags "all=-buildid '1618378600' -linkmode external -extldflags '-L/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib -L/workdir/openwrt/staging_dir/target-x86_64_musl/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'"        -ldflags "-s -w -X main.VERSION=v4.37.3 -buildid '1618378600' -linkmode external -extldflags '-L/workdir/openwrt/staging_dir/target-x86_64_musl/usr/lib -L/workdir/openwrt/staging_dir/target-x86_64_musl/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/lib -Wl,-z,now -Wl,-z,relro'"  
2021-04-15T03:12:55.9515563Z Finding targets
2021-04-15T03:12:55.9760925Z go: downloading github.com/v2fly/v2ray-core/v4 v4.37.3
2021-04-15T03:12:57.3419590Z verifying github.com/v2fly/v2ray-core/v4@v4.37.3: checksum mismatch
2021-04-15T03:12:57.3421350Z 	downloaded: h1:FGmf5rtUo3xGHyZGJi8jEa7AKin05ROfPtegmGbkCKg=
2021-04-15T03:12:57.3422836Z 	go.sum:     h1:Kf9Qot2lLYccEzQfoCdt9img6zo5Vf+PEaoTMfa2S8k=
2021-04-15T03:12:57.3424297Z 
2021-04-15T03:12:57.3425108Z SECURITY ERROR
2021-04-15T03:12:57.3425903Z This download does NOT match an earlier download recorded in go.sum.
2021-04-15T03:12:57.3426625Z The bits may have been replaced on the origin server, or an attacker may
2021-04-15T03:12:57.3427822Z have intercepted the download attempt.
2021-04-15T03:12:57.3428334Z 
2021-04-15T03:12:57.3429228Z For more information, see 'go help module-auth'.
2021-04-15T03:12:57.3445046Z 
2021-04-15T03:12:57.3445457Z Building targets
2021-04-15T03:12:57.3695707Z no Go files in /workdir/openwrt/build_dir/target-x86_64_musl/v2ray-plugin-4.37.3/.go_work/build
2021-04-15T03:12:57.3706844Z 
2021-04-15T03:12:57.4749135Z make[3]: *** [Makefile:74: /workdir/openwrt/build_dir/target-x86_64_musl/v2ray-plugin-4.37.3/.built] Error 1
2021-04-15T03:12:57.4751751Z make[3]: Leaving directory '/workdir/openwrt/package/lean/v2ray-plugin'
2021-04-15T03:12:57.4759678Z make[2]: *** [package/Makefile:113: package/lean/v2ray-plugin/compile] Error 2
2021-04-15T03:12:57.4760613Z time: package/lean/v2ray-plugin/compile#0.35#0.17#1.69
2021-04-15T03:12:57.4768085Z make[2]: Leaving directory '/workdir/openwrt'
2021-04-15T03:12:57.4799741Z make[1]: *** [package/Makefile:107: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
2021-04-15T03:12:57.4800715Z make[1]: Leaving directory '/workdir/openwrt'
2021-04-15T03:12:57.4804861Z make: *** [/workdir/openwrt/include/toplevel.mk:224: world] Error 2
2021-04-15T03:12:57.4899407Z ##[error]Process completed with exit code 2.

Beginner-Go, 2021-04-15:
上游问题 等修复

Dimitri2020007, 2021-04-15:
知道了

txtfqy, 2021-04-15:
应该是https://github.com/teddysun/v2ray-plugin/blob/master/go.sum没有更新造成的问题

teasiu, 2021-04-15:
同问。

skaterjuice, 2021-04-16:
同问。

ma20at, 2021-04-16:
我把这个删了就可以编译了

skaterjuice, 2021-04-16:
> I can compile by deleting this
删除什么？ 我可以不使用Passwall进行编译

JohyC, 2021-04-17:
+1!!! 目前也是这个问题。

closed 2021-04-17 by Dimitri2020007