coolsnowwolf/lede #6698: 频繁出现拨号断开重播，老哥们有遇到过吗

ostrichii, 2021-04-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
用了昨天的最新代码，一早上起来发现DDNS记录更新了好几次，看日志发现拨号断开重播了好几次
Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
用了昨天的最新代码，一早上起来发现DDNS记录更新了好几次，看日志发现拨号断开重播了好几次，不知道是不是和PT或者大量的流量有关
### (1) 具体问题
看下面日志，iptv相关的行可跳过
A：
拨号被断开
### (2) 路由器型号和固件版本
Powered by LuCI Master (git-21.096.35788-b200c7b) / OpenWrt R21.3.27
A：

### (3) 详细日志

A：
Fri Apr 16 05:42:36 2021 daemon.info pppd[5734]: LCP terminated by peer
Fri Apr 16 05:42:36 2021 daemon.info pppd[5734]: Connect time 1368.1 minutes.
Fri Apr 16 05:42:36 2021 daemon.info pppd[5734]: Sent 2496571566 bytes, received 33430112 bytes.
Fri Apr 16 05:42:36 2021 daemon.notice netifd: Network device 'pppoe-wan' link is down
Fri Apr 16 05:42:37 2021 daemon.notice pppd[5734]: Modem hangup
Fri Apr 16 05:42:37 2021 daemon.notice pppd[5734]: Connection terminated.
Fri Apr 16 05:42:37 2021 daemon.notice netifd: Interface 'wan' has lost the connection
Fri Apr 16 05:42:37 2021 daemon.info pppd[5734]: Sent PADT
Fri Apr 16 05:42:37 2021 daemon.info pppd[5734]: Exit.
Fri Apr 16 05:42:37 2021 daemon.notice netifd: Interface 'wan' is now down
Fri Apr 16 05:42:37 2021 daemon.notice netifd: Interface 'wan' is disabled
Fri Apr 16 05:42:37 2021 daemon.notice netifd: Interface 'wan' is enabled
Fri Apr 16 05:42:37 2021 daemon.notice netifd: Interface 'wan' is setting up now
Fri Apr 16 05:42:37 2021 daemon.notice netifd: Network device 'eth1' link is down
Fri Apr 16 05:42:37 2021 daemon.notice netifd: Interface 'wan' has link connectivity loss
Fri Apr 16 05:42:37 2021 kern.info kernel: [82102.080184] 8021q: adding VLAN 0 to HW filter on device eth1
Fri Apr 16 05:42:37 2021 daemon.err insmod: module is already loaded - slhc
Fri Apr 16 05:42:37 2021 daemon.err insmod: module is already loaded - ppp_generic
Fri Apr 16 05:42:37 2021 daemon.err insmod: module is already loaded - pppox
Fri Apr 16 05:42:37 2021 daemon.err insmod: module is already loaded - pppoe
Fri Apr 16 05:42:37 2021 daemon.notice netifd: wan (16228): uci: Entry not found
Fri Apr 16 05:42:37 2021 daemon.notice netifd: wan (16228): sh: out of range
Fri Apr 16 05:42:37 2021 daemon.notice netifd: wan (16228): Command failed: Permission denied
Fri Apr 16 05:42:37 2021 daemon.notice netifd: Interface 'wan' is now down
Fri Apr 16 05:42:37 2021 daemon.notice netifd: Interface 'wan' is disabled
Fri Apr 16 05:42:37 2021 kern.info kernel: [82102.208553] 8021q: adding VLAN 0 to HW filter on device eth1
Fri Apr 16 05:42:37 2021 daemon.notice netifd: Interface 'wan' is enabled
Fri Apr 16 05:42:40 2021 daemon.notice netifd: iptv (5693): udhcpc: sending renew to 10.206.64.1
Fri Apr 16 05:42:40 2021 daemon.notice netifd: iptv (5693): udhcpc: lease of 10.206.114.26 obtained, lease time 600
Fri Apr 16 05:42:40 2021 kern.info kernel: [82105.069467] igb 0000:02:00.0 eth1: igb: eth1 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
Fri Apr 16 05:42:40 2021 daemon.notice netifd: Network device 'eth1' link is up
Fri Apr 16 05:42:40 2021 daemon.notice netifd: Interface 'wan' has link connectivity
Fri Apr 16 05:42:40 2021 daemon.notice netifd: Interface 'wan' is setting up now
Fri Apr 16 05:42:40 2021 daemon.err insmod: module is already loaded - slhc
Fri Apr 16 05:42:40 2021 daemon.err insmod: module is already loaded - ppp_generic
Fri Apr 16 05:42:40 2021 daemon.err insmod: module is already loaded - pppox
Fri Apr 16 05:42:40 2021 daemon.err insmod: module is already loaded - pppoe
Fri Apr 16 05:42:40 2021 daemon.notice netifd: wan (16416): uci: Entry not found
Fri Apr 16 05:42:40 2021 daemon.notice netifd: wan (16416): sh: out of range
Fri Apr 16 05:42:40 2021 daemon.info pppd[16426]: Plugin rp-pppoe.so loaded.
Fri Apr 16 05:42:40 2021 daemon.info pppd[16426]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Fri Apr 16 05:42:40 2021 daemon.notice pppd[16426]: pppd 2.4.8 started by root, uid 0
Fri Apr 16 05:42:45 2021 daemon.info pppd[16426]: PPP session is 661
Fri Apr 16 05:42:45 2021 daemon.warn pppd[16426]: Connected to 10:1b:54:81:f8:37 via interface eth1
Fri Apr 16 05:42:45 2021 kern.info kernel: [82110.487029] pppoe-wan: renamed from ppp0
Fri Apr 16 05:42:45 2021 daemon.info pppd[16426]: Renamed interface ppp0 to pppoe-wan
Fri Apr 16 05:42:45 2021 daemon.info pppd[16426]: Using interface pppoe-wan
Fri Apr 16 05:42:45 2021 daemon.notice pppd[16426]: Connect: pppoe-wan <--> eth1
Fri Apr 16 05:42:48 2021 daemon.info pppd[16426]: syncppp not active
Fri Apr 16 05:42:48 2021 daemon.info pppd[16426]: Remote message: Authentication success,Welcome!
Fri Apr 16 05:42:48 2021 daemon.notice pppd[16426]: PAP authentication succeeded
Fri Apr 16 05:42:48 2021 daemon.notice pppd[16426]: peer from calling number 10:1B:54:81:F8:37 authorized
Fri Apr 16 05:42:48 2021 daemon.notice pppd[16426]: local  IP address 115.194.9.0
Fri Apr 16 05:42:48 2021 daemon.notice pppd[16426]: remote IP address 115.194.8.1
Fri Apr 16 05:42:48 2021 daemon.notice pppd[16426]: primary   DNS address 202.101.172.35
Fri Apr 16 05:42:48 2021 daemon.notice pppd[16426]: secondary DNS address 202.101.172.47
Fri Apr 16 05:42:48 2021 daemon.notice netifd: Network device 'pppoe-wan' link is up
Fri Apr 16 05:42:48 2021 daemon.notice netifd: Interface 'wan' is now up
Fri Apr 16 05:42:48 2021 daemon.info netdata[7680]: RRDSET: chart name 'net.pppoe_wan' on host 'IXP' already exists.
Fri Apr 16 05:42:48 2021 daemon.info netdata[7680]: RRDSET: chart name 'net_packets.pppoe_wan' on host 'IXP' already exists.
Fri Apr 16 05:42:59 2021 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Fri Apr 16 05:43:00 2021 cron.err crond[16935]: crond (busybox 1.31.1) started, log level 8
Fri Apr 16 05:43:00 2021 cron.err crond[17116]: crond (busybox 1.31.1) started, log level 8
Fri Apr 16 05:43:01 2021 user.notice ddns-scripts[17367]: myddns_ipv4: PID '17367' started at 2021-04-16 05:43
Fri Apr 16 05:43:02 2021 user.warn ddns-scripts[17367]: myddns_ipv4: Service section disabled! - TERMINATE
Fri Apr 16 05:43:03 2021 user.warn ddns-scripts[17367]: myddns_ipv4: PID '17367' exit WITH ERROR '1' at 2021-04-16 05:43
Fri Apr 16 05:45:28 2021 daemon.info netdata[7680]: RRDSET: chart name 'net_drops.pppoe_wan' on host 'IXP' already exists.
Fri Apr 16 05:47:40 2021 daemon.notice netifd: iptv (5693): udhcpc: sending renew to 10.206.64.1

xydche, 2021-04-16:
我是三月底编译的，今天也遇到一次，不知道是不是电信的问题

xydche, 2021-04-20:
破案了，我的是因为光猫电源坏了，换了个12V电源就好了，困扰了一个星期。

ostrichii, 2021-04-20:
> 破案了，我的是因为光猫电源坏了，换了个12V电源就好了，困扰了一个星期。

有可能，我试试去