coolsnowwolf/lede #6710: 最新固件编译报错

dlj390, 2021-04-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-ssr-plus:
 * 	v2ray-plugin
 * opkg_install_cmd: Cannot install package luci-app-ssr-plus.
make[2]: *** [package/Makefile:68: package/install] Error 255
make[2]: Leaving directory '/home/djl390/lede'
make[1]: *** [package/Makefile:108: /home/djl390/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/djl390/lede'
make: *** [/home/djl390/lede/include/toplevel.mk:224：world] 错误 2
### (1) 具体问题

A：编译报错
### (2) 路由器型号和固件版本

A：Intel X86

### (3) 详细日志

A：Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-ssr-plus:
 * 	v2ray-plugin
 * opkg_install_cmd: Cannot install package luci-app-ssr-plus.
make[2]: *** [package/Makefile:68: package/install] Error 255
make[2]: Leaving directory '/home/djl390/lede'
make[1]: *** [package/Makefile:108: /home/djl390/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/djl390/lede'
make: *** [/home/djl390/lede/include/toplevel.mk:224：world] 错误 2

EnnawYang, 2021-04-18:
`./scripts/feeds clean && ./scripts/feeds update -a && ./scripts/feeds install -a`

GitGuilin, 2021-04-18:
学习学习 帮顶 我这边也是同样的问题 显示ssr缺少v2ray-plugin的依赖

dlj390, 2021-04-18:
> `./scripts/feeds clean && ./scripts/feeds update -a && ./scripts/feeds install -a`

not working

GitGuilin, 2021-04-18:
我之前清华源 刚换到官方源 更新了一下 8个软件包 就好了

starxu1971, 2021-04-18:
原作者那里删除了

ricksuzade-maker, 2021-04-18:
除了全部删除重新pull外，还有没有别的办法解决？

starxu1971, 2021-04-18:
https://github.com/jwchenzju/teddysunss/v2ray-plugin/tar.gz/v$(PKG_VERSION)?
 Makefile文件那行地址改成上面那个这是人家保存的库

starxu1971, 2021-04-18:
不一定能行我也在试

starxu1971, 2021-04-18:
如果不信只能取消v2ray-plugin了，因为现在用v2ray的真不多

starxu1971, 2021-04-18:
作者已经改成xray-plugin，v2ray-plugin已经没了

EnnawYang, 2021-04-18:
> 作者已经改成xray-plugin，v2ray-plugin已经没了

哦？没了？

starxu1971, 2021-04-18:
我用原版代码编译完成没有出现上面的问题。但是./scripts/feeds update -a && ./scripts/feeds install -a时的错误提示v2ray-plugin不存在。

alepennyq, 2021-04-18:
> `./scripts/feeds clean && ./scripts/feeds update -a && ./scripts/feeds install -a`

这个命令可以啊

loydcheung, 2021-04-18:
打开*/lede/feeds.conf.default文件最后一行hello world前面的#去掉，然后./scripts/feeds update -a && ./scripts/feeds install -a就行了

kissyouhunter, 2021-04-18:
> > `./scripts/feeds clean && ./scripts/feeds update -a && ./scripts/feeds install -a`
> 
> 这个命令可以啊
试了
不行
取消编译可以成功
但是有提示看着闹心

openwrt-try, 2021-04-19:
WARNING: Makefile 'package/feeds/passwall/luci-app-passwall/Makefile' has a dependency on 'v2ray-plugin', which does not exist
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'v2ray-plugin', which does not exist

同问，怎么把这个问题消灭了，

EnnawYang, 2021-04-19:
- 修改 `feeds.conf.default` ，删除helloworld行最前面的注释，保存  
- `./scripts/feeds update -a && ./scripts/feeds install -a`

ziyangdaoren, 2021-04-19:
> WARNING: Makefile 'package/feeds/passwall/luci-app-passwall/Makefile' has a dependency on 'v2ray-plugin', which does not exist
> WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'v2ray-plugin', which does not exist
> 
> 同问，怎么把这个问题消灭了，

把package/feeds/passwall/luci-app-passwall/Makefile中的v2ray-plugin那一行注释掉，就不提示了。

openwrt-try, 2021-04-21:
解决了，先注释掉，
  +PACKAGE_$(PKG_NAME)_INCLUDE_V2ray-plugin:v2ray-plugin \
执行命令，
./scripts/feeds update -a && ./scripts/feeds install -a
再取消注释，新执行命令，

shu0802, 2021-05-11:
> +PACKAGE_$(PKG_NAME)_INCLUDE_V2ray-plugin:v2ray-plugin

在哪里注释

shu0802, 2021-05-15:
重新拉取lede源后编译，可以编译成功了。可以关闭了。

closed 2021-06-17 by coolsnowwolf