coolsnowwolf/lede #6718: Argument list too long编译报错，怎么解决

galaxyskyknight, 2021-04-18:
make[2]: execvp: /usr/bin/env: Argument list too long
make[2]: *** [package/Makefile:68: package/install] Error 127
make[2]: Leaving directory '/home/xxx/lede'
make[1]: *** [package/Makefile:108: /home/xxx/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/xxx/lede'
make: *** [/home/xxx/lede/include/toplevel.mk:224：world] 错误 2

网上查了一下应该是命令带的参数过长，这种不知道怎么解决了，有说unlimit -s 65535的，试了没用，然后我减少了luci-app的包个数，也还是没用，到底怎么解决，谁能给个有效方案？

Dimitri2020007, 2021-04-18:
实体机编译非常容易出现这个问题，而且一定会出现在编译好所有的模块和luci-app之后执行make install这些的时候出现的错误。

即使按照readme上的提示安装了软件包还是容易出现这个错误，只好转战github action编译了

使用的hyper-v ubuntu 18.04服务器版本

make -j1 V=s 虚拟机编译的时间比其他人说的慢不少，我也不知道是咋回事。github action编译是6小时左右完成。

改ulimit和减少包个数对于这个问题是没用的。

galaxyskyknight, 2021-04-19:
> 实体机编译非常容易出现这个问题，而且一定会出现在编译好所有的模块和luci-app之后执行make install这些的时候出现的错误。
> 
> 即使按照readme上的提示安装了软件包还是容易出现这个错误，只好转战github action编译了
> 
> 使用的hyper-v 第2代虚拟机虚拟的ubuntu 18.04服务器版本
> 
> make -j1 V=s 虚拟机编译的时间比其他人说的慢不少，我也不知道是咋回事。github action编译是6小时左右完成。
> 
> 改ulimit和减少包个数对于这个问题是没用的。

关键是我在vmware的虚拟机里编译的，兄dei

galaxyskyknight, 2021-04-19:
求大佬改下makefile看能规避这个问题不，makefile太复杂了，不熟的话很难定位在哪里做workaround

Dimitri2020007, 2021-04-19:
> > 实体机编译非常容易出现这个问题，而且一定会出现在编译好所有的模块和luci-app之后执行make install这些的时候出现的错误。
> > 即使按照readme上的提示安装了软件包还是容易出现这个错误，只好转战github action编译了
> > 使用的hyper-v 第2代虚拟机虚拟的ubuntu 18.04服务器版本
> > make -j1 V=s 虚拟机编译的时间比其他人说的慢不少，我也不知道是咋回事。github action编译是6小时左右完成。
> > 改ulimit和减少包个数对于这个问题是没用的。
> 
> 关键是我在vmware的虚拟机里编译的，兄dei

vmware我也试过 也会出这个错误

1715173329, 2021-04-21:
See openwrt/openwrt#4065.

Dimitri2020007, 2021-05-10:
> See [openwrt/openwrt#4065](https://github.com/openwrt/openwrt/pull/4065).

按照他的提示我今天测试了一下那段代码

修改package/makefile文件

替换原内容

	$(call opkg,$(TARGET_DIR)) install \
		$(call opkg_package_files,$(foreach pkg,$(shell cat $(PACKAGE_INSTALL_FILES) 2>/dev/null),$(pkg)$(call GetABISuffix,$(pkg))))

为下面这个（就在69行）

	$(file >$(TMP_DIR)/opkg_install_list,\
	  $(call opkg_package_files,\
	    $(foreach pkg,$(shell cat $(PACKAGE_INSTALL_FILES) 2>/dev/null),$(pkg)$(call GetABISuffix,$(pkg)))))
	$(call opkg,$(TARGET_DIR)) install $$(cat $(TMP_DIR)/opkg_install_list)

测试编译840个openwrt软件包，使用openwrt-21.02分支测试没有再出现Argument list too long的错误。

lede也可以尝试这样改下看看效果。

Dimitri2020007, 2021-05-12:
openwrt今天已经更新源码修复了Argument list too long的错误了

Dimitri2020007, 2021-05-20:
> See [openwrt/openwrt#4065](https://github.com/openwrt/openwrt/pull/4065).

修改package/Makefile文件

ctrl+f查找这个$(curdir)/merge-index)

定位到68行

将内容从原来的

	$(call opkg,$(TARGET_DIR)) install \		$(call opkg_package_files,$(foreach pkg,$(shell cat $(PACKAGE_INSTALL_FILES) 2&gt;/dev/null),$(pkg)$(call GetABISuffix,$(pkg))))

替换为

	$(file &gt;$(TMP_DIR)/opkg_install_list,\	  $(call opkg_package_files,\	    $(foreach pkg,$(shell cat $(PACKAGE_INSTALL_FILES) 2&gt;/dev/null),$(pkg)$(call GetABISuffix,$(pkg)))))	$(call opkg,$(TARGET_DIR)) install $$(cat $(TMP_DIR)/opkg_install_list)

我测试安装了840个软件包

测试者编译了916个软件包

使用修改后的package/makefile文件执行make install安装操作后没有再出现Argument list too long的错误

closed 2021-06-17 by coolsnowwolf