coolsnowwolf/lede #6721: 最新版本里面系统设置找不到zram 的设置了？

hxlls, 2021-04-18:
编译了个AC58U的固件，用的最新源码，因为要用XRAY 但是内存不足，经常导致跑VMESS 时候会退出，所以我把把zram 编译进去了，但是好像在系统设置里面没找到zram 的设置了？默认也是zram 没有启动，需要手动启动
[image]
百度了下，看到大雕之前的介绍，早点的版本有
[image]
 请问怎么添加进去？

Beginner-Go, 2021-04-20:
Fix https://github.com/coolsnowwolf/luci/commit/2c780ccfc56182366a40fd961223ef25f5342862

coolsnowwolf, 2021-04-20:
已经添加了 ZRAM 压缩内存支持，请更新 feeds 重新编译

hxlls, 2021-04-20:
测试在X86构架上，ZRAM 在新版中无法正常启动，luci 已经支持
相关系统日志如下
Tue Apr 20 23:28:37 2021 daemon.debug zram_start: activating '/dev/zram0' for swapping (128 MegaBytes)
Tue Apr 20 23:28:37 2021 daemon.err uhttpd[4209]: zram_start: activating '/dev/zram0' for swapping (128 MegaBytes)
Tue Apr 20 23:28:37 2021 daemon.debug zram_reset: enforcing defaults via /sys/block/zram0/reset
Tue Apr 20 23:28:37 2021 daemon.err uhttpd[4209]: zram_reset: enforcing defaults via /sys/block/zram0/reset
Tue Apr 20 23:28:37 2021 daemon.debug zram_comp_algo: Compression algorithm 'lz4' is not supported for '/dev/zram0'
Tue Apr 20 23:28:37 2021 daemon.err uhttpd[4209]: zram_comp_algo: Compression algorithm 'lz4' is not supported for '/dev/zram0'
Tue Apr 20 23:28:37 2021 daemon.debug zram_comp_streams: Set max compression streams to '1' for zram '/dev/zram0'
Tue Apr 20 23:28:37 2021 daemon.err uhttpd[4209]: zram_comp_streams: Set max compression streams to '1' for zram '/dev/zram0'
Tue Apr 20 23:28:37 2021 kern.err kernel: [  756.334585] Can't allocate a compression stream
Tue Apr 20 23:28:37 2021 kern.err kernel: [  756.340132] zram: Cannot initialise lzo-rle compressing backend
Tue Apr 20 23:28:37 2021 daemon.err uhttpd[4209]: sh: write error: Out of memory
Tue Apr 20 23:28:37 2021 daemon.err uhttpd[4209]: mkswap: image is too small
Tue Apr 20 23:28:37 2021 daemon.err uhttpd[4209]: swapon: /dev/zram0: Invalid argument
Tue Apr 20 23:28:37 2021 kern.err kernel: [  756.365106] Unable to find swap-space signature
相关内核日志如下
[  222.045314] Can't allocate a compression stream
[  222.050825] zram: Cannot initialise lzo-rle compressing backend
[  222.079680] Unable to find swap-space signature
[  756.334585] Can't allocate a compression stream
[  756.340132] zram: Cannot initialise lzo-rle compressing backend
[  756.365106] Unable to find swap-space signature

该问题貌似只出现在新版的5.x 内核中，因为之前AC58U我也用LEDE源码编译了一次，发现也是重启后需要手动启动ZRAM ，但是在4.14内核下，这个问题不存在，以下为未加入luci 更新时候的截图

[image]

PS 有个题外的问题
X86构架上ps3netsrv 无法正常工作，系统日志如下
Tue Apr 20 23:36:36 2021 daemon.info ps3netsrv: Instance "main" disabled.
Tue Apr 20 23:36:41 2021 daemon.info ps3netsrv: ps3netsrv will run as user 'root'.
Tue Apr 20 23:36:42 2021 kern.info kernel: [ 1241.128565] ps3netsrv[5152]: segfault at 400040 ip 0000000000418fbf sp 00007ffdd248a270 error 4 in ps3netsrv[401000+1e000]
Tue Apr 20 23:36:42 2021 kern.info kernel: [ 1241.141807] Code: 44 8b 05 b0 e2 00 00 45 31 d2 31 c9 55 31 d2 bd 00 00 80 00 53 4c 8b 5f 18 bb 00 00 00 00 4c 8b 4f 28 4c 89 d8 4d 85 c9 74 59 <8b> 30 83 fe 06 75 05 4c 89 d9 eb 14 83 fe 02 74 07 83 fe 07 74 37
Tue Apr 20 23:36:47 2021 kern.info kernel: [ 1246.169493] ps3netsrv[5225]: segfault at 400040 ip 0000000000418fbf sp 00007ffff5f1ac50 error 4 in ps3netsrv[401000+1e000]
Tue Apr 20 23:36:47 2021 kern.info kernel: [ 1246.182768] Code: 44 8b 05 b0 e2 00 00 45 31 d2 31 c9 55 31 d2 bd 00 00 80 00 53 4c 8b 5f 18 bb 00 00 00 00 4c 8b 4f 28 4c 89 d8 4d 85 c9 74 59 <8b> 30 83 fe 06 75 05 4c 89 d9 eb 14 83 fe 02 74 07 83 fe 07 74 37

内核日志如下
[ 1241.128565] ps3netsrv[5152]: segfault at 400040 ip 0000000000418fbf sp 00007ffdd248a270 error 4 in ps3netsrv[401000+1e000]
[ 1241.141807] Code: 44 8b 05 b0 e2 00 00 45 31 d2 31 c9 55 31 d2 bd 00 00 80 00 53 4c 8b 5f 18 bb 00 00 00 00 4c 8b 4f 28 4c 89 d8 4d 85 c9 74 59 <8b> 30 83 fe 06 75 05 4c 89 d9 eb 14 83 fe 02 74 07 83 fe 07 74 37
[ 1246.169493] ps3netsrv[5225]: segfault at 400040 ip 0000000000418fbf sp 00007ffff5f1ac50 error 4 in ps3netsrv[401000+1e000]
[ 1246.182768] Code: 44 8b 05 b0 e2 00 00 45 31 d2 31 c9 55 31 d2 bd 00 00 80 00 53 4c 8b 5f 18 bb 00 00 00 00 4c 8b 4f 28 4c 89 d8 4d 85 c9 74 59 <8b> 30 83 fe 06 75 05 4c 89 d9 eb 14 83 fe 02 74 07 83 fe 07 74 37
[ 1251.209667] ps3netsrv[5266]: segfault at 400040 ip 0000000000418fbf sp 00007ffc0282a8e0 error 4 in ps3netsrv[401000+1e000]
[ 1251.222995] Code: 44 8b 05 b0 e2 00 00 45 31 d2 31 c9 55 31 d2 bd 00 00 80 00 53 4c 8b 5f 18 bb 00 00 00 00 4c 8b 4f 28 4c 89 d8 4d 85 c9 74 59 <8b> 30 83 fe 06 75 05 4c 89 d9 eb 14 83 fe 02 74 07 83 fe 07 74 37
[ 1256.251856] ps3netsrv[5476]: segfault at 400040 ip 0000000000418fbf sp 00007ffe73a74a40 error 4 in ps3netsrv[401000+1e000]
[ 1256.265123] Code: 44 8b 05 b0 e2 00 00 45 31 d2 31 c9 55 31 d2 bd 00 00 80 00 53 4c 8b 5f 18 bb 00 00 00 00 4c 8b 4f 28 4c 89 d8 4d 85 c9 74 59 <8b> 30 83 fe 06 75 05 4c 89 d9 eb 14 83 fe 02 74 07 83 fe 07 74 37
[ 1261.293338] ps3netsrv[7044]: segfault at 400040 ip 0000000000418fbf sp 00007fff220d8b90 error 4 in ps3netsrv[401000+1e000]
[ 1261.306617] Code: 44 8b 05 b0 e2 00 00 45 31 d2 31 c9 55 31 d2 bd 00 00 80 00 53 4c 8b 5f 18 bb 00 00 00 00 4c 8b 4f 28 4c 89 d8 4d 85 c9 74 59 <8b> 30 83 fe 06 75 05 4c 89 d9 eb 14 83 fe 02 74 07 83 fe 07 74 37
[ 1266.333289] ps3netsrv[7070]: segfault at 400040 ip 0000000000418fbf sp 00007ffe1e37ea40 error 4 in ps3netsrv[401000+1e000]
[ 1266.346592] Code: 44 8b 05 b0 e2 00 00 45 31 d2 31 c9 55 31 d2 bd 00 00 80 00 53 4c 8b 5f 18 bb 00 00 00 00 4c 8b 4f 28 4c 89 d8 4d 85 c9 74 59 <8b> 30 83 fe 06 75 05 4c 89 d9 eb 14 83 fe 02 74 07 83 fe 07 74 37

closed 2021-06-17 by coolsnowwolf