coolsnowwolf/lede #6729: avahi编译错误，求救如何解决！

galaxyskyknight, 2021-04-20:
cp -fpR /home/XXX/lede/build_dir/target-x86_64_musl/avahi-dbus/avahi-0.8/ipkg-install/usr/lib/libdns_sd.so* /home/XXX/lede/build_dir/target-x86_64_musl/avahi-dbus/avahi-0.8/ipkg-x86_64/libavahi-compat-libdnssd/usr/lib/
cp: cannot stat '/home/XXX/lede/build_dir/target-x86_64_musl/avahi-dbus/avahi-0.8/ipkg-install/usr/lib/libdns_sd.so*': No such file or directory
make[3]: *** [Makefile:398: /home/XXX/lede/bin/packages/x86_64/packages/libavahi-compat-libdnssd_0.8-6_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/XXX/lede/feeds/packages/libs/avahi'
time: package/feeds/packages/avahi/dbus/compile#0.15#0.07#0.21
make[2]: *** [package/Makefile:113: package/feeds/packages/avahi/compile] Error 2
make[2]: Leaving directory '/home/XXX/lede'
make[1]: *** [package/Makefile:107: /home/XXX/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/XXX/lede'
make: *** [/home/XXX/lede/include/toplevel.mk:224：world] 错误 2

galaxyskyknight, 2021-04-20:
真的是要哭了，各种坑，随便多选一个包或者少选一个包，就error了。application里实在也没有avahi的app不知道为何又跑到avahi里说编译找不到so库。

Deerczl, 2024-07-08:
遇到了同样的问题，请问是怎么解决的呢？
```
cp -fpR /home/reval/lede/build_dir/target-x86_64_musl/avahi-dbus/avahi-0.8/ipkg-install/usr/lib/libdns_sd.so* /home/reval/lede/build_dir/target-x86_64_musl/avahi-dbus/avahi-0.8/ipkg-x86_64/libavahi-compat-libdnssd/usr/lib/
cp: cannot stat '/home/reval/lede/build_dir/target-x86_64_musl/avahi-dbus/avahi-0.8/ipkg-install/usr/lib/libdns_sd.so*': No such file or directory
Makefile:396: recipe for target '/home/reval/lede/bin/packages/x86_64/packages/libavahi-compat-libdnssd_0.8-7_x86_64.ipk' failed
make[3]: *** [/home/reval/lede/bin/packages/x86_64/packages/libavahi-compat-libdnssd_0.8-7_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/reval/lede/feeds/packages/libs/avahi'
time: package/feeds/packages/avahi/dbus/compile#0.25#0.06#0.27
    ERROR: package/feeds/packages/avahi failed to build (build variant: dbus).
package/Makefile:114: recipe for target 'package/feeds/packages/avahi/compile' failed
make[2]: *** [package/feeds/packages/avahi/compile] Error 1
make[2]: Leaving directory '/home/reval/lede'
package/Makefile:110: recipe for target '/home/reval/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/reval/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/reval/lede'
/home/reval/lede/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2
```

closed 2024-07-09 by coolsnowwolf