coolsnowwolf/lede #6731: libfido2 not exist

doumeki, 2021-04-21:
WARNING: Makefile 'package/feeds/packages/openssh/Makefile' has a dependency on 'libfido2', which does not exist
WARNING: Makefile 'package/feeds/packages/openssh/Makefile' has a build dependency on 'libfido2', which does not exist

jxph1123, 2021-04-22:
同样警告。

shedkiss, 2021-04-22:
same problem and please fix it ,thanks!

Cathgao, 2021-04-22:
just ignore it if complie return no error

Myoko, 2021-04-22:
忽略这个

shedkiss, 2021-04-22:
if you can help to fix that ?thanks

Cathgao, 2021-04-22:
> if you can help to fix that ?thanks

No need to fix if no error. It doesn't have any influence on anything

CuteTony, 2021-04-22:
[packages feed](https://github.com/coolsnowwolf/packages)已经添加 `libcbor` 和 `libfido2` 依赖，不影响。
若不放心再执行一次 `./scripts/feeds update -a && ./scripts/feeds install -a` 即可看到警告消除。

closed 2021-04-22 by doumeki