coolsnowwolf/lede #6732: 最新5.4内核k2p怎么把wifi密码提前预设到固件里呢

kuole-o, 2021-04-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
在之前4.多内核时，斐讯k2p固件里，我通过替换文件大法，mv base-files/files/etc/config/my-files/mt7615.1*.dat ./lean/mt/drivers/mt_wifi/files 来实现刷完估计，双频WiFi自动配置、自动设置我要的密码

现在5.4内核里，升级完，没找到 mt7615.1*.dat 这样的WiFi配置文件，只有/tmp/profiles 目录下有mt_dbdc*.dat 文件，好像是相关的。用之前的方法，一直没法编译前放到base-files/files/tmp/profiles 目录下，提示没有“tmp/profiles ”这样的目录（其实已经提前 mkdir创建过）

**求问，怎样不改源码方式实现WiFi配置进固件里呢？求大佬指点** （k2p需要5g设置149信道，2.4g设置11信道，国家为us才能正常使用WiFi，改源码里的.sh脚本，好像很复杂。。）

### (1) 具体问题

A：

### (2) 路由器型号和固件版本
斐讯 k2p
A：

### (3) 详细日志
[image]

A：

EnnawYang, 2021-04-22:
机器上设置好wifi后把配置备份出来，通过files大法打包进去，目录 `/etc/wireless/mt7615/`

kuole-o, 2021-04-22:
已解决！

5.4内核上，k2p把机器上 etc/config下 network wireless shadowsocksr 3个文件备份，然后file大法放到 base-files/files/etc/config 下，编译到包里就行

如果刷上宽带拨号异常，重启一次路由器即可。

相问个别的问题，5.4内核，编译完估计14MB 好大啊……是什么导致的呢  以前都7MB左右

closed 2021-04-26 by coolsnowwolf