coolsnowwolf/lede #6734: 有没有遇到过DNS莫名其妙挂掉的（晚上还好好的，一觉睡醒就挂了）

fhbyljj, 2021-04-22:
DNS问题

 前天晚上还好好的，一觉睡醒路由器DNS挂了（我设定了wan和wan6每天晚上自动重新拨号）

硬件和版本 I5 8400 ESXI    内核版本是5.4.113那个

日志： 这个错误我真不知道放那个日志，只是想问问有没有遇到一样问题的

我打开了DNS加速，然后又好了,隐约感觉是系统内置的DNS出问题??????

galaxyskyknight, 2021-04-23:
这个真帮不了你，都不知道你本地dns server用的啥daemon，有用smartdns的，有用dnsmasq的，有用dnsforwarder,有用pdnsd的，天知道你用的啥，啥原因。

fhbyljj, 2021-04-24:
> 这个真帮不了你，都不知道你本地的dns服务器用的啥daemon，有用的smartdns的，有用的dnsmasq的，有用的dnsforwarder，有用的pdnsd的，天知道你用的啥，啥原因。

已经解决了

deleisoft, 2021-04-24:
> > 这个真帮不了你，都不知道你本地的dns服务器用的啥daemon，有用的smartdns的，有用的dnsmasq的，有用的dnsforwarder，有用的pdnsd的，天知道你用的啥，啥原因。
> 
> 已经解决了

如何解决的呢？我好像也遇到过类似问题

starxu1971, 2021-04-25:
最近发生因为去广告插件用了这个加载ADgk主规则（https://github.com/banbendalao/ADgk）后，因为其中||114la.com^$third-party,domain=~ylmf.com^这条规则，会导致dnsfilter故障，使openwrt整个DNS系统无响应

closed 2021-04-25 by coolsnowwolf