coolsnowwolf/lede #6735: Package openssh missing dependency libfido2

nojd11, 2021-04-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
WARNING: Makefile 'package/feeds/packages/openssh/Makefile' has a dependency on 'libfido2', which does not exist
WARNING: Makefile 'package/feeds/packages/openssh/Makefile' has a build dependency on 'libfido2', which does not exist

jxph1123, 2021-04-22:
https://github.com/coolsnowwolf/packages/pull/146

jxph1123, 2021-04-22:
https://github.com/coolsnowwolf/lede/issues/6731

MXJNZ6, 2021-04-22:
我这应该是和你一样错误吧
checking for pkg-config... /workdir/openwrt/staging_dir/host/bin/pkg-config
10496
configure: WARNING: using cross tools not prefixed with host triplet
10497
checking if /workdir/openwrt/staging_dir/host/bin/pkg-config knows about libfido2... no
10498
checking for fido_init in -lfido2... no
10499
configure: error: no usable libfido2 found
10500
Makefile:263: recipe for target '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/openssh-without-pam/openssh-8.6p1/.configured_a708d0c3539a2f3a09217c5527a5f9e6' failed
10501
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/openssh'
10502
time: package/feeds/packages/openssh/without-pam/compile#15.31#7.81#19.93
10503
make[3]: *** [/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/openssh-without-pam/openssh-8.6p1/.configured_a708d0c3539a2f3a09217c5527a5f9e6] Error 1
10504
package/Makefile:111: recipe for target 'package/feeds/packages/openssh/compile' failed
10505
make[2]: Leaving directory '/workdir/openwrt'
10506
package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile' failed
10507
make[1]: Leaving directory '/workdir/openwrt'
10508
make[2]: *** [package/feeds/packages/openssh/compile] Error 2
10509
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
10510
make[1]: *** [/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
10511
make: *** [world] Error 2
10512
Error: Process completed with exit code 2.

closed 2021-04-24 by coolsnowwolf