coolsnowwolf/lede #6751: 编译出错

hhhfsj, 2021-04-24:
## 1.关于你要提交的问题
编译报错

## 2. 详细叙述
...failed updating 1 target...
...updated 15565 targets...
make[3]: *** [Makefile:505: /home/fsj/lede/build_dir/hostpkg/boost_1_73_0/.built] Error 1
make[3]: Leaving directory '/home/fsj/lede/feeds/packages/libs/boost'
time: package/feeds/packages/boost/host-compile#349.04#267.26#230.08
make[2]: *** [package/Makefile:111: package/feeds/packages/boost/host/compile] Error 2
make[2]: Leaving directory '/home/fsj/lede'
make[1]: *** [package/Makefile:107: /home/fsj/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/fsj/lede'
make: *** [/home/fsj/lede/include/toplevel.mk:224: world] Error 2

dodogao, 2021-04-24:
mkdir -p /home/gaoh/openwrt/dl
SHELL= flock /home/gaoh/openwrt/tmp/.linux-4.19.tar.xz.flock -c '  	/home/gaoh/openwrt/scripts/download.pl "/home/gaoh/openwrt/dl" "linux-4.19.tar.xz" "x" "" "@KERNEL/linux/kernel/v4.x"    '
Cannot find appropriate hash command, ensure the provided hash is either a MD5 or SHA256 checksum.
Makefile:104: recipe for target '/home/gaoh/openwrt/dl/linux-4.19.tar.xz' failed
make[3]: *** [/home/gaoh/openwrt/dl/linux-4.19.tar.xz] Error 255
make[3]: Leaving directory '/home/gaoh/openwrt/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#0.02#0.08#0.16
toolchain/Makefile:98: recipe for target 'toolchain/kernel-headers/compile' failed
make[2]: *** [toolchain/kernel-headers/compile] Error 2
make[2]: Leaving directory '/home/gaoh/openwrt'
toolchain/Makefile:96: recipe for target '/home/gaoh/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/stamp/.toolchain_compile' failed
make[1]: *** [/home/gaoh/openwrt/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-7.5.0_musl_eabi/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/gaoh/openwrt'
/home/gaoh/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
gaoh@ubuntu18:~/openwrt$

closed 2021-04-24 by coolsnowwolf

lihonfu, 2021-11-30:
/home/q/openwrt/dl/linux-4.19.tar.xz] Error 255
怎么解决的?