coolsnowwolf/lede #6753: ShadowSocksR Plus+   无法科学上网

ufo1cn, 2021-04-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
ShadowSocksR Plus+   无法科学上网
Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue
## 2. 详细叙述
 直接导入VPN提供商的Vmess Link,
SSRplus+运行后,能正常连接服务器端口,但访问代理的所有国外IP都Time out.
关闭SSRplus+服务,手机连接该路由器wifi后,打开手机客户端V2rayNG,导入此Vmess Link,能正常科学上网.
### (1) 具体问题
A：导入Vmess Link后,除了ssr plus+ 客户端页面主服务器选择此节点应用外,是否还需额外设置?

### (2) 路由器型号和固件版本

A：
主机型号 | NETGEAR WNDR4300
架构 | Atheros AR9344 rev 2 x 1
固件版本 | OpenWrt R21.2.1 / LuCI Master (git-21.083.60403-2f1c883)
内核版本 | 4.14.195

### (3) 详细日志

A：
2021-04-25 09:54:32: -----------end------------
2021-04-25 09:54:27: 禁止连接的域名加载完毕。
2021-04-25 09:54:27: 未启动主节点，禁止连接的域名正在加载。
2021-04-25 09:54:27: ----------start------------
2021-04-25 09:51:09: -----------end------------
2021-04-25 09:50:51: Main node:Xray 1.4.0 (Xray, Penetrates Everything.) OpenWrt (go1.16.2 linux/mips) Started!
2021-04-25 09:50:40: ----------start------------
2021-04-25 09:36:14: -----------end------------
2021-04-25 09:36:10: 禁止连接的域名加载完毕。
2021-04-25 09:36:10: 未启动主节点，禁止连接的域名正在加载。
2021-04-25 09:36:10: ----------start------------
2021-04-25 09:25:21: -----------end------------
2021-04-25 09:25:04: Main node:Xray 1.4.0 (Xray, Penetrates Everything.) OpenWrt (go1.16.2 linux/mips) Started!
2021-04-25 09:24:54: ----------start------------
2021-04-25 09:15:25: -----------end------------
2021-04-25 09:15:21: 禁止连接的域名加载完毕。
2021-04-25 09:15:21: 未启动主节点，禁止连接的域名正在加载。
2021-04-25 09:15:20: ----------start------------
2021-04-25 09:12:06: -----------end------------
2021-04-25 09:11:48: Main node:Xray 1.4.0 (Xray, Penetrates Everything.) OpenWrt (go1.16.2 linux/mips) Started!
2021-04-25 09:11:37: ----------start------------
2021-04-25 09:08:09: -----------end------------
2021-04-25 09:08:04: 禁止连接的域名加载完毕。
2021-04-25 09:08:04: 未启动主节点，禁止连接的域名正在加载。
2021-04-25 09:08:04: ----------start------------
2021-04-25 09:01:59: -----------end------------
2021-04-25 09:01:54: 禁止连接的域名加载完毕。
2021-04-25 09:01:54: 未启动主节点，禁止连接的域名正在加载。
2021-04-25 09:01:53: ----------start------------
2021-04-25 08:54:16: -----------end------------
2021-04-25 08:53:41: Main node:Xray 1.4.0 (Xray, Penetrates Everything.) OpenWrt (go1.16.2 linux/mips) Started!
2021-04-25 08:53:23: ----------start------------
2021-04-25 08:51:43: -----------end------------
2021-04-25 08:51:38: 禁止连接的域名加载完毕。
2021-04-25 08:51:38: 未启动主节点，禁止连接的域名正在加载。
2021-04-25 08:51:37: ----------start------------
2021-04-25 08:51:01: -----------end------------
2021-04-25 08:50:56: 禁止连接的域名加载完毕。
2021-04-25 08:50:56: 未启动主节点，禁止连接的域名正在加载。
2021-04-25 08:50:56: ----------start------------
2021-04-25 08:48:32: -----------end------------
2021-04-25 08:48:27: 禁止连接的域名加载完毕。
2021-04-25 08:48:27: 未启动主节点，禁止连接的域名正在加载。
2021-04-25 08:48:27: ----------start------------

系统日志如下:
[... 511 lines cut ...]
Sun Apr 25 09:29:18 2021 daemon.info hostapd: wlan1: STA d0:c0:bf:54:ec:1c IEEE 802.11: disassociated
Sun Apr 25 09:29:19 2021 daemon.info hostapd: wlan1: STA d0:c0:bf:54:ec:1c IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sun Apr 25 09:29:28 2021 authpriv.info dropbear[19097]: Child connection from 192.168.0.201:46234
Sun Apr 25 09:32:35 2021 kern.info kernel: [ 4894.324335] sh (19250): drop_caches: 3
Sun Apr 25 09:33:07 2021 authpriv.info dropbear[5669]: Exit (root) from <192.168.0.201:1937>: Error reading: Connection reset by peer
Sun Apr 25 09:34:15 2021 authpriv.info dropbear[19077]: Exit before auth from <192.168.0.200:46109>: Timeout before auth
Sun Apr 25 09:34:28 2021 authpriv.info dropbear[19097]: Exit before auth from <192.168.0.201:46234>: (user 'root', 0 fails): Timeout before auth
Sun Apr 25 09:36:06 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Apr 25 09:36:06 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Apr 25 09:36:10 2021 daemon.info dnsmasq[19700]: Connected to system UBus
Sun Apr 25 09:36:11 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Apr 25 09:36:11 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Apr 25 09:36:14 2021 daemon.info dnsmasq[19855]: Connected to system UBus
Sun Apr 25 09:50:36 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Apr 25 09:50:36 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Apr 25 09:50:40 2021 daemon.info dnsmasq[21155]: Connected to system UBus
Sun Apr 25 09:51:05 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Apr 25 09:51:05 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Apr 25 09:51:10 2021 daemon.info dnsmasq[21639]: Connected to system UBus
Sun Apr 25 09:54:23 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Apr 25 09:54:23 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Apr 25 09:54:26 2021 daemon.info dnsmasq[22201]: Connected to system UBus
Sun Apr 25 09:54:28 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sun Apr 25 09:54:28 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sun Apr 25 09:54:32 2021 daemon.info dnsmasq[22356]: Connected to system UBus

programmejava, 2021-04-25:
大概的原因是你这个破路由器性能太差劲了   我的gl-inet  ar300m  qca9531比你的还高50hz   我强行安装ssr plus后用trojan卡到炸

libinanshu, 2021-04-25:
1，这个机器跑:Xray确实有点费劲  2，你先更新到最新源码编译固件再尝试一下？ 你这玩意儿几个月都没更新了。

closed 2021-04-26 by coolsnowwolf

KnowledegAndAction, 2022-05-11:
> 大概的原因是你这个破路由器性能太差劲了 我的gl-inet ar300m qca9531比你的还高50hz 我强行安装ssr plus后用trojan卡到炸
我也有这个AR300M，也是用的glinet官方openwrt固件塞进去的ssrp，请问能不能直接用lean的来编译？我好像没找到适配这个型号。