coolsnowwolf/lede #6758: 能否加入微信推送的源

galaxyskyknight, 2021-04-26:
自己本地编译发现没有微信推送的ipk,然后那个argon的主题设置luci-app也没有，编译菜单里luci->application找不到，自己也不知道怎么包含，能否老大把这两个加默认包含？或者谁能指导一下具体步骤，我如何把这两个东东加到本地编译里。微信推送那个作者的git库放到feeds.defualt.conf里引用，编译好像报错。argon的config工具，就没看到哪里有代码。

qianxu2001, 2021-04-26:
在 package 目录 git clone 插件仓库就行了

galaxyskyknight, 2021-04-26:
> 在 package 目录 git clone 插件仓库就行了

make menuconfig里选不到的也能自动识别编译进去吗？还是要做什么配置？

EnnawYang, 2021-04-26:
建议网上(指搜索引擎)多搜搜相关教程，很多。

jimgo188, 2021-04-26:
自己搜 luci-app-serverchan 加进去就可以了

jxph1123, 2021-04-27:
> 自己搜 luci-app-serverchan 加进去就可以了

我是手动复制到package\lean目录下的，能否有什么办法也自动检测更新呢，谢谢。

galaxyskyknight, 2021-04-27:
> 在 package 目录 git clone 插件仓库就行了

在package/lean下git clone了代码，但是做make后，安装完Image,并没有发现系统里有该插件，在编译后的package产物里似乎也没有改插件的ipk包啊，似乎直接略过了？需要make clean吗

AmadeusGhost, 2021-04-27:
`git clone` 完你要 `make menuconfig` 选上这个包啊

galaxyskyknight, 2021-04-27:
> `git clone` 完你要 `make menuconfig` 选上这个包啊

兄弟，我现在的问题就是git clone完，make menuconfig里没这个包啊 luci->application里就没这个玩意，没看到有luci-app-serverchan这个！！

Dimitri2020007, 2021-04-27:
可以fo一下lean的luci源码然后把luci-app-serverchan放进luci/application目录里，接着修改下feeds.conf.default文件。

将luci feeds链接改成你fo的源

最后git pull、script/feeds update和install就行了。

kleisth, 2021-05-27:
feeds.conf.default文件加入源
这是别人收集统合的包，里面含了挺多常用插件，也包括你要的serverchan

 src-git kenzo https://github.com/kenzok8/openwrt-packages

closed 2021-06-17 by coolsnowwolf