coolsnowwolf/lede #6775: php+nginx无法解析php

yaott2020-666, 2021-04-28:
openwrt arm64 安装nginx php7-fpm 配置后无法解析 *.php 提示：

FastCGI sent in stderr: "Unable to open primary script : /mnt/sdb/www/file/test.php (No such file or directory)" while reading response header from upstream, client: 127.0.0.1, server: file, request: "GET /test.php HTTP/1.1", upstream: "fastcgi://127.0.0.1:9000", host: "localhost:8080"

/mnt/sdb/www/file/test.php 文件存在且赋权0777 root所有   

nginx php7-fpm均正常工作没有报错 .html 正常

php.ini:
################################
[PHP]
zend.ze1_compatibility_mode = Off

; Language Options

engine = On
;short_open_tag = Off
precision    =  12
y2k_compliance = On
output_buffering = Off
;output_handler =
zlib.output_compression = Off
;zlib.output_compression_level = -1
;zlib.output_handler =
implicit_flush = Off
unserialize_callback_func =
serialize_precision = 100

;open_basedir =
disable_functions =
disable_classes =

; Colors for Syntax Highlighting mode.  Anything that's acceptable in
; <span style="color: ???????"> would work.
;highlight.string  = #DD0000
;highlight.comment = #FF9900
;highlight.keyword = #007700
;highlight.bg      = #FFFFFF
;highlight.default = #0000BB
;highlight.html    = #000000

;ignore_user_abort = On
;realpath_cache_size = 16k
;realpath_cache_ttl = 120

; Miscellaneous

expose_php = On

; Resource Limits

max_execution_time = 30 ; Maximum execution time of each script, in seconds.
max_input_time = 60     ; Maximum amount of time each script may spend parsing r                                                                                                             equest data.
;max_input_nesting_level = 64
memory_limit = 8M       ; Maximum amount of memory a script may consume.

; Error handling and logging

; Error Level Constants:
; E_ALL             - All errors and warnings (includes E_STRICT as of PHP 6.0.0                                                                                                             )
; E_ERROR           - fatal run-time errors
; E_RECOVERABLE_ERROR  - almost fatal run-time errors
; E_WARNING         - run-time warnings (non-fatal errors)
; E_PARSE           - compile-time parse errors
; E_NOTICE          - run-time notices (these are warnings which often result
;                     from a bug in your code, but it's possible that it was
;                     intentional (e.g., using an uninitialized variable and
;                     relying on the fact it's automatically initialized to an
;                     empty string)
; E_STRICT                      - run-time notices, enable to have PHP suggest c                                                                                                             hanges
[... 653 lines cut ...]
################################
server {
        listen       8080;
        server_name  file;

        access_log /tmp/log/nginx/file.access.log;
        error_log /tmp/log/nginx/file.error.log;
        root   /mnt/sdb/www/file;
        index  index.html index.htm;

        location ~ [^/]\.php$ {
    fastcgi_split_path_info ^(.+?\.php)(/.*)$;
    if (!-f $document_root$fastcgi_script_name) {
        return 404;
    }
    fastcgi_index index.php;
    fastcgi_param  SCRIPT_FILENAME $document_root$fastcgi_script_name;
    fastcgi_pass    127.0.0.1:9000;

    include fastcgi_params;
}
}
################################

求帮助！！！

hyird, 2021-04-28:
看下可道云 那个luci咋写的 去抄下作业

yaott2020-666, 2021-04-29:
懂了，晚上试试

closed 2021-04-30 by yaott2020-666