coolsnowwolf/lede #6784: 编译报错

azgh, 2021-05-01:
反馈bug/问题模板

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [*] 没有类似的issue

## 2. 上周编译成功，本周编译报错

### (1) 具体问题

A：执行'./scripts/feeds update -a && ./scripts/feeds install -a'提示如下：
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'v2ray-plugin', which does not exist

### (2) 路由器型号和固件版本

A：nanopi-r2s

### (3) 详细日志
```
:~$ cd lede/
:~/lede$ git pull
Already up to date.
:~/lede$ ./scripts/feeds update -a && ./scripts/feeds install -a
Updating feed 'packages' from 'https://github.com/coolsnowwolf/packages' ...
Already up to date.
Create index file './feeds/packages.index' 
Updating feed 'luci' from 'https://github.com/coolsnowwolf/luci' ...
Already up to date.
Create index file './feeds/luci.index' 
Updating feed 'routing' from 'https://git.openwrt.org/feed/routing.git' ...
Already up to date.
Create index file './feeds/routing.index' 
Updating feed 'telephony' from 'https://git.openwrt.org/feed/telephony.git' ...
Already up to date.
Create index file './feeds/telephony.index' 
Updating feed 'helloworld' from 'https://github.com/fw876/helloworld' ...
Already up to date.
Create index file './feeds/helloworld.index' 
WARNING: Makefile 'package/feeds/helloworld/luci-app-ssr-plus/Makefile' has a dependency on 'v2ray-plugin', which does not exist
Installing all packages from feed packages.
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
Installing all packages from feed helloworld.
```
执行`make -j1 V=s`仍然提示‘v2ray-plugin’相关错误提示。

kexix, 2021-05-04:
建议先git pull ，然后update install，再次二次编译

azgh, 2021-05-05:
> 建议先git pull ，然后update install，再次二次编译

是的，先git pull，然后update install，仍提示v2ray-plugin不存在。

Cathgao, 2021-05-06:
别用以前的配置文件，重新menuconfig

azgh, 2021-05-07:
> 别用以前的配置文件，重新menuconfig

重新menuconfig，取消v2ray-plugin后，编译成功了，但是不是以后v2ray-plugin就不可用了？

Cathgao, 2021-05-07:
> ，取消v2ray-plugin后，编译成功了，但是不是以后v2ray-plugin就不可用了？

被xray替代

closed 2021-05-27 by azgh