coolsnowwolf/lede #6785: 内核问题请教

Guxing0321, 2021-05-02:
Hi Team,

最近发现X86的固件总是报一个内核文件找不到，找了几个帖子，我没找到一个好的解决办法

Failed to load DMC firmware i915/kbl_dmc_ver1_04.bin. Disabling runtime power management

[image]

BR
Robin

sxx1314, 2021-05-02:
自行把相应firmware编译进去即可

Guxing0321, 2021-05-03:
Hi sxx1314，

可以提供下教程不，我这边找了几个链接，都没有说清楚
https://koolshare.cn/thread-178689-1-1.html
[image]
这里面提供的步骤，我没在menuconfig 里面找到

BR
Robin

Promix953, 2021-05-04:
```
echo 'CONFIG_EXTRA_FIRMWARE="i915/skl_dmc_ver1_27.bin"' >> target/linux/x86/config-5.4
echo 'CONFIG_EXTRA_FIRMWARE_DIR="/lib/firmware"' >> target/linux/x86/config-5.4
```

Guxing0321, 2021-05-04:
Hi @Promix953

好的，我来试下

BR
RobiN

Guxing0321, 2021-05-04:
Hi @Promix953

添加了两行配置之后，报了一个错误
` No rule to make target '/lib/firmware/i915/kbl_dmc_ver1_04.bin', needed by 'drivers/base/firmware_loader/builtin/i915/kbl_dmc_ver1_04.bin.gen.o'.  Stop.`

[image]

编译的全部日志如下
[11_Compile the firmware.txt](https://github.com/coolsnowwolf/lede/files/6421163/11_Compile.the.firmware.txt)

请问，我需要怎么操作，还是说哪边少了一步？

BR
Robin

Promix953, 2021-05-05:
我用的云编译一直是没问题的，这个错误是找不到`kbl_dmc_ver1_04.bin`文件，刚去看了下[https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/i915](url)是有这个文件的，你检查下自己的环境

Promix953, 2021-05-05:
想起来了，编译的系统记得安装linux-firmware
`sudo apt install linux-firmware`

Guxing0321, 2021-05-06:
Hi @Promix953  @sxx1314 ,

我这边找到了解决方案了，修改的代码如下，感谢大佬们的协助
`
# 创建文件夹并下载文件
sudo mkdir /lib/firmware/i915
sudo curl -L https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/i915/kbl_dmc_ver1_04.bin -o /lib/firmware/i915/kbl_dmc_ver1_04.bin

# 添加 'i915/kbl_dmc_ver1_04.bin'
CONFIG_FIRMWARE_IN_KERNEL=y
echo 'CONFIG_FIRMWARE_IN_KERNEL=y' >> target/linux/x86/config-5.4
echo 'CONFIG_EXTRA_FIRMWARE="i915/kbl_dmc_ver1_04.bin"' >> target/linux/x86/config-5.4
echo 'CONFIG_EXTRA_FIRMWARE_DIR="/lib/firmware"' >> target/linux/x86/config-5.4

`

BR
Robin

closed 2021-05-06 by Guxing0321

newtamax, 2022-07-13:
> Hi @Promix953 @sxx1314 ,
> 
> 我这边找到了解决方案了，修改的代码如下，感谢大佬们的协助 `
> 
> ##### 1. 创建文件夹并下载文件
> sudo mkdir /lib/firmware/i915 sudo curl -L https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/plain/i915/kbl_dmc_ver1_04.bin -o /lib/firmware/i915/kbl_dmc_ver1_04.bin
> 
> ##### 2. 添加 'i915/kbl_dmc_ver1_04.bin'
> echo 'CONFIG_FIRMWARE_IN_KERNEL=y' >> target/linux/x86/config-5.4 echo 'CONFIG_EXTRA_FIRMWARE="i915/kbl_dmc_ver1_04.bin"' >> target/linux/x86/config-5.4 echo 'CONFIG_EXTRA_FIRMWARE_DIR="/lib/firmware"' >> target/linux/x86/config-5.4
> 
> `
> 
> BR Robin

你好，我也有这个问题，请问你的两个步骤是不是，步骤1是添加文件到自己的编译系统，步骤2是添加支持到openwrt的配置文件？如果是这样，我这么做了还是有出错信息，是我哪里出了问题？谢谢。