coolsnowwolf/lede #6789: 不知道单线多拨最近改了什么代码，重启路由器就怎么都拨不上了。

xxp2277, 2021-05-04:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
[x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：wndr3800路由，ath79分支编译的，最近编译的版本会出现：单线直接wan口pppoe拨号正常，luci界面改为单线多拨正常（3拨）；保持多拨状态重启，无法拨号（日志随后附上），然后重启后关掉多拨（从luci界面）改为wan口直接pppoe正常，再改为多拨也正常。个人怀疑是多拨相关的脚本没启动或者有可能冲突之类的。
大神麻烦看下可能是哪里的问题，我这边积极配合查找原因。

### (2) 路由器型号和固件版本

A：wndr3800

### (3) 详细日志

A：Tue May  4 12:58:23 2021 kern.notice kernel: [    0.000000] Linux version 4.19.138 (xxp@Home) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3184-e2c456836)) #0 Mon May 3 15:57:20 2021
Tue May  4 12:58:23 2021 kern.info kernel: [    0.000000] bootconsole [early0] enabled
Tue May  4 12:58:23 2021 kern.info kernel: [    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
Tue May  4 12:58:23 2021 kern.info kernel: [    0.000000] MIPS: machine is Netgear WNDR3800
Tue May  4 12:58:23 2021 kern.info kernel: [    0.000000] SoC: Atheros AR7161 rev 2
Tue May  4 12:58:23 2021 kern.info kernel: [    0.000000] Determined physical RAM map:
Tue May  4 12:58:23 2021 kern.info kernel: [    0.000000]  memory: 08000000 @ 00000000 (usable)
Tue May  4 12:58:23 2021 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Tue May  4 12:58:23 2021 kern.warn kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Tue May  4 12:58:23 2021 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
Tue May  4 12:58:23 2021 kern.info kernel: [    0.000000] Zone ranges:
Tue May  4 12:58:23 2021 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
Tue May  4 12:58:23 2021 kern.info kernel: [    0.000000] Movable zone start for each node
Tue May  4 12:58:23 2021 kern.info kernel: [    0.000000] Early memory node ranges
Tue May  4 12:58:23 2021 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
Tue May  4 12:58:23 2021 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
Tue May  4 12:58:23 2021 kern.debug kernel: [    0.000000] On node 0 totalpages: 32768
Tue May  4 12:58:23 2021 kern.debug kernel: [    0.000000]   Normal zone: 288 pages used for memmap
Tue May  4 12:58:23 2021 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Tue May  4 12:58:23 2021 kern.debug kernel: [    0.000000]   Normal zone: 32768 pages, LIFO batch:7
Tue May  4 12:58:23 2021 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x98/0x4a4 with crng_init=0
Tue May  4 12:58:23 2021 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Tue May  4 12:58:23 2021 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0
Tue May  4 12:58:23 2021 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
Tue May  4 12:58:23 2021 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs,jffs2
Tue May  4 12:58:23 2021 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
Tue May  4 12:58:23 2021 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
Tue May  4 12:58:23 2021 kern.info kernel: [    0.000000] Writing ErrCtl register=00000000
Tue May  4 12:58:23 2021 kern.info kernel: [    0.000000] Readback ErrCtl register=00000000
Tue May  4 12:58:23 2021 kern.info kernel: [    0.000000] Memory: 122236K/131072K available (4439K kernel code, 180K rwdata, 1048K rodata, 1236K init, 206K bss, 8836K reserved, 0K cma-reserved)
Tue May  4 12:58:23 2021 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Tue May  4 12:58:23 2021 kern.info kernel: [    0.000000] NR_IRQS: 51
Tue May  4 12:58:23 2021 kern.info kernel: [    0.000000] CPU clock: 800.000 MHz
Tue May  4 12:58:23 2021 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4778151116 ns
Tue May  4 12:58:23 2021 kern.info kernel: [    0.000007] sched_clock: 32 bits at 400MHz, resolution 2ns, wraps every 5368709118ns
Tue May  4 12:58:23 2021 kern.info kernel: [    0.007842] Calibrating delay loop... 531.45 BogoMIPS (lpj=1062912)
Tue May  4 12:58:23 2021 kern.info kernel: [    0.046111] pid_max: default: 32768 minimum: 301
Tue May  4 12:58:23 2021 kern.info kernel: [    0.050949] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
Tue May  4 12:58:23 2021 kern.info kernel: [    0.057614] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
Tue May  4 12:58:23 2021 kern.info kernel: [    0.068410] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
Tue May  4 12:58:23 2021 kern.info kernel: [    0.078245] futex hash table entries: 256 (order: -1, 3072 bytes)
Tue May  4 12:58:23 2021 kern.info kernel: [    0.084480] pinctrl core: initialized pinctrl subsystem
Tue May  4 12:58:23 2021 kern.info kernel: [    0.090471] NET: Registered protocol family 16
Tue May  4 12:58:23 2021 kern.info kernel: [    0.400050] PCI host bridge /ahb/apb/pcie-controller@17010000 ranges:
Tue May  4 12:58:23 2021 kern.info kernel: [    0.406544]  MEM 0x0000000010000000..0x0000000016ffffff
Tue May  4 12:58:23 2021 kern.info kernel: [    0.411799]   IO 0x0000000000000000..0x0000000000000000
Tue May  4 12:58:23 2021 kern.info kernel: [    0.432702] PCI host bridge to bus 0000:00
Tue May  4 12:58:23 2021 kern.info kernel: [    0.436877] pci_bus 0000:00: root bus resource [mem 0x10000000-0x16ffffff]
Tue May  4 12:58:23 2021 kern.info kernel: [    0.443801] pci_bus 0000:00: root bus resource [io  0x0000]
Tue May  4 12:58:23 2021 kern.info kernel: [    0.449411] pci_bus 0000:00: root bus resource [??? 0x00000000 flags 0x0]
Tue May  4 12:58:23 2021 kern.info kernel: [    0.456235] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
Tue May  4 12:58:23 2021 kern.debug kernel: [    0.464617] pci 0000:00:11.0: [168c:ff1d] type 00 class 0x020000
[... 308 lines cut ...]
Tue May  4 12:58:58 2021 daemon.notice procd: /etc/rc.d/S96led: setting up led WAN LED (green)
Tue May  4 12:58:58 2021 daemon.notice procd: /etc/rc.d/S96led: setting up led USB
Tue May  4 12:58:59 2021 daemon.info dnsmasq[3487]: Connected to system UBus
Tue May  4 12:58:59 2021 user.notice mwan3[1802]: Using firewall mask 0x3F00
Tue May  4 12:58:59 2021 user.notice mwan3[1802]: Max interface count is 60
Tue May  4 12:59:01 2021 cron.err crond[3699]: crond (busybox 1.31.1) started, log level 8
Tue May  4 12:59:02 2021 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Tue May  4 12:59:04 2021 user.info wsdd2[4103]: starting.
Tue May  4 12:59:05 2021 daemon.notice procd: /etc/rc.d/S99zerotier: disabled in config
Tue May  4 12:59:05 2021 daemon.info procd: - init complete -
Tue May  4 12:59:08 2021 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 00:00:00:00:d4:ef
Tue May  4 12:59:10 2021 daemon.info hostapd: wlan0: STA 00:00:00:00:d4:ef IEEE 802.11: authenticated
Tue May  4 12:59:10 2021 daemon.info hostapd: wlan0: STA 00:00:00:00:d4:ef IEEE 802.11: associated (aid 1)
Tue May  4 12:59:10 2021 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 00:00:00:00:d4:ef
Tue May  4 12:59:10 2021 daemon.info hostapd: wlan0: STA 00:00:00:00:d4:ef WPA: pairwise key handshake completed (RSN)
Tue May  4 12:59:14 2021 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 00:00:00:00:d4:ef
Tue May  4 12:59:14 2021 daemon.info hostapd: wlan0: STA 00:00:00:00:d4:ef IEEE 802.11: authenticated
Tue May  4 12:59:14 2021 daemon.info hostapd: wlan0: STA 00:00:00:00:d4:ef IEEE 802.11: associated (aid 1)
Tue May  4 12:59:14 2021 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 00:00:00:00:d4:ef
Tue May  4 12:59:14 2021 daemon.info hostapd: wlan0: STA 00:00:00:00:d4:ef WPA: pairwise key handshake completed (RSN)
Tue May  4 12:59:42 2021 daemon.info hostapd: wlan1: STA 00:00:00:00:bc:af IEEE 802.11: authenticated
Tue May  4 12:59:42 2021 daemon.info hostapd: wlan1: STA 00:00:00:00:bc:af IEEE 802.11: associated (aid 1)
Tue May  4 12:59:42 2021 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 00:00:00:00:bc:af
Tue May  4 12:59:42 2021 daemon.info hostapd: wlan1: STA 00:00:00:00:bc:af WPA: pairwise key handshake completed (RSN)
Tue May  4 13:00:20 2021 daemon.err uhttpd[2374]: killall: pppd: no process killed

zxlhhyccc, 2021-05-05:
添加一个检测并自动拨号脚本吧。。。

xxp2277, 2021-05-06:
能否给个链接？

galaxyskyknight, 2021-05-06:
吧STP关掉试试

xxp2277, 2021-05-08:
> 
> 
> 吧STP关掉试试

没懂，是指我的网络，交换机划分的vlan么？

xxp2277, 2021-05-15:
2021-05-15：找到表面的原因，感觉是macvlan没有建立对应的虚拟网卡：手动添加如下代码到启动项可以正常双拨。（但明显下面的代码没有开机启动）哪位告知下如何排查原因
```
ip link add link eth1 name macvlan1 type macvlan
ifconfig macvlan1 hw ether xx:xx:xx:xx:xx:xx
ifconfig macvlan1 up
ip link add link eth1 name macvlan2 type macvlan
ifconfig macvlan2 hw ether xx:xx:xx:xx:xx:xx
ifconfig macvlan2 up
```

closed 2021-06-17 by coolsnowwolf