coolsnowwolf/lede #6790: 可道云新增的依赖导致编译失败

HoldOnBro, 2021-05-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：可道云新增的依赖导致编译失败

### (2) 路由器型号和固件版本

A：所有型号

### (3) 详细日志

A：

`Configuring extensions
checking io.h usability... no
checking io.h presence... no
checking for io.h... no
checking for strtoll... yes
checking for atoll... yes
checking for use of system timezone data... no
checking whether to build with LIBXML support... no
checking for OpenSSL support... no
checking for Kerberos support... no
checking whether to use system default cipher list instead of hardcoded value... no
checking for libpcre2-8 >= 10.30... yes
checking for JIT support in PCRE2... (cached) no
checking whether to enable the SQLite3 extension... yes, shared
checking for sqlite3 > 3.7.4... yes
checking for sqlite3_errstr in -lsqlite3... yes
checking for sqlite3_load_extension in -lsqlite3... yes
checking for ZLIB support... yes
checking for zlib >= 1.2.0.4... yes
checking whether to enable bc style precision math functions... no
checking for BZip2 support... no
checking whether to enable calendar conversion support... no
checking whether to enable ctype functions... no
checking for cURL support... yes, shared
checking for libcurl >= 7.15.5... yes
checking for SSL support in libcurl... yes
checking for libcurl linked against old openssl... no
checking for curl_easy_perform in -lcurl... yes
checking for QDBM support... no
checking for GDBM support... no
checking for NDBM support... no
checking for TCADB support... no
checking for LMDB support... no
checking for Berkeley DB4 support... no
checking for Berkeley DB3 support... no
checking for Berkeley DB2 support... no
checking for DB1 support... no
checking for DBM support... no
checking for CDB support... no
checking for INI File support... no
checking for FlatFile support... no
checking whether to enable DBA interface... no
checking whether to enable DOM support... yes, shared
**configure: error: DOM extension requires LIBXML extension, add --with-libxml**
make[3]: *** [Makefile:588: /home/xyz/lede/build_dir/target-aarch64_cortex-a53_musl/php-7.4.7/.configured_8429fb15ef37454d409faa428ef01aa3] Error 1
make[3]: Leaving directory '/home/xyz/lede/feeds/packages/lang/php7'
time: package/feeds/packages/php7/compile#6.96#1.70#6.76
make[2]: *** [package/Makefile:113: package/feeds/packages/php7/compile] Error 2
make[2]: Leaving directory '/home/xyz/lede'
make[1]: *** [package/Makefile:107: /home/xyz/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2`

mingxiaoyu, 2021-05-04:
Makefile:584: recipe for target '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/php-7.4.7/.configured_58231bf0c0b4e6c1c27cd20617428c39' failed
make[3]: *** [/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/php-7.4.7/.configured_58231bf0c0b4e6c1c27cd20617428c39] Error 1
make[2]: *** [package/feeds/packages/php7/compile] Error 2
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/lang/php7'
make[1]: *** [/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
time: package/feeds/packages/php7/compile#8.28#4.78#10.51
package/Makefile:111: recipe for target 'package/feeds/packages/php7/compile' failed
make[2]: Leaving directory '/workdir/openwrt'
package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/workdir/openwrt'
make: *** [world] Error 2
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed

----------------
same issue

liwenjie119, 2021-05-04:
checking whether to enable DOM support... yes, shared
configure: error: DOM extension requires LIBXML extension, add --with-libxml
Makefile:584: recipe for target '/workdir/openwrt/build_dir/target-x86_64_musl/php-7.4.7/.configured_58231bf0c0b4e6c1c27cd20617428c39' failed
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/lang/php7'
make[3]: *** [/workdir/openwrt/build_dir/target-x86_64_musl/php-7.4.7/.configured_58231bf0c0b4e6c1c27cd20617428c39] Error 1
time: package/feeds/packages/php7/compile#8.39#3.69#10.58
make[2]: *** [package/feeds/packages/php7/compile] Error 2
package/Makefile:111: recipe for target 'package/feeds/packages/php7/compile' failed
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make: *** [world] Error 2
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
Error: Process completed with exit code 2.

WXfeng-boy, 2021-05-05:
一样

bleach1991, 2021-05-05:
差这个 PHP7_LIBXML
到languages -》php-》php7-cgi选中就有了

bleach1991, 2021-05-05:
已经修复了

closed 2021-05-05 by coolsnowwolf