coolsnowwolf/lede #6797: LINKSYS WRT1900AC  mvebu/400-find_active_root.patch ERROR

codesnas, 2021-05-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：Applying /workdir/openwrt/target/linux/mvebu/patches-5.4/400-find_active_root.patch using plaintext: 
can't find file to patch at input line 8

### (2) 路由器型号和固件版本

A：LINKSYS WRT1900AC

### (3) 详细日志

A：
Applying /workdir/openwrt/target/linux/mvebu/patches-5.4/318-ARM-dts-turris-omnia-fix-hardware-buffer-management.patch using plaintext: 
patching file arch/arm/boot/dts/armada-385-turris-omnia.dts

Applying /workdir/openwrt/target/linux/mvebu/patches-5.4/318-armada-xp-linksys-mamba-resize-kernel.patch using plaintext: 
patching file arch/arm/boot/dts/armada-xp-linksys-mamba.dts

Applying /workdir/openwrt/target/linux/mvebu/patches-5.4/320-armada-370-dts-fix-crypto-engine.patch using plaintext: 
patching file arch/arm/boot/dts/armada-370.dtsi

Applying /workdir/openwrt/target/linux/mvebu/patches-5.4/400-find_active_root.patch using plaintext: 
can't find file to patch at input line 8
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|The WRT1900AC among other Linksys routers uses a dual-firmware layout.
|Dynamically rename the active partition to "ubi".
|
|Signed-off-by: Imre Kaloz <kaloz@openwrt.org>
|
|--- a/drivers/mtd/parsers/ofpart_core.c
|+++ b/drivers/mtd/parsers/ofpart_core.c
make[4]: *** [/workdir/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/linux-5.4.115/.prepared_4d345f5b7d084c7d7ccf40e2c353beff] Error 1
--------------------------
No file to patch.  Skipping patch.
4 out of 4 hunks ignored
Patch failed!  Please fix /workdir/openwrt/target/linux/mvebu/patches-5.4/400-find_active_root.patch!
make[3]: *** [compile] Error 2
Makefile:19: recipe for target '/workdir/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/linux-5.4.115/.prepared_4d345f5b7d084c7d7ccf40e2c353beff' failed
make[4]: Leaving directory '/workdir/openwrt/target/linux/mvebu'
Makefile:13: recipe for target 'compile' failed
make[3]: Leaving directory '/workdir/openwrt/target/linux'
make[2]: *** [target/linux/compile] Error 2
time: target/linux/compile#11.03#5.13#13.40
target/Makefile:23: recipe for target 'target/linux/compile' failed
make[2]: Leaving directory '/workdir/openwrt'
target/Makefile:18: recipe for target '/workdir/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/stamp/.target_compile' failed
make[1]: *** [/workdir/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/stamp/.target_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2
Error: Process completed with exit code 2.

whoshoe, 2021-05-07:
我linksys wrt32x编译也是这个错误，开了一个issue被关掉了。看到了一个相关的upstream更新的commit，今天再试试。

whoshoe, 2021-05-07:
刚刚重试了，也是同样的问题。

Applying /workdir/openwrt/target/linux/mvebu/patches-5.4/400-find_active_root.patch using plaintext: 
can't find file to patch at input line 8
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|The WRT1900AC among other Linksys routers uses a dual-firmware layout.
|Dynamically rename the active partition to "ubi".
make[4]: *** [/workdir/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/linux-5.4.115/.prepared_4d345f5b7d084c7d7ccf40e2c353beff] Error 1
make[3]: *** [compile] Error 2
|
|Signed-off-by: Imre Kaloz <kaloz@openwrt.org>
|
|--- a/drivers/mtd/parsers/ofpart_core.c
|+++ b/drivers/mtd/parsers/ofpart_core.c
--------------------------
No file to patch.  Skipping patch.
4 out of 4 hunks ignored
Patch failed!  Please fix /workdir/openwrt/target/linux/mvebu/patches-5.4/400-find_active_root.patch!
Makefile:19: recipe for target '/workdir/openwrt/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/linux-mvebu_cortexa9/linux-5.4.115/.prepared_4d345f5b7d084c7d7ccf40e2c353beff' failed
make[4]: Leaving directory '/workdir/openwrt/target/linux/mvebu'
Makefile:13: recipe for target 'compile' failed
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/compile#11.93#5.69#14.70
target/Makefile:23: recipe for target 'target/linux/compile' failed
make[2]: Leaving directory '/workdir/openwrt'
make[2]: *** [target/linux/compile] Error 2
target/Makefile:18: recipe for target '/workdir/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/stamp/.target_compile' failed
make[1]: *** [/workdir/openwrt/staging_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/stamp/.target_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [world] Error 2
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
Error: Process completed with exit code 2.

allenfoxtc, 2021-05-07:
> 我linksys wrt32x编译也是这个错误，开了一个issue被关掉了。看到了一个相关的upstream更新的commit，今天再试试。

是这个commit吗？https://github.com/openwrt/openwrt/commit/7a7b2fd809809fbd7045bd3dad4fc896a6fef06f#diff-c9baa3bff8edfbf9aa8abf75efd95bfded3c5dd742e1cf1921859952568eb166

这个也不行的 原因是ofpart_core.c这个文件在linux kernel的master分支 openwrt用的5.4.115-116里面还是老的ofpart.c
具体可以看
kernel tree: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/mtd/parsers?h=v5.4.115
ofpart_core.c commit: https://github.com/torvalds/linux/commits/master/drivers/mtd/parsers/ofpart_core.c

我试过把400-find_active_root.patch恢复成ofpart.c的版本 但是编译最后会报另一个错
`drivers/net/dsa/mv88e6xxx/chip.c:5079:4: error: 'struct dsa_switch' has no member named 'assisted_learning_on_cpu_port'`
这个应该也是同样问题

codesnas, 2021-05-07:
Linksys_WRT1900ACS
Linksys_WRT3200ACM
Linksys_WRT32X

编译时报相同的错误

nivalxer, 2021-05-07:
上游在patch https://github.com/openwrt/openwrt/blob/master/target/linux/generic/backport-5.4/403-v5.13-mtd-parsers-ofpart-support-BCM4908-fixed-partitions.patch 中做了一个{ofpart.c => ofpart_core.c}，这个patch在这边没有，所以会找不到ofpart_core.c这个文件。

AmadeusGhost, 2021-05-08:
有人可以试试这个 commit 能编译过吗：https://github.com/AmadeusGhost/lede/commit/347df8630cc43104b1176b62da761bbbcdad5b50

jiahanbryant, 2021-05-08:
hm sorry, 请问怎么加，我尝试了
```
git cherry-pick -n d01f12a2374390329b372216402046bacd2a76d9
fatal: bad object d01f12a2374390329b372216402046bacd2a76d9
```

AmadeusGhost, 2021-05-08:
> hm sorry, 请问怎么加，我尝试了

```
https://github.com/AmadeusGhost/lede/commit/d01f12a2374390329b372216402046bacd2a76d9.patch
git apply d01f12a2374390329b372216402046bacd2a76d9.patch
```

jiahanbryant, 2021-05-08:
```
$ git apply d01f12a2374390329b372216402046bacd2a76d9.patch
d01f12a2374390329b372216402046bacd2a76d9.patch:181: trailing whitespace.

d01f12a2374390329b372216402046bacd2a76d9.patch:182: space before tab in indent.
 			/* fastpath: update of existing entry */
d01f12a2374390329b372216402046bacd2a76d9.patch:183: space before tab in indent.
 			if (unlikely(source != fdb->dst && !fdb->is_sticky)) {
d01f12a2374390329b372216402046bacd2a76d9.patch:185: space before tab in indent.
 				fdb->dst = source;
d01f12a2374390329b372216402046bacd2a76d9.patch:186: space before tab in indent.
 				fdb_modified = true;
warning: squelched 226 whitespace errors
warning: 231 lines add whitespace errors.
```
有一些warning..不过可以继续往下build了

jiahanbryant, 2021-05-08:
ok 我build成功了

whoshoe, 2021-05-08:
> ok 我build成功了
试用一下，没问题的话可以提交一个PR了。

closed 2021-05-08 by coolsnowwolf

jiahanbryant, 2021-05-08:
so far so good!