coolsnowwolf/lede #6799: RPC: fragement too large

galaxyskyknight, 2021-05-07:
网卡高负载下，每一分钟打一个。

Fri May  7 00:00:27 2021 kern.notice kernel: [516015.563335] RPC: fragment too large: 323119476
Fri May  7 00:01:27 2021 kern.notice kernel: [516075.752707] RPC: fragment too large: 323119476
Fri May  7 00:02:27 2021 kern.notice kernel: [516135.840975] RPC: fragment too large: 323119476
Fri May  7 00:03:27 2021 kern.notice kernel: [516195.941714] RPC: fragment too large: 323119476
Fri May  7 00:04:27 2021 kern.notice kernel: [516256.020894] RPC: fragment too large: 323119476
Fri May  7 00:05:26 2021 kern.notice kernel: [516315.118608] RPC: fragment too large: 323119476
Fri May  7 00:06:26 2021 kern.notice kernel: [516375.222337] RPC: fragment too large: 323119476
Fri May  7 00:07:26 2021 kern.notice kernel: [516435.302458] RPC: fragment too large: 323119476
Fri May  7 00:08:26 2021 kern.notice kernel: [516495.516479] RPC: fragment too large: 323119476
Fri May  7 00:09:27 2021 kern.notice kernel: [516555.736149] RPC: fragment too large: 323119476
Fri May  7 00:10:27 2021 kern.notice kernel: [516615.764187] RPC: fragment too large: 323119476
Fri May  7 00:11:27 2021 kern.notice kernel: [516675.902361] RPC: fragment too large: 323119476
Fri May  7 00:12:27 2021 kern.notice kernel: [516736.029621] RPC: fragment too large: 323119476
Fri May  7 00:13:26 2021 kern.notice kernel: [516795.173663] RPC: fragment too large: 323119476
Fri May  7 00:14:27 2021 kern.notice kernel: [516855.550711] RPC: fragment too large: 323119476
Fri May  7 00:15:27 2021 kern.notice kernel: [516915.634004] RPC: fragment too large: 323119476
Fri May  7 00:16:27 2021 kern.notice kernel: [516975.753756] RPC: fragment too large: 323119476
Fri May  7 00:17:27 2021 kern.notice kernel: [517035.971182] RPC: fragment too large: 323119476
Fri May  7 00:18:26 2021 kern.notice kernel: [517095.426758] RPC: fragment too large: 323119476
Fri May  7 00:19:26 2021 kern.notice kernel: [517155.490551] RPC: fragment too large: 323119476
Fri May  7 00:20:27 2021 kern.notice kernel: [517215.773619] RPC: fragment too large: 323119476
Fri May  7 00:21:26 2021 kern.notice kernel: [517275.318027] RPC: fragment too large: 323119476
Fri May  7 00:22:26 2021 kern.notice kernel: [517335.453286] RPC: fragment too large: 323119476
Fri May  7 00:23:27 2021 kern.notice kernel: [517395.569006] RPC: fragment too large: 323119476
Fri May  7 00:24:27 2021 kern.notice kernel: [517455.705735] RPC: fragment too large: 323119476
Fri May  7 00:25:27 2021 kern.notice kernel: [517515.982439] RPC: fragment too large: 323119476
Fri May  7 00:26:26 2021 kern.notice kernel: [517575.180585] RPC: fragment too large: 323119476
Fri May  7 00:27:26 2021 kern.notice kernel: [517635.316886] RPC: fragment too large: 323119476
Fri May  7 00:28:27 2021 kern.notice kernel: [517695.839279] RPC: fragment too large: 323119476
Fri May  7 00:29:27 2021 kern.notice kernel: [517755.888452] RPC: fragment too large: 323119476
Fri May  7 00:30:27 2021 kern.notice kernel: [517816.019783] RPC: fragment too large: 323119476
Fri May  7 00:31:26 2021 kern.notice kernel: [517875.207804] RPC: fragment too large: 323119476
Fri May  7 00:32:26 2021 kern.notice kernel: [517935.297727] RPC: fragment too large: 323119476
Fri May  7 00:33:26 2021 kern.notice kernel: [517995.429754] RPC: fragment too large: 323119476
Fri May  7 00:34:27 2021 kern.notice kernel: [518055.646985] RPC: fragment too large: 323119476
Fri May  7 00:35:27 2021 kern.notice kernel: [518115.762570] RPC: fragment too large: 323119476
Fri May  7 00:36:28 2021 kern.notice kernel: [518177.039674] RPC: fragment too large: 323119476
Fri May  7 00:37:27 2021 kern.notice kernel: [518236.172860] RPC: fragment too large: 323119476
Fri May  7 00:38:27 2021 kern.notice kernel: [518296.324403] RPC: fragment too large: 323119476
Fri May  7 00:39:27 2021 kern.notice kernel: [518356.521857] RPC: fragment too large: 323119476
Fri May  7 00:40:28 2021 kern.notice kernel: [518416.813541] RPC: fragment too large: 323119476
Fri May  7 00:41:08 2021 daemon.notice miniupnpd[16759]: removed 34 unused rules
Fri May  7 00:41:28 2021 kern.notice kernel: [518476.931793] RPC: fragment too large: 323119476
Fri May  7 00:42:27 2021 kern.notice kernel: [518536.068928] RPC: fragment too large: 323119476
Fri May  7 00:43:27 2021 kern.notice kernel: [518596.211349] RPC: fragment too large: 323119476
Fri May  7 00:44:27 2021 kern.notice kernel: [518656.359952] RPC: fragment too large: 323119476
Fri May  7 00:45:27 2021 kern.notice kernel: [518716.549671] RPC: fragment too large: 323119476
Fri May  7 00:46:28 2021 kern.notice kernel: [518776.678004] RPC: fragment too large: 323119476
Fri May  7 00:47:28 2021 kern.notice kernel: [518836.867249] RPC: fragment too large: 323119476
Fri May  7 00:48:28 2021 kern.notice kernel: [518896.956275] RPC: fragment too large: 323119476
Fri May  7 00:49:27 2021 kern.notice kernel: [518956.149954] RPC: fragment too large: 323119476
Fri May  7 00:50:27 2021 kern.notice kernel: [519016.262663] RPC: fragment too large: 323119476
Fri May  7 00:51:27 2021 kern.notice kernel: [519076.405858] RPC: fragment too large: 323119476
Fri May  7 00:52:27 2021 kern.notice kernel: [519136.453286] RPC: fragment too large: 323119476
Fri May  7 00:53:28 2021 kern.notice kernel: [519196.787454] RPC: fragment too large: 323119476
Fri May  7 00:54:28 2021 kern.notice kernel: [519257.053654] RPC: fragment too large: 323119476
Fri May  7 00:55:27 2021 kern.notice kernel: [519316.197307] RPC: fragment too large: 323119476
Fri May  7 00:56:27 2021 kern.notice kernel: [519376.291339] RPC: fragment too large: 323119476
Fri May  7 00:57:27 2021 kern.notice kernel: [519436.530951] RPC: fragment too large: 323119476
Fri May  7 00:58:28 2021 kern.notice kernel: [519496.656214] RPC: fragment too large: 323119476
Fri May  7 00:59:28 2021 kern.notice kernel: [519556.859857] RPC: fragment too large: 323119476
Fri May  7 01:00:28 2021 kern.notice kernel: [519616.953385] RPC: fragment too large: 323119476
Fri May  7 01:01:00 2021 cron.err crond[2814]: USER root pid 11821 cmd sh /usr/share/jd-dailybonus/newapp.sh -w
Fri May  7 01:01:00 2021 cron.err crond[2814]: USER root pid 11822 cmd sh /usr/share/jd-dailybonus/newapp.sh -u
Fri May  7 01:01:27 2021 kern.notice kernel: [519676.137864] RPC: fragment too large: 323119476
Fri May  7 01:02:27 2021 kern.notice kernel: [519736.505683] RPC: fragment too large: 323119476
Fri May  7 01:03:27 2021 kern.notice kernel: [519796.613100] RPC: fragment too large: 323119476
Fri May  7 01:04:28 2021 kern.notice kernel: [519856.747781] RPC: fragment too large: 323119476
[... 1468 lines cut ...]
Fri May  7 21:08:30 2021 kern.notice kernel: [592101.139936] RPC: fragment too large: 323119476
Fri May  7 21:09:31 2021 kern.notice kernel: [592162.331563] RPC: fragment too large: 323119476
Fri May  7 21:10:32 2021 kern.notice kernel: [592222.608955] RPC: fragment too large: 323119476
Fri May  7 21:11:32 2021 kern.notice kernel: [592282.810233] RPC: fragment too large: 323119476
Fri May  7 21:12:31 2021 kern.notice kernel: [592342.048937] RPC: fragment too large: 323119476
Fri May  7 21:13:32 2021 kern.notice kernel: [592403.212841] RPC: fragment too large: 323119476
Fri May  7 21:14:33 2021 kern.notice kernel: [592464.287246] RPC: fragment too large: 323119476
Fri May  7 21:15:33 2021 kern.notice kernel: [592524.489729] RPC: fragment too large: 323119476
Fri May  7 21:16:34 2021 kern.notice kernel: [592584.706193] RPC: fragment too large: 323119476
Fri May  7 21:17:34 2021 kern.notice kernel: [592644.876389] RPC: fragment too large: 323119476
Fri May  7 21:18:33 2021 kern.notice kernel: [592704.110904] RPC: fragment too large: 323119476
Fri May  7 21:19:33 2021 kern.notice kernel: [592764.299411] RPC: fragment too large: 323119476
Fri May  7 21:20:35 2021 kern.notice kernel: [592825.539500] RPC: fragment too large: 323119476
Fri May  7 21:21:36 2021 kern.notice kernel: [592886.793222] RPC: fragment too large: 323119476
Fri May  7 21:22:36 2021 kern.notice kernel: [592947.029760] RPC: fragment too large: 323119476
Fri May  7 21:23:36 2021 kern.notice kernel: [593007.092385] RPC: fragment too large: 323119476
Fri May  7 21:24:36 2021 kern.notice kernel: [593067.398027] RPC: fragment too large: 323119476
Fri May  7 21:25:37 2021 kern.notice kernel: [593127.566306] RPC: fragment too large: 323119476
Fri May  7 21:26:37 2021 kern.notice kernel: [593187.748064] RPC: fragment too large: 323119476
[... cut ...] 19476
Fri May  7 21:28:36 2021 kern.notice kernel: [593307.117697] RPC: fragment too large: 323119476
Fri May  7 21:28:37 2021 authpriv.info dropbear[10095]: Exit (root) from <192.168.2.14:63370>: Exited normally
Fri May  7 21:29:36 2021 kern.notice kernel: [593367.401973] RPC: fragment too large: 323119476
Fri May  7 21:30:38 2021 kern.notice kernel: [593428.626891] RPC: fragment too large: 323119476
Fri May  7 21:31:38 2021 kern.notice kernel: [593488.873759] RPC: fragment too large: 323119476

coolsnowwolf, 2021-05-08:
关闭 NFS

galaxyskyknight, 2021-05-18:
> 关闭 NFS

关键是我nfs是关闭的啊

closed 2021-06-17 by coolsnowwolf